PX4/PX4-Autopilot #17485: V 1.12 vtol pwm limits get always lost with transition to fw

taileron, 2021-04-29:
**Describe the bug**
vtol pwm limits get always lost with transition to fw and remain lost 

**To Reproduce**
Steps to reproduce the behavior:
set the limits required for the vehicle in pwm settings for a optimal tuning
it affects all vtol applications the examples shown with Convergence Airframe selected

**Expected behavior**
All correctly set limits remain in place
behaviour on Fcs with IO 
**Log Files and Screenshots**
MC Mode:
pwm info device: /dev/pwm_output0 channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00) channel 5: 1984 us (default rate: 50 Hz failsafe: 0, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00) channel 6: 1030 us (default rate: 50 Hz failsafe: 0, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00) channel 7: 1500 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00) channel 8: 1500 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00) channel group 0: channels 1 2 channel group 1: channels 5 6 7 8 channel group 2: channels 3 4

transition to FW all upper servo and lower esc pwm limits get lost: 
but limit shift for vt mot off Id ch3 works:
pwm info device: /dev/pwm_output0 channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 900 us, max: 1943 us, trim:  0.00) channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 900 us, max: 1943 us, trim:  0.00) channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 900 us, max: 1000 us, trim:  0.00) channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00) channel 5: 925 us (default rate: 50 Hz failsafe: 0, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00) channel 6: 1665 us (default rate: 50 Hz failsafe: 0, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00) channel 7: 1419 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00) channel 8: 1420 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00) channel group 0: channels 1 2 channel group 1: channels 5 6 7 8 channel group 2: channels 3 4

transition back to MC limits still lost:
device: /dev/pwm_output0 channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00) channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00) channel 5: 1651 us (default rate: 50 Hz failsafe: 0, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00) channel 6: 935 us (default rate: 50 Hz failsafe: 0, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00) channel 7: 1400 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00) channel 8: 1400 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00) channel group 0: channels 1 2 channel group 1: channels 5 6 7 8 channel group 2: channels 3 4

behavior on only FMU fc or IO disabled:
MC mode:
pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1043 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1985 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 1030 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00)
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6

FW mode:
no upper limits are lost here, however the level shift of mot ch 3 (vt mot off id) does not work.
pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1043 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1021 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 2014 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00)
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6

dgeorge83616, 2021-05-05:
We have a similar problem. See flight logs here: https://logs.px4.io/plot_app?log=ad68e025-48f8-42d5-9846-b78e167ea261

This plot is zoomed into the event. Our fork has a modification to automatically initiate a back transition when the front transition completes (yes, it is a temporary hack). The event coincides with the front-to-back transition. The PWM_MAIN_MAX7/8 (1670/1668) parameters become ignored. Prior to that point, the signals were pegged to MAX as expected.
[image]
[image]

LorenzMeier, 2021-05-05:
@RomanBapst @sfuhrer Do you have a quick idea what could be wrong and causing this?

LorenzMeier, 2021-05-05:
@davids5 Could this be PWM driver changes related?

LorenzMeier, 2021-05-05:
@dgeorge83616 Could you re-test with current master? I just merged https://github.com/PX4/PX4-Autopilot/pull/17484, which is likely going to address the root cause. @taileron could you test as well (on the ground)?

sfuhrer, 2021-05-05:
Not sure about the context to this issue here, but https://github.com/PX4/PX4-Autopilot/pull/17534 would change the default min PWM in fixed-wing to 1000 instead of 900.

taileron, 2021-05-05:
seems not to solve it ... only if vt_elev_mc_lock gets enabled and then disabled, then the proper limits are set.

LorenzMeier, 2021-05-06:
@taileron Did you test with #17484 applied? Just want to be sure I understand what you tested.

taileron, 2021-05-06:
took the master from 12h ago thought that #17484 was already merged, as soon as the status of mc elevon lock is changed , the limits get applied:
pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00)
channel 5: 1651 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 6: 935 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 7: 1400 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel 8: 1400 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> param set VT_ELEV_MC_LOCK 0
+ VT_ELEV_MC_LOCK: curr: 1 -> new: 0
nsh> pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1658 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2150 us, trim:  0.00)
channel 6: 978 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2150 us, trim:  0.00)
channel 7: 1440 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2150 us, trim:  0.00)
channel 8: 1717 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2150 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh>

dgeorge83616, 2021-05-06:
> seems not to solve it ... only if vt_elev_mc_lock gets enabled and then disabled, then the proper limits are set.

We use that same workaround for @taileron's issue. My comments above probably pertain to a different bug. I'll log a new issue, if I can verify.

sfuhrer, 2021-05-21:
@dgeorge83616 @taileron have you got closer to finding the issue here?

I encountered something similar: For me, every time a parameter is set, the PWM_MIN and PWM_MAX are updated (I guess in pwm_out) to what is specified in the corresponding params (e.g. `PWM_AUX_MIN`). That's a problem for VTOL because there we actually want the PWM_MIN of the motors being set to `VT_IDLE_PWM_MC`, but only apply the limits when the VTOL mode is changed (transition to FW, transition to MC).
You could test if https://github.com/PX4/PX4-Autopilot/pull/17622 solves anything for you, but I don't think that's the right way forward anyway.

taileron, 2021-05-21:
Unfortunately doesn´t it solve the issue proper limits still are only after change of elevon lock:
with #17622 not only the upper limits are now replaced by default (2000) but also the lower limits by their default (1000).

pwm info
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00)
channel 5: 1651 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 6: 935 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 7: 1400 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel 8: 1400 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> param set VT_ELEV_MC_LOCK 0;sleep 1;param set VT_ELEV_MC_LOCK 1
+ VT_ELEV_MC_LOCK: curr: 1 -> new: 0
* VT_ELEV_MC_LOCK: curr: 0 -> new: 1
nsh> pwm info
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1983 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 1029 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 7: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00)
channel 8: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4

-> Trans to FW now the upper servo and lower esc limits get all lost (replaced by the default):

pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 1943 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 1943 us, trim:  0.00)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1000 us, max: 900 us, trim:  0.00)
channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00)
channel 5: 938 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 6: 1652 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 7: 1449 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel 8: 1392 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4

taileron, 2021-05-25:
After more tests with different fc some only fmu without any io #17622 behaves absolutely identically like before.
The IO outputs always go from the defined to the default limits. The fmu outs always have the desired limits, so this error is not present at fmu, but the mot off id shift does not take place there either.

dagar, 2021-06-01:
@taileron can you give this a try? https://github.com/PX4/PX4-Autopilot/pull/17683

taileron, 2021-06-01:
#17683 tested: limits still lost and the previous workaround no longer has any effect.
the desired limits are now completely replaced by the defaults.
pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1000 us, max: 1943 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1000 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1000 us, max: 1000 us, trim:  0.00)
channel 4: 1990 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2000 us, trim:  0.00)
channel 5: 1025 us (default rate: 50 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 6: 1582 us (default rate: 50 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2000 us, trim:  0.00)
channel 7: 1470 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)
channel 8: 1327 us (default rate: 50 Hz failsafe: 1500, disarmed: 1500 us, min: 800 us, max: 2000 us, trim:  0.00)

HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 0c7075cba966668fc507c08a37ed13ddab5542c9
FW version: 1.12.0 0 (17563648)
FW git-branch: pr-pwm_param_update_limit
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 20376b65d1a8cb2c02bbfe6a494add6238b29051
Build datetime: Jun  2 2021 00:59:41
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00010000000031323634343551160049002d
MCU: STM32F42x, rev. 3

dagar, 2021-06-02:
@taileron do you want to open a PR with your PWM max increase to 2500 us? It seems relatively harmless.

sfuhrer, 2021-06-02:
https://github.com/PX4/PX4-Autopilot/blob/1b671877708098b7c803b940f4a52402bcc64db9/src/modules/vtol_att_control/vtol_type.cpp#L95
That should return the set PWM_MAX right? Is that one not working properly?
@taileron are you sure that it's a regression, has it worked before? I'm just used to PWM ranges from 1000-2000, so personally don't have experience about it.

taileron, 2021-06-02:
@sfuhrer in 1.11 and early 1.12 masters that still worked on I/O outs.
And at fmu outs the limits were and are always treated right (expect the vtol mot off id limit issue that never worked on aux/fmu or uavcan outputs).

dagar, 2021-06-03:
It's a race condition in module startup/configuration. I'm updating the VTOL module to record the PWM min/max/disarm values before they're changed for the first time, rather than only at init.

bkueng, 2021-06-25:
@taileron @dgeorge83616 not sure if https://github.com/PX4/PX4-Autopilot/pull/17803 fixes your problem, but can you test that?

taileron, 2021-06-26:
#17803 seems to have it completely solved. All limits now are correct and don´t change with transitions anymore.
And so far I have not noticed any side effects.
pwm info
device: /dev/pwm_output0
channel 1: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 1000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 1000 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 2000, disarmed: 2000 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1985 us (default rate: 50 Hz failsafe: 0, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 1030 us (default rate: 50 Hz failsafe: 0, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 7: 1500 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00)
channel 8: 1500 us (default rate: 50 Hz failsafe: 0, disarmed: 1500 us, min: 800 us, max: 2200 us, trim:  0.00)

LorenzMeier, 2021-06-28:
I'm closing this since this is reported as fixed. We will still run a full flight test.

closed 2021-06-28 by LorenzMeier

dgeorge83616, 2021-06-28:
> @taileron @dgeorge83616 not sure if #17803 fixes your problem, but can you test that?

Sorry, I only just saw this. I'll try it and open a new issue if necessary.