PX4/PX4-Autopilot #17486: There is packet loss when UAVCAN sends large data

CUAVcaijie, 2021-04-30:
**Describe the bug**
RTK base is connected to QGC and forwarded to the CAN RTK mobile terminal using the CAN port. The mobile terminal detects a packet loss error in UAVCAN data (flip error...)

**To Reproduce**
Use f6a34e1e80ff259838dd15b17b7eb97fd347ee01 version firmware

> equipment:
CUAVv5Nano
F9P base, CAN RTK Mobile
QGC

> parameter settings：
UAVCAN_ENABLE = Sensors Manual Config

**Expected behavior**
You can check the packet loss on the CAN of the mobile terminal

 **The following is the code connection I tested and modified**
https://github.com/cuav/libuavcan/tree/cuav_test
https://github.com/cuav/Firmware/tree/cuav_uavcan_test

**Problem finding process**
Print the UAVACAN package at the send function entry of the src/drivers/uavcan/libuavcan/libuavcan/src/transport/uc_can_io.cpp file. No data was lost.
Print the UAVACAN packet in the sendToIface function and find that it is the same as the data received by CAN RTK Mobile. There is already a packet loss. So I guess the problem is in the queue

CUAVcaijie, 2021-04-30:
This picture shows only the first frame of data
[image]
This is the queue is full
[image]

davids5, 2021-04-30:
@pavel-kirienko ideas?

pavel-kirienko, 2021-04-30:
Try increasing the memory allocation arena here and see if it fixes the problem:

https://github.com/PX4/PX4-Autopilot/blob/4f52c0b6daa93f01b3c02e9457265be107af74be/src/drivers/uavcan/allocator.hpp#L54-L55

CUAVcaijie, 2021-05-03:
Okay, test it when I go back during the holidays

CUAVcaijie, 2021-05-06:
Hello, increase these two values by 200, but the packet loss error still occurs.
[image]

CUAVcaijie, 2021-05-06:
@pavel-kirienko 
Hello, I added a number of enqueue and dequeue data volume statistics, and found that when the queue does not overflow, the queue dequeue data has been lost.

[image]
1 is to print directly using the queue length. 2 is the difference between entering and leaving the team. What is more obvious here is that suddenly there is no data in the queue.

A few minutes later, the difference between entering and leaving the team reached tens of thousands
[image]

This is the code I added to print：
https://github.com/cuav/libuavcan/tree/cuav_test

CUAVcaijie, 2021-05-06:
I blocked my print information, and then increased the allocated memory. There is no packet loss.
More than 160 memory blocks were used, but only 126 were originally allocated.
[image]

CUAVcaijie, 2021-05-07:
CUAV X7

Here is to allocate memory to print
[image]

Here, there is packet loss when the queue is not full, and the difference between the incoming and outgoing queues is getting bigger and bigger.
[image]

CUAVcaijie, 2021-05-07:
CUAV V5nano

Here is to allocate memory to print
[image]

The number of enqueues and the number of exits remain the same
[image]

CUAVcaijie, 2021-05-07:
@pavel-kirienko 
After the CUAV v5Nano flight controller increases the allocated memory, the packet loss problem can be solved.

In addition to insufficient memory allocation, the CUAV X7 flight controller has other problems that cause packet loss. Not found yet

CUAVcaijie, 2021-05-07:
Any good suggestions?

CUAVcaijie, 2021-05-07:
Hello, I found that the entrance sent by CUAV X7 is different from the entrance sent by CUAV v5Nano. Will it affect the subsequent data loss?

CUAV V5 sending entrance:
src/drivers/uavcan/libuavcan/libuavcan/include/uavcan/node/publisher.hpp
int broadcast(const DataType& message)

CUAV X7 send entrance:
src/drivers/uavcan/uavcan_main.cpp
void UavcanNode::node_spin_once()

CUAVcaijie, 2021-05-07:
@pavel-kirienko 
Sorry, it turns out that there is a problem with the baud rate setting of my CAN device. Configured to 87.5, it can be used normally. No packet loss

CUAVcaijie, 2021-05-07:
@davids5 @pavel-kirienko 
Thank you

closed 2021-05-07 by CUAVcaijie

davids5, 2021-05-07:
@CUAVcaijie - Would you please make a PR with any relevant changes needed.