PX4/PX4-Autopilot #17509: ADIS16448 IMU Driver Unreliable with ADIS16448AMLZ

Jaeyoung-Lim, 2021-05-02:
**Describe the bug**
Starting the ADIS16448AMLZ driver on a external SPI port seems to be unreliable on latest master

There are two issues using this IMU (logs provided below)
- Unreliable start of the driver: It takes a few tries to get the driver to detect the imu
- Low rate of IMU messages: After the driver has been started, the rate of the imu message seems to be quite low. 

**Expected behavior**
Reliably start the imu and receive high rate imu messages

**Log Files and Screenshots**
- Unreliable start of the driver: It takes a few tries to get the driver to detect the imu
```
nsh> adis16448 start -S
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
nsh> adis16448 start -S
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
nsh> adis16448 start -S
adis16448 #0 on SPI bus 5 (devid=0x0) (external, equal to '-b 1')
```

- Low rate of IMU messages: After the driver has been started, the rate of the imu message seems to be quite low. 
```
update: 1s, num topics: 86
TOPIC NAME      INST #SUB RATE #Q SIZE
sensor_accel       0    5  399  8   48 
sensor_accel       1    4  200  8   48 
sensor_accel       2    3    0  8   48 
sensor_accel_fifo  0    1  200  1  224 
sensor_accel_fifo  1    0  399  1  224 
```

**Drone (please complete the following information):**
- Tested on a pixhawk4

**Additional context**
- This seems to be related to https://github.com/PX4/PX4-Autopilot/issues/16839

dagar, 2021-05-02:
It might need more time to startup after the power is cycled.

I would add a loop in the probe that attempts maybe 5 times or so before returning an error. 
https://github.com/PX4/PX4-Autopilot/blob/fa9fdce6e6f26a1fbf5970aeb5eb7be3cf4941d3/src/drivers/imu/analog_devices/adis16448/ADIS16448.cpp#L162-L178

The other pattern I want to get in place eventually is an option that starts these sensors and keeps running even if they aren't found initially so that they could be plugged in later in flight.

Jaeyoung-Lim, 2021-05-02:
One thing I noticed is that the configure step is returning false: https://github.com/PX4/PX4-Autopilot/blob/fa9fdce6e6f26a1fbf5970aeb5eb7be3cf4941d3/src/drivers/imu/analog_devices/adis16448/ADIS16448.cpp#L436

Wondering if the register maps are different between the two versions?

Jaeyoung-Lim, 2021-05-02:
Register check failing seems to be the blame: https://github.com/PX4/PX4-Autopilot/blob/fa9fdce6e6f26a1fbf5970aeb5eb7be3cf4941d3/src/drivers/imu/analog_devices/adis16448/ADIS16448.cpp#L539

dagar, 2021-05-02:
> Wondering if the register maps are different between the two versions?

Seems possible given the CRC difference. Let's gather up the exact datasheets and compare.

Jaeyoung-Lim, 2021-05-02:
If it helps here are the datasheets:
ADIS16448AMLZ: [datasheet](https://drive.google.com/file/d/1li3ocPU1JNSfhOjI4UF8xM1UDavRQhmn/view?usp=sharing)
ADIS16448BMLZ: [datasheet](https://drive.google.com/file/d/12sohyHdLVzy3463K6hA7qd_0b_bJyp3O/view?usp=sharing)

dagar, 2021-05-18:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/17510.

closed 2021-05-18 by dagar