PX4/PX4-Autopilot #17532: UAVCAN bootloader file naming and Jenkins artifacts

PetervdPerk-NXP, 2021-05-05:
**Describe the bug**
Is there a easy way to get a static URL to the latest binary that works with the UAVCANv0 bootloader.

For example building the UCANS32K146 firmware yields 2 identical files.
`build/nxp_ucans32k146_default/34-0.1.ac6e7a1c.uavcan.bin`
`build/nxp_ucans32k146_default/deploy/34.bin`

Only the 34-0.1.ac6e7a1c.uavcan.bin will be available in the Jenkins CI artifact see but the filename changes with every commit.
http://ci.px4.io/job/PX4_misc/job/Firmware-compile/job/master/lastSuccessfulBuild/artifact/build/nxp_ucans32k146_default/

A solution could be including the build/*/deploy/*.bin regex in the Jenkins CI command but then we would duplicate files in the Jenkins CI artifact.
https://github.com/PX4/PX4-Autopilot/blob/6e2343a485a51308706757b6d2a2344dcd522571/.ci/Jenkinsfile-compile#L185

dagar, 2021-05-05:
I'm trying to transition all the remaining builds to github actions. Here's the UAVCAN firmware builds. https://github.com/PX4/PX4-Autopilot/actions/workflows/compile_nuttx_cannode.yml

One of the next goals is to work with the GCS to deploy the firmware to the flight controller (on request) so that the flight controller can automatically update the notes.

All that is to say, yes I'm in favor of finding a solution here that let's us trivially grab "latest".

davids5, 2021-05-05:
> Is there a easy way to get a static URL to the latest binary that works with the UAVCANv0 bootloader?
>
> For example building the UCANS32K146 firmware yields 2 identical files. [...]

The deploy sub directory was created with the copy just for deploying generically to say a PX4FMUV5 build's ROMFS.

So that would work. But let @dagar sort out the build, and then we can see if deploy/* is still workable.