PX4/PX4-Autopilot #17536: GIMBAL_DEVICE_ATTITUDE_STATUS parsing result in BAD_DATA type

sei-project, 2021-05-06:
Hello, 
I'm using SITL simulation.
I'm using Gimbal Protocol v2 with typhoon model.
I need to receive GIMBAL_DEVICE_ATTITUDE_STATUS in ROS.
I don't see mavros equivalent message yet.
So I want to listen mavlink messages directly using pymavlink utils.
All messages coming correctly but GIMBAL_DEVICE_ATTITUDE_STATUS (and other new messages) messages are parsing as BAD_DATA.
I see in mavlink_px4.py and in mavlink_map array this message doesn't exists.
I see mavlink_px4.py is generated from common.xml where this message definition exists.
How is possible? 
How mavlink_px4.py can be regenerated?
Thanks.

julianoes, 2021-05-07:
Hi, so taking a step back, what are you trying to do?

Note that pymavlink is using the ardupilotmega.xml MAVLink dialect which probably does not contain the new GIMBAL_MANAGER and GIMBAL_DEVICE messages as they are still marked WIP.

sei-project, 2021-05-07:
Hi julianoes, what I'm trying to do is to receive from my ROS node that handles the navigation of the UAV the gimbal orientation and position messages measured by the gimbal itself to locate images. The gimbal is operating with the Gimbal Protocol V2.
Pymavlink is something I've read around and I set the environment variables MAVLINK20 = 1 and MAVLINK_DIALECT = common, so I think, pymavlink use common.xml dialect.
What you say, that these messages are WIP I had noticed (also in common.xml) but I don't know what effect they have in deserializing the messages.
Does that mean that as long as they are WIP it is not possible to deserialize them?
And are you aware of what the release date may be?
I think that as long as there is no support for these messages I can deserialize them myself since I receive them anyway.
Is there an alternative to using pymavlink?
Thank you.
 

Translated with www.DeepL.com/Translator (free version)

julianoes, 2021-05-08:
Aha, I understand. You could use MAVROS but you probably would have to add the message there. Or MAVSDK where it's already implemented. Or any other direct mavlink implementation.

sei-project, 2021-05-10:
If possible I will try to use the gimbal with protocol 1.
I will give MAVSDK a try in case it is not possible to do so.
Thanks for the info.

julianoes, 2021-05-10:
MAVSDK supports the mavlink gimbal v1 and v2 protocols.