PX4/PX4-Autopilot #17546: Magnetometer Calibration Fails (unable to fit mag 0)

Pskoog3, 2021-05-07:
I am working on a new driver for Honeywell's 3-axis magnetometer HMC6343 and it seems to work fine with PX4. However, the magnetometer calibration fails in QGroundControl... the calibration process goes well but at the end says:
Retry calibration (unable to fit mag 0)
[cal] calibration failed: mag

I have tested it on a Raspberry Pi 3 board with SparkFun's HMC6343 breakout board. The output of the magnetometer is quite stable and other than the calibration issue the sensor behaves well. Is the issue related to scale factor or something else?
The log is as under.

[cal] calibration started: 2 mag
[cal] pending: back front left right up down
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] down orientation detected
[cal] down orientation detected
[cal] Rotate vehicle
[cal] down side done, rotate to a different side
[cal] down side done, rotate to a different side
[cal] down side done, rotate to a different side
[cal] pending: back front left right up
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] up orientation detected
[cal] up orientation detected
[cal] Rotate vehicle
[cal] up side done, rotate to a different side
[cal] up side done, rotate to a different side
[cal] up side done, rotate to a different side
[cal] pending: back front left right
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] front orientation detected
[cal] front orientation detected
[cal] Rotate vehicle
[cal] front side done, rotate to a different side
[cal] front side done, rotate to a different side
[cal] front side done, rotate to a different side
[cal] pending: back left right
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] back orientation detected
[cal] back orientation detected
[cal] Rotate vehicle
[cal] back side done, rotate to a different side
[cal] back side done, rotate to a different side
[cal] back side done, rotate to a different side
[cal] pending: left right
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] left orientation detected
[cal] left orientation detected
[cal] Rotate vehicle
[cal] left side done, rotate to a different side
[cal] left side done, rotate to a different side
[cal] left side done, rotate to a different side
[cal] pending: right
[cal] hold vehicle still on a pending side
[cal] detected rest position, hold still...
[cal] right orientation detected
[cal] right orientation detected
[cal] Rotate vehicle
[cal] right side done, rotate to a different side
[cal] right side done, rotate to a different side
[cal] right side done, rotate to a different side
Retry calibration (unable to fit mag 0)
[cal] calibration failed: mag

dagar, 2021-05-07:
Can you share a log from boot (SDLOG_MODE)? There's detailed logging from the mag cal in `mag_worker_data`. https://github.com/PX4/PX4-Autopilot/blob/master/msg/mag_worker_data.msg

SalimTerryLi, 2021-05-07:
BTW, I have done a test flight several days ago and found something similar.

First time I uploaded a fresh new PX4 to RPi and try to do sensor calibration, all passed but not mag. I did not notice what console say but on QGC side it passed through all six direction and done the calibration with red mag icon. But soon I found that due to different hardware versions I have made, the mag driver didn't launch at all! After modified the startup script it finally works without any problem.

Seems that mag calibration goes even without any valid data source? And there is no related error report about that.

dagar, 2021-05-07:
> Seems that mag calibration goes even without any valid data source? And there is no related error report about that.

That's certainly not the intention, but it's also not explicitly tested as the majority of boards have built in mags.

dagar, 2021-05-07:
@SalimTerryLi PR for that particular situation. https://github.com/PX4/PX4-Autopilot/pull/17549

@Pskoog3 if possible having another known good mag in the same log will help with debugging. It could be things like the scale factor in the new driver being wrong.

dagar, 2021-05-07:
> Here is the log...
> https://review.px4.io/plot_app?log=446d9d85-7268-422a-8312-ea146fb0da3a

Either the scaling in the driver is off or its next to a strong magnetic field.

### Magnetometer magnitude
[image]

Earth's magnetic field ranges from 0.25 to 0.65 gauss, but your magnetometer is reading over 2 Gauss at times.

### Magnetometer per Axis
[image]

Let's take a closer look at that driver.

Pskoog3, 2021-05-11:
After hard iron calibration of HMC6343 the raw magnetic field looks like this...
[image]

and failed to calibrate...
ERROR [commander] Retry calibration (unable to fit mag 0)
WARN  [commander] [cal] calibration failed: mag

BUT, the sensor seems to work fine in QGroundControl compass instrument. It points to the North and when the platform is rotated 90 degrees, the compass indeed shows 90 degrees rotation. The dynamic response and stability of compass is excellent, but couldn't get it calibrated and the raw magnetic field strength seems to be zero! Why?

Here is the fresh log...
https://review.px4.io/plot_app?log=364a3bc6-d851-4976-b646-a15c95ae6654

LorenzMeier, 2021-05-11:
Please read the response from @dagar. You have the scaling wrong for your sensor. You will need to fix the driver to have the correct units.

esamet, 2022-10-15:
ı got that error and  ı just try a little far away from my laptop and problem solved, ı think magnetic field source from issue

xiangyu541, 2024-07-20:
> ı 遇到了那个错误，ı 只是尝试离我的笔记本电脑有点远并解决了问题，ı 认为问题的磁场源
Thanks, because of your suggestion, I solved this problem.