PX4/PX4-Autopilot #17558: X2.1-777 1.12 bug in UART RX and TX DMA causes most of the remaining errors

taileron, 2021-05-10:
**Describe the bug**
GPS not recognized
IO auto update not working
all other consequences of missing IO
telemetry unstable
log download by wifi not working
**To Reproduce**
Steps to reproduce the behavior:
after disabling all rx and tx dma in defconfig
the above functions can be successfully used individually 

**Expected behavior**
A clear and concise description of what you expected to happen.
rx and tx DMA does work properly and all serial stuff works simultaneously
**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

taileron, 2021-05-17:
If you go back in time: also in all V1.11 with NuttX 8.2.0 both F777 rx and tx dma are also affected only as soon as both are off in defconfig, the IO is automatically flashed and Ublox gps modules are detected.
ver all
HW arch: MRO_X21_777
FW git-hash: 7e5e8259f3bc644185569cb2292fd5ebd6bb532f
FW version: 1.11.0 0 (17498112)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 863ed53e3fb1cebbb0832ec28fbc11e16566e89e
Build datetime: May 17 2021 13:01:42
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0002000000003237323134385108002d0022
MCU: STM32F76xxx, rev. Z
nsh> px4io status
WARN  [px4io] loaded
protocol 4 hardware 2 bootloader 3 buffer 64B crc 0x5f575f9d
8 controls 8 actuators 18 R/C inputs 2 analog inputs 0 relays
 px4io_status_s
    timestamp: 76483917  (0.829196 seconds ago)
    voltage_v: 5.0440
    rssi_v: 1.6860
    free_memory_bytes: 824
    actuators: [-10000, -10000, -10000, 9808, 4838, -5772, 0, 0]
    servos: [900, 900, 900, 2000, 1985, 1030, 1500, 1500]
    raw_inputs: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    status_outputs_armed: True
    status_override: False
    status_rc_ok: False
    status_rc_ppm: False
    status_rc_dsm: False
    status_rc_sbus: False
    status_fmu_ok: True
    status_raw_pwm: False
    status_mixer_ok: True
    status_arm_sync: True
    status_init_ok: True
    status_failsafe: False
    status_safety_off: True
    status_fmu_initialized: True
    status_rc_st24: False
    status_rc_sumd: False
    alarm_vbatt_low: False
    alarm_temperature: False
    alarm_servo_current: False
    alarm_acc_current: False
    alarm_fmu_lost: True
    alarm_rc_lost: True
    alarm_pwm_error: False
    alarm_vservo_fault: False
    arming_io_arm_ok: True
    arming_fmu_armed: False
    arming_fmu_prearmed: True
    arming_manual_override_ok: False
    arming_failsafe_custom: True
    arming_inair_restart_ok: True
    arming_always_pwm_enable: True
    arming_rc_handling_disabled: False
    arming_lockdown: False
    arming_force_failsafe: False
    arming_termination_failsafe: False
    arming_override_immediate: True

reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
0 raw R/C inputs
R/C flags: 0x0000
mapped R/C inputs 0x0000
ADC inputs 2081 2066
features 0x0008 RSSI_ADC
rates 0x000f default 50 alt 400 sbus 72
debuglevel 0
controls 0: 761 412 8 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 899 center 1500 max 2101 deadzone 10 assigned 0 options 0x0001 ENABLED
input 1 min 900 center 1500 max 2098 deadzone 10 assigned 1 options 0x0001 ENABLED
input 2 min 1056 center 1056 max 2035 deadzone 10 assigned 3 options 0x0001 ENABLED
input 3 min 899 center 1501 max 2101 deadzone 10 assigned 2 options 0x0003 ENABLED REVERSED
input 4 min 1099 center 1514 max 1901 deadzone 10 assigned 255 options 0x0000
input 5 min 1059 center 1551 max 2043 deadzone 10 assigned 255 options 0x0000
input 6 min 899 center 1410 max 1921 deadzone 10 assigned 5 options 0x0003 ENABLED REVERSED
input 7 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 8 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 9 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 10 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 11 min 1000 center 1500 max 2000 deadzone 0 assigned 7 options 0x0001 ENABLED
input 12 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 13 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 14 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 15 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 900 900 900 2000 1985 1030 1500 1500
disarmed values 900 900 900 2000 1985 1030 1500 1500

taileron, 2021-08-17:
@dagar Unfortunately, I haven't really been able to track down the problem of the DMA of all UARTs of the controller.
I have tried it with other timers without DCACHE etc. As soon as rxdma is activated, gps, wifi telemetry ftp and flashing the IO no longer works. 
Do you have an additional hint where this could be hidden in the board configuration? The CPU is exactly the same only on ZERO F7 and Cube Yellow, and there seem to be problems as well. It must be something in the cache allocation or in the DMA polling, as it has exactly the same effect on all UARTS. F777 still has small differences to the F7s that are used in FMU V5 and above boards.

taileron, 2021-08-18:
How can be done a kind of loop test to send something to the UART´s tx and get it back through it´s rx or the rx of another uart. The type or what gets lost in the process could provide more insight into fault finding.

taileron, 2021-08-19:
unfortunately, the result of the serial_test is not as nice and black and white as I had imagined. The errors shown below occur when rxdma is active, but errors also occur without rxdma because the app itself brings the CPU load above 100% much earlier:
Error, count: 114826, expected 77, got 41 Error, count: 114845, expected 54, got 55 Error, count: 115046, expected 1e, got 1f Error, count: 115145, expected 82, got 83 Error, count: 115229, expected d7, got d8 Error, count: 115271, expected 02, got 03 Error, count: 115309, expected 29, got 2a Error, count: 115429, expected a2, got a3 Error, count: 115590, expected 44, got 47 Error, count: 115652, expected 85, got 98 Error, count: 115692, expected c0, got c1 Error, count: 116070, expected 3b, got 3c Error, count: 116259, expected f9, got bf Error, count: 116327, expected 03, got 04 Error, count: 116748, expected a9, got aa Error, count: 116866, expected 20, got ea Error, count: 117209, expected 41, got 42 Error, count: 117473, expected 4a, got ae Error, count: 117615, expected 3c, got 3d

taileron, 2021-12-01:
solved ... with the more recent AP bootloader now all rx and tx dma work as desired

taileron, 2022-01-11:
sorry I have to reopen this issue because I made a mistake, in defconfig rxdma=Y I entered accidentally with capital letters therefore interpreted as off, my mistake therefore I thought it was fixed.
It is still the only issue with this board

dagar, 2022-01-11:
Have you reviewed the DMA map? https://github.com/PX4/PX4-Autopilot/blob/master/boards/mro/x21-777/nuttx-config/include/board_dma_map.h

Let me get my hands on one of these boards and we can follow up here or on slack.

taileron, 2022-01-11:
was not able to find any faulty DMA assignments and I tried some alternatives the strange thing is that it affects all serial ports in the same way. Is it possible that if only one port´s DMA map does not fit, then all ports no longer function properly? The simplest test was a GPS module no longer works on any port as soon as DMA is activated there.
sorry, currently im traveling with no slack access and the x2.1-777 board is not part of my luggage so further hw test I can only do from Friday on.