PX4/PX4-Autopilot #17569: Wrong RPM in ESC_STATUS

jpowen898, 2021-05-12:
@mcsauder was also involved in helping me debug and diagnose this issue.
The uorb esc_status message appears to be outputting the wrong rpms. Through the FMU debug port I have been querying the ESC reported rpms. The following data was collected while the aircraft was armed and in acro mode without props. The RPMs should all be about the same but they have a 4000rpm difference between motors. However, temperature and voltage in this message are correct implying that the ESC data is being parsed correctly but that the rpms are getting corrupted somewhere downstream in the pixhawk code.
```
nsh> Listener esc_status
TOPIC: esc_status
 esc_status_s
	timestamp: 219508377  (0.021385 seconds ago)
	counter: 11337
	esc_count: 4
	esc_connectiontype: 5
	esc_online_flags: 15 (0b0000'1111)
	esc_armed_flags: 15 (0b0000'1111)
	px4/esc_report[8] esc[0] esc_report_s
	timestamp: 219508376  (0.022104 seconds ago)
	esc_errorcount: 0
	esc_rpm: 1000
	esc_voltage: 23.4700
	esc_current: 0.0400
	esc_temperature: 49
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[1] esc_report_s
	timestamp: 219498309  (0.033913 seconds ago)
	esc_errorcount: 0
	esc_rpm: 5533
	esc_voltage: 23.4300
	esc_current: 0.0400
	esc_temperature: 59
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[2] esc_report_s
	timestamp: 219502110  (0.030654 seconds ago)
	esc_errorcount: 0
	esc_rpm: 1466
	esc_voltage: 23.6600
	esc_current: 0.0400
	esc_temperature: 48
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[3] esc_report_s
	timestamp: 219505826  (0.027732 seconds ago)
	esc_errorcount: 0
	esc_rpm: 4733
	esc_voltage: 23.4300
	esc_current: 0.0400
	esc_temperature: 60
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[4] esc_report_s

	esc_errorcount: 0
	esc_rpm: 0
	esc_voltage: 0.0000
	esc_current: 0.0000
	esc_temperature: 0
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[5] esc_report_s

	esc_errorcount: 0
	esc_rpm: 0
	esc_voltage: 0.0000
	esc_current: 0.0000
	esc_temperature: 0
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[6] esc_report_s

	esc_errorcount: 0
	esc_rpm: 0
	esc_voltage: 0.0000
	esc_current: 0.0000
	esc_temperature: 0
	esc_address: 0
	esc_state: 0
	failures: 0
	px4/esc_report[8] esc[7] esc_report_s

	esc_errorcount: 0
	esc_rpm: 0
	esc_voltage: 0.0000
	esc_current: 0.0000
	esc_temperature: 0
	esc_address: 0
	esc_state: 0
```

  
I also tried gathering the esc info through the dshot command. This command returned saying that no telemetry data was received. However, I know it is being received because the esc_status messages are being sent with the correct voltage and temperature.
```
nsh> dshot esc_info -m 1
ERROR [dshot] No data received. If telemetry is setup correctly, try again
nsh> 
nsh> dshot esc_info -m 2
ERROR [dshot] No data received. If telemetry is setup correctly, try again
nsh> 
nsh> dshot esc_info -m 3
ERROR [dshot] No data received. If telemetry is setup correctly, try again
nsh> 
nsh> dshot esc_info -m 4
ERROR [dshot] No data received. If telemetry is setup correctly, try again
```

  
In order to double check that the ESC's were actually reporting telemetry I plugged into the telemetry line into a serial device and wrote a script to parse the incoming data. Again I verified that reported voltage and temperature were correct. Bellow is a collection of reported rpm's. I tried to associate them, but every few cycles one of the esc's fails to report and so the data association shifts to the left. Again this data was gathered while in armed in acro mode without props.
```
[m1, m2, m3, m4]
[84, 204, 138, 150]
[84, 204, 138, 150]
[84, 204, 138, 144]
[84, 204, 138, 144]
[84, 204, 138, 144]
[84, 204, 144, 144]
[84, 210, 144, 90]
[210, 144, 150, 90]
[210, 144, 150, 204]
[138, 150, 90, 204]
[138, 150, 84, 204]
[138, 150, 84, 204]
[150, 84, 204, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[144, 84, 144, 144]
[84, 210, 144, 144]
[210, 144, 150, 84]
[210, 144, 84, 204]
[138, 150, 84, 144]
[84, 204, 138, 144]
[210, 138, 144, 84]
[210, 138, 144, 84]
[210, 138, 150, 84]
[210, 138, 150, 84]
[210, 144, 150, 84]
[144, 150, 84, 204]
[144, 84, 204, 144]
[144, 84, 204, 138]
[84, 210, 138, 144]
[84, 210, 138, 150]
[84, 210, 138, 150]
[84, 210, 138, 150]
[84, 210, 138, 150]
[84, 210, 138, 90]
[210, 138, 150, 90]
[210, 138, 150, 90]
[210, 138, 150, 90]
[210, 144, 150, 90]
[204, 144, 150, 90]
[144, 150, 84, 204]
[144, 144, 84, 204]
[144, 144, 84, 204]
[144, 144, 84, 204]
[138, 144, 84, 210]
[138, 144, 210, 138]
[150, 210, 138, 150]
[84, 204, 138, 150]
[84, 204, 84, 204]
[150, 84, 210, 144]
[150, 84, 210, 144]
[84, 210, 144, 144]
[84, 210, 138, 144]
[84, 138, 144, 84]
[210, 138, 150, 210]
[138, 150, 84, 204]
[138, 150, 84, 204]
[138, 150, 204, 138]
[150, 84, 204, 138]
[150, 84, 204, 138]
[150, 216, 138, 150]
```

After verifying that the ESC's were reporting correctly I ran another test and gathered the rpms from the listener over time. This data was gathered by running the "listener esc_status" command and then parsing the output to get the 4 rpms I am looking for. Again this data was gathered while in armed in acro mode without props.
```
[m1, m2, m3, m4]
[1066, 200, 1466, 0]
[0, 4866, 1533, 4066]
[1000, 4866, 0, 4133]
[1000, 4800, 1533, 4066]
[1000, 4866, 1533, 4133]
[933, 4866, 1466, 4133]
[1000, 4933, 1466, 4066]
[1000, 4933, 1466, 4133]
[1000, 4933, 1533, 4133]
[1000, 4933, 1466, 4133]
[1000, 4933, 1466, 4133]
[1000, 4933, 1533, 4133]
[1000, 5000, 1533, 0]
[1000, 4933, 1466, 4200]
[1000, 5000, 1466, 4200]
[1000, 5000, 1466, 4200]
[1000, 5000, 1466, 4266]
[1000, 0, 1533, 4200]
[1000, 0, 1466, 4266]
[1000, 5066, 1466, 4200]
[1000, 5066, 1466, 4200]
[1000, 5066, 0, 4266]
[1000, 0, 1466, 4266]
[1000, 5000, 1466, 4200]
[1000, 5000, 1466, 4266]
[1000, 5000, 1466, 4266]
```

This is the rpms plotted from the log file. As you can see they correspond well with the rpms gathered from the listener.
[image]

Here are the Dshot and Motor pole parameters I am using.
```
DSHOT_CONFIG = DShot600
DSHOT_MIN = 0.05%
DSHOT_TEL_CFG = TELEM/SERIAL4
MOT_POLE_COUNT = 18
```

The code I wrote to convert the raw serial data from the ESCs was basically copied from the decodeByte() function in telemetry.cpp so the bug seems like it must be downstream from there.

bkueng, 2021-05-17:
Which ESC's are those?
Did you compare against the commanded levels?
If you look at the code, there's only a constant scaling between the telemetry parsing and `esc_status.esc_rpm`.

jpowen898, 2021-05-17:
I am using the APD 80F3[x] ESCs
https://www.racedayquads.com/products/advanced-power-drives-80f3x-8s-80a-esc-apd-x-class-esc?currency=USD&variant=31301323391089&gclid=CjwKCAjwqIiFBhAHEiwANg9szv6MDKEHwSW9utObASSpRj82h6rvjrbYhcMkdOTKKemgUbZmjOZi2hoCm8oQAvD_BwE

I did not compare them against the commanded levels. I just left it on idle and compared the rpms outputed from the listener to what was measured from a tachometer.

Yes I realized that it is only being scaled before it is passed into the ESC_STATUS message, but it is still coming out wrong on the other side. Could it be a type conversion or something? EscData.erpm is a int16_t which is being scaled and passed into esc_status.esc_rpm which is a int32_t. I know this shouldn't cause an issue, but is it getting converted again somewhere that would cause it to lose precision or something? Also if you look at the plot from the log the reported esc_rpms drop to zero occasionally even though the motors were being held at a relatively constant rpm, indicating that there might be some kind of connection issue causing it to lose the data.

bkueng, 2021-05-18:
Maybe I misinterpreted your data, but doesn't your raw data also show big differences between the motors (`[84, 204, 138, 150]`)?

> I just left it on idle and compared the rpms outputed from the listener to what was measured from a tachometer.

Do you have a comparison plot for that?

> Also if you look at the plot from the log the reported esc_rpms drop to zero occasionally even though the motors were being held at a relatively constant rpm, indicating that there might be some kind of connection issue causing it to lose the data.

Yes that is a known issue for the APD ESC's: https://github.com/PX4/PX4-Autopilot/issues/16735

jpowen898, 2021-05-21:
Yes there are relatively large differences in the motor rpms, but they are not as large as the pixhawk reports them to be. This is the code the pixhawk uses to scale the rpm. My pole count param is set to 18.
`esc_status.esc[motor_index].esc_rpm = ((int)data.erpm * 100) / (_param_mot_pole_count.get() / 2);`
If we apply this to the `[84, 204, 138, 150]` data set we get `[933, 2266, 1533, 1666]` which is significantly different from what the pixhawk is actually reporting `[1000, 4933, 1533, 4133]`

Also I do not have a plot of the raw data this is becasue the esc's occasionally miss a telemetry report when commanded making it hard to associate the raw data with individual motors. As you can see in the data below the motor rpms are staying constant but the reported motor number is changing this happens when the ESC's fail to report.
```
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[150, 84, 210, 138]
[144, 84, 144, 144]
[84, 210, 144, 144]
[210, 144, 150, 84]
[210, 144, 84, 204]
[138, 150, 84, 144]
[84, 204, 138, 144]
[210, 138, 144, 84]
[210, 138, 144, 84]
[210, 138, 150, 84]
[210, 138, 150, 84]
[210, 144, 150, 84]
```

closed 2025-12-01 by github-actions[bot]