PX4/PX4-Autopilot #17573: compile problems about PX4 on ubuntu

bakawtjun, 2021-05-14:
**I had updated and install what is needed for compiling PX4. And I followed the suggestion to complete the environment. But it's still not succeed. Here is what happed below.I hope you can help me to solve it. Thanks!**

ninja: Entering directory `/home/wtj/src/Firmware/build/nuttx_px4fmu-v2_default'
[4/617] Building px4io-v2
ninja: Entering directory `/home/wtj/src/Firmware/build/nuttx_px4io-v2_default'
[4/186] Generating mixer_multirotor_6dof.generated.h
FAILED: cd /home/wtj/src/Firmware/build/nuttx_px4io-v2_default/src/lib/mixer && /usr/bin/python /home/wtj/src/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_t.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_h.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/tri_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
('python import error: ', ImportError('No module named numpy',))

Required python packages not installed.

On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

[4/186] Generating mixer_multirotor_normalized.generated.h
FAILED: cd /home/wtj/src/Firmware/build/nuttx_px4io-v2_default/src/lib/mixer && /usr/bin/python /home/wtj/src/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --normalize -f /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_t.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_h.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/tri_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
('python import error: ', ImportError('No module named numpy',))

Required python packages not installed.

On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

[4/186] Generating mixer_multirotor.generated.h
FAILED: cd /home/wtj/src/Firmware/build/nuttx_px4io-v2_default/src/lib/mixer && /usr/bin/python /home/wtj/src/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py -f /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_t.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_h.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/tri_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
('python import error: ', ImportError('No module named numpy',))

Required python packages not installed.

On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'nuttx_px4io-v2_default' failed
make[1]: *** [nuttx_px4io-v2_default] Error 1
[6/617] Generating mixer_multirotor_6dof.generated.h
FAILED: cd /home/wtj/src/Firmware/build/nuttx_px4fmu-v2_default/src/lib/mixer && /usr/bin/python /home/wtj/src/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_t.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/octa_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_h.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/quad_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/tri_y.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
('python import error: ', ImportError('No module named numpy',))

Required python packages not installed.

On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

[6/617] Generating mixer_multirotor.generated.h
FAILED: cd /home/wtj/src/Firmware/build/nuttx_px4fmu-v2_default/src/lib/mixer && /usr/bin/python /home/wtj/src/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py -f /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_t.toml /home/wtj/src/Firmware/src/lib/mixer/geometries/hex_x.toml /home/wtj/src/Firmware/ [... cut ...]
On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

[6/617] Copying NuttX/apps to /home/wt...re/build/nuttx_px4fmu-v2_default/NuttX
FAILED: cd /home/wtj/src/Firmware && make --no-print-directory nuttx_px4io-v2_default
[6/617] Copying NuttX/nuttx to /home/w...re/build/nuttx_px4fmu-v2_default/NuttX
ninja: build stopped: subcommand failed.
Makefile:154: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

bakawtjun, 2021-05-14:
I tried to install the numpy toml. But it still gives me this wrong again.

antonio-sc66, 2021-05-15:
@bakawtjun It looks like a Python2 vs Python3 problem

julianoes, 2021-05-17:
@bakawtjun this looks like an old version of PX4.

bakawtjun, 2021-05-26:
> @bakawtjun this looks like an old version of PX4.

Thanks! I had succeeded last weekend by using PX4 ver1.8.2 with gcc compiler ver2017q4 on Ubuntu 18.04. It seemed that the reason why I failed were incompatibility problems. I shouldn't use the latest version  to test.

julianoes, 2021-05-27:
@bakawtjun so it's fixed?