PX4/PX4-Autopilot #17574: MacOS make error: No package 'bullet2.82' found

Dives44, 2021-05-14:
## Question
How to run to successfully run `make px4_sitl gazebo` without the error: `--   No package 'bullet' found`

## Context
I'm trying to run PX4 stil on gazebo on macOS

###Details
I have followed these [instructions](https://docs.px4.io/master/en/dev_setup/dev_env_mac.html) but at this [step](https://docs.px4.io/master/en/dev_setup/dev_env_mac.html#gazebo-simulation) after running `make px4_sitl gazebo` I get the follow gin error:
```
[29/595] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
CMake Warning (dev) at /opt/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /opt/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'bullet>=2.82'
--   No package 'bullet' found
-- Checking for module 'bullet2.82>=2.82'
--   No package 'bullet2.82' found
CMake Error at /usr/local/lib/cmake/gazebo/gazebo-config.cmake:134 (message):
  Error: Bullet > 2.82 not found, please install libbullet2.82-dev.
Call Stack (most recent call first):
  CMakeLists.txt:48 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/julien/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/Users/julien/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[35/595] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /Users/julien/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /opt/local/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /Users/julien/PX4-Autopilot/Tools/sitl_gazebo && /opt/local/bin/cmake -E touch /Users/julien/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[37/595] Linking CXX static library msg/libuorb_msgs.a
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

### What I did
I tried to find a way to install libbullet2.82-dev but the only way I found was to run `sudo apt-get install libbullet2.82-dev` which I cannot do as I'm on MacOS which does not support apt-get

## System
MBP Intel on MacOS BigSur 11.2.3
xcode-select version 2384

Can someone help me plz 🙏

julianoes, 2021-05-17:
Are all brew dependencies up to date?

mwshafer, 2022-05-20:
@julianoes I am having this same issue. When I run `brew install bullet` I get:
```
Running `brew update --preinstall`...
Warning: bullet 3.24 is already installed.
The currently linked version is: 3.17
```