PX4/PX4-Autopilot #17576: Airspeed sensor calibration only checks the piping polarity

roman-dvorak, 2021-05-14:
**Describe the bug**
I tried to use our [TFSLOT](https://github.com/ThunderFly-aerospace/TFSLOT01) sensor with PX4  at our [TF-G2/4s](https://github.com/ThunderFly-aerospace/TF-G2/tree/4s) and we encountered several problems. The [TFASPDIMU](https://github.com/ThunderFly-aerospace/TFASPDIMU01) sensor board has a fixed SDP33 diff pressure sensor. And a positive speed value causes negative pressure in the pressure sensor. And because of that, I was not able to do a normal calibration. 

After examining the callibration code, I found that the calibration only detects the correct connection of the sensor. In case of negative pressure it will end. 

Subsequently, I compared Airspeed via TFSLOT and from GPS. There I obtained this measurements:
[image]

It is necessary to add a constant into the calculation of airspeed, which will depend on the TFSLOT profile. 

**Expected behavior**
Expected behavior is that the autopilot remembers the measured polarity and this will multiply the pressure obtained form pressure sensor. For example, in the form of a parameter. 

**Additional context**
1. Calibration i have solved with blowing from behind of TFSLOT (TFSLOT it allows against pittot tube). The autopilot thinks it's OK. 
2. And for calculation of airspeed I have created this modifications of FW: https://github.com/ThunderFly-aerospace/PX4Firmware/commit/c6d93979bf7ab55e949f5726aaf129ea280d32a9

I created a new profile for calculation of airspeed from TFSLOT, which is turned on by a new value in the parameter `CAL_AIR_CMODEL` This profile is almost same as profile for pitot tube. But diff pressure is negated. 

**Questions**
* Are these adjustments acceptable to PX4? (New profile for TFSLOT, custom calculation of airspeed)
* Does it have any reason that calibration cannot reverse the measunig direction? Or is it just not programmed? I understand that this is a problem with some sensors. But in the case of SDP3x, it shouldn't matter.

dagar, 2021-05-14:
What if we simply expose the differential scale factor calibration and allow it to be negative?

roman-dvorak, 2021-05-14:
Hi, yes, it sounds good. I saw this [parameter](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#ASPD_SCALE). But I wasn't sure if it would affect (break) something else. 

Do you mean that this parameter well be automatically negated during the calibration with a negative pressure? Should I do it?