PX4/PX4-Autopilot #17578: ekf2 replay does not work for SDLOG_PROFILE = 2

priseborough, 2021-05-16:
A log generated with upstream master using a single EKF instance with only ekf2 replay messages is unable to be replayed. Here is an example log.

https://logs.px4.io/plot_app?log=31ebbb10-97a4-469e-8092-2d3ac1ecbfe4

generated using:

SDLOG_MODE, 2
SDLOG_PROFILE, 2

There are two errors that occur when replaying this log:

1) unrecogonised message definition, ig: ERROR [replay] unknown log definition type 81, size 28 (offset 55043). This was fixed by https://github.com/priseborough/PX4-Autopilot/commit/13b36e1a5de6829c76762ecb553a976a58d8f844

2) Repeated baro timeouts, eg: WARN  [ecl/EKF] baro hgt timeout - reset to baro

The ability to replay logs with SDLOG_PROFILE=2 is essential for the debug on F4 boards that do ot have adequate resources to log without  dropouts if other log data is included, eg SDLOG_PROFILE = 3

dagar, 2021-05-16:
The log is missing a large portion of the `sensor_combined` publications. The overall cpu load seems reasonable (~55%), although we only have a snapshot from preflight.

[image]

priseborough, 2021-05-20:
Interesting. I'm also getting the same problem when trying to replay a log generated with SITL gazebo https://logs.px4.io/plot_app?log=b916034a-6b56-432a-98b3-d1864a1ca6c2 with multi ekf disabled.

Abner0907, 2022-05-27:
@dagar  hi, Do you have this issue? 
I have the same error when replay.  This is my sitl log. https://logs.px4.io/plot_app?log=caca5908-34e9-4b0c-b27b-4f2aab13fcfc
when I replay, it shows the error below: 
[image]

And I add the skip default parameter as shows
https://github.com/priseborough/PX4-Autopilot/blob/13b36e1a5de6829c76762ecb553a976a58d8f844/src/modules/replay/Replay.cpp#L250-L254

		case (int)ULogMessageType::INFO: //skip
		case (int)ULogMessageType::INFO_MULTIPLE: //skip
		case (int)ULogMessageType::PARAMETER_DEFAULT: //skip
			file.seekg(message_header.msg_size, ios::cur);
			break;

I still replay failed, the replayed log is empty(always 75.1kb). shows informations as below
[image]

bresch, 2022-08-18:
replay now works again on `main`

closed 2022-08-18 by bresch

hamishwillee, 2022-08-23:
Do we test replay on CI? I ask because periodically we get questions. Usually it is because people are trying it with multi-EKF, but would be good to be able to point to something and say "nope, it works".