PX4/PX4-Autopilot #17586: UAVCAN cannode bootloader stuck in init on MRO f303

dirksavage88, 2021-05-16:
On my local branch baselined to 'pr-mro_cannode' i've been having issues with getting the canbootloader to become operational/maintenance through the uavcan gui tool. 

It seems to be stuck in initialization, but it is assigned a dynamic node ID from the gui tool + zubax babel. Flashed the bootloader to the MRO f303 can node. I can see the node status from the node through the bus monitor, so it seems to have gotten past dynamic node ID allocation part.

 If I flash firmware through jlink commander (SWD), I can see the software version and crc filled in but the same issue occurs with the MRO still stuck in init mode. I believe with just a bootloader, the mode should change to maintenance (w/o firmware), not initialization.

 Of course, my autopilot is having issues with enabling uavcan and updating the fw through that, so the SWD is my only way to flash and the zubax babel is my only means of testing. Updating firmware through the uavcan gui tool just results in the node getting into a bad state and the firmware doesn't take.

Reproduce by compiling the mro_cannode_canbootloader and mro_cannode_default from pr-mro_cannode branch, flashing the bootloader to an MRO can node (via Jlink/stlink) to 0x08000000. Optional to flash firmware via jlink/stlink/Px4 autopilot but not required.

Expected canbootloader shows up with dynamic node ID assigned, but does not come out of init mode. Node status messages are displayed in bus monitor.

[image]

Additional info: 
I made some changes on my fork that mess with the flash base and size, since it looks like @dagar might have copied from the cuav can gps node. The f303cct6 has 256k flash and 40k sram. The F303 node I have is similar to the KitCan node MRO also sells and has almost the same board layout and peripherals.

dagar, 2021-05-16:
At this point it's probably worth getting the initial board support in so we can better coordinate. Can you open a pull request?

dirksavage88, 2021-05-16:
> At this point it's probably worth getting the initial board support in so we can better coordinate. Can you open a pull request?

yes, am I doing a PR to base master or pr-mro_cannode?

dagar, 2021-05-16:
Current master with the goal of getting a baseline working version merged to continue iterating on full support.

closed 2022-10-23 by dirksavage88