PX4/PX4-Autopilot #17610: PX4 in WSL2 + AirSim + QGC

kistlin, 2021-05-19:
**Describe problem solved by the proposed feature**
Use QGC and other tools together with a WSL2 setup.

**Describe your preferred solution**
A similar solution to #16252, #15965.

In `src/modules/mavlink/mavlink_main.cpp` I had to change
```
	/* set default target address, but not for onboard mode (will be set on first received packet) */
	if (!_src_addr_initialized) {
		_src_addr.sin_family = AF_INET;
		inet_aton("<IP of WSL2 interface>", &_src_addr.sin_addr);  //inet_aton("127.0.0.1", &_src_addr.sin_addr); 
	}
```
after this change, I could use QGC + AirSim + PX4 on WSL2

[image]

**Describe possible alternatives**
Is there already a solution passing some arguments or other variables?

**Additional context**
Setup description (what is running where)
Windows 10: QGC, AirSim
WSL2 (Ubuntu 20.04): PX4 based on 5f14ee2a97 (`make px4_sitl_default none_iris`)

Variable `PX4_SIM_HOST_ADDR` is set to the IP of my WSL interface.
Basically following https://microsoft.github.io/AirSim/px4_sitl/.

mrpollo, 2021-05-28:
Hey @kistlin I'm afraid I'm not following on your issue, what is your problem here?

kistlin, 2021-05-29:
To be able to run SITL on another host, the issues  #16252 and #15965 were opened. This is necessary because communication between Ubuntu in WSL2 and Windows is not happening using localhost (127.0.0.1) but a custom IP address that changes from time to time.
Now the mentioned issues added
```
$PX4_SIM_HOST_ADDR"
      simulator start -t $simulator_tcp_port $PX4_SIM_HOST_ADDR
```
but that only changes the TCP address passed in `PX4_SIM_HOST_ADDR` with `set_tcp_remote_ipaddr`.
Therefore I could not use QGC on Windows, because no UDP mavlink messages on `<IP of WSL2 interface>:14550` are received.

By changing a line in `src/modules/mavlink/mavlink_main.cpp` as mentioned above, the QGC communication worked.
But I have no overview of all the possibilities and if this is the only place that would require changes.

mrpollo, 2021-06-07:
Have you tried the changes from @lovettchris on #17275 ?

He also published a video tutorial https://www.youtube.com/watch?v=DiqgsWIOoW4

Let me know if this works for you

lovettchris, 2021-06-07:
Actually @kistlin may have found a new bug where issues #16252 and #15965 where not fixed completely.  The `ROMFS\px4fmu_common\init.d-posix\px4-rc.simulator` passes `PX4_SIM_HOST_ADDR` through to the "simulator" module for the `simulator` module, but it seems @kistlin has found one more localhost hiding in `mavlink_main.cpp`, and I do not see any  code passing  `PX4_SIM_HOST_ADDR`  through to mavlink in `px4-rc.mavlink`, perhaps what happened to `px4-rc.simulator` needs to also happen here, using `-t` argument on mavlink also since `_src_addr_initialized` seems to also be set by a `-t` argument on line 1939 of `mavlink_main.cpp`.