PX4/PX4-Autopilot #17613: RTK GPS over CAN

FreshGresh, 2021-05-19:
RTK GPS over CAN (from Here3) is not currently compatible with PX4 (but is with Ardupilot).

Is this in the works or can I request it be added as a feature?

For reference, I am using a Pixhawk Cube Orange with Here3 connected via the CAN. I have demonstrated RTK GPS (fixed) using the Here+ over I2C.

FreshGresh, 2021-05-21:
Worked with v1.12-beta5, thanks! 
3D RTK GPS lock (fixed) just fine with Quadrotor X airframe. I'll try in flight with a fixed wing normal airplane soon. Looking forward to the v1.12 stable release!

closed 2021-08-31 by mrpollo

Franzi13, 2022-09-27:
Hi everyone,
I think there is still a problem using RTK GPS over CAN with the Pixhawk 4 and also with the new Pixhawk 6c board.
Is there a way to make it work there too?
Thank you in advance!