PX4/PX4-Autopilot #17614: 1.12 beta4 cpu load too high to arm

benjinne, 2021-05-19:
**Describe the bug**
Moving from 1.11 to 1.12 with a similar setup has increased the cpu load from around 70% to a little over 90% on a modalai fc-v1 board. Top report on 1.12 beta4:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                    4960  3.307   264/  512   0 (  0)  READY  3
   1 hpwork                          0  0.000   332/ 1268 249 (249)  w:sig  3
   2 lpwork                          0  0.000   844/ 1620  50 ( 50)  w:sig  3
   3 init                            0  0.000  2324/ 2932 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   436/ 1260 255 (255)  w:sem  5
 337 wq:INS1                       902  6.950  4412/ 6004 240 (240)  w:sem  5
  25 wq:hp_default                  78  0.582  1032/ 1900 237 (237)  w:sem  5
  29 dataman                         0  0.000   796/ 1204  90 ( 90)  w:sem  4
  31 wq:lp_default                 595  5.078  1244/ 1924 205 (205)  w:sem  5
 241 wq:uavcan                     283  2.170  1852/ 3628 236 (236)  w:sem  5
 244 uavcan_fw_srv                 212  1.608  1564/ 6004 120 (120)  READY  5
 252 wq:I2C3                        30  0.219   820/ 2340 244 (244)  w:sem  5
 255 wq:SPI1                       558  4.258  1628/ 2340 253 (253)  w:sem  5
 262 wq:SPI2                      1177  9.002  1844/ 2340 252 (252)  w:sem  5
 264 wq:SPI6                      1275  9.736  1428/ 2340 248 (248)  w:sem  5
 267 wq:I2C4                        46  0.349   884/ 2340 243 (243)  w:sem  5
 312 wq:I2C1                        81  0.611   920/ 2340 246 (246)  w:sem  5
 330 wq:nav_and_controllers        857  6.547  1332/ 2244 242 (242)  w:sem  5
 332 wq:rate_ctrl                 1443 11.039  1556/ 1956 255 (255)  w:sem  5
 333 wq:INS0                       806  6.191  4412/ 6004 241 (241)  w:sem  5
 476 micrortps_client_send         553  4.250  1828/ 2252 100 (100)  w:sig  4
 339 commander                     207  1.590  1244/ 3220 140 (140)  w:sig  5
 340 wq:INS2                       873  6.683  4412/ 6004 239 (239)  w:sem  5
 346 mavlink_if0                   891  6.708  1852/ 2828 100 (100)  READY  6
 399 gps                            41  0.345  1052/ 1684 205 (205)  w:sem  4
 441 mavlink_if1                   842  6.271  1788/ 2740 100 (100)  READY  4
 442 mavlink_rcv_if1                70  0.519  1588/ 4404 175 (175)  w:sem  4
 475 micrortps_client_rcv           67  0.507  2012/ 2532 100 (100)  READY  4
 553 mavlink_rcv_if0                59  0.439  1588/ 4404 175 (175)  w:sem  6
 480 wq:UART5                       91  0.692   844/ 1508 229 (229)  w:sem  5
 548 logger                         65  0.512  2436/ 3644 230 (230)  w:sem  3
 502 navigator                      14  0.115  1044/ 1772 105 (105)  w:sem  6
 550 log_writer_file                 0  0.000   388/ 1172  60 ( 60)  w:sem  3
 555 mavlink_shell                   0  0.000  1052/ 2028 100 (100)  w:sem  3
 556 top                           149  1.209  1964/ 4084 237 (237)  RUN    3

Processes: 35 total, 6 running, 29 sleeping, max FDs: 12
CPU usage: 94.19% tasks, 2.51% sched, 3.31% idle
DMA Memory: 5120 total, 1024 used 1536 peak
Uptime: 69.117s total, 4.960s idle
```

Current solution is to reduce IMU_GYRO_RATE max from 800 to 600, and set UAVCAN_ENABLE to disabled. This will cause the cpu load to get low enough to arm, but while flying the cpu load will raise above 90%.

My UAS is equipped with a mavlink radio, rtps connection, lidarlite-v3, GPS and mag, and crossfire receiver with telemetry enabled. The UAS has not changed from 1.11 to 1.12 and the cpu load increase issue is consistent on 1.12 beta4. 

Is there any other easy way to significantly reduce cpu load?

dagar, 2021-05-19:
What does `work_queue status` look like?

dagar, 2021-05-19:
Run `mavlink status` as well. Are you using both instances (nearly 14% cpu total)?

Note that top itself adds a bit of overhead when active (more than just the `top` frontend).

benjinne, 2021-05-20:
```

Work Queue: 14 threads                        RATE        INTERVAL
|__ 1) wq:rate_ctrl    
|   |__ 1) dshot                         620.9 Hz         1611 us
|   |__ 2) mc_rate_control               620.9 Hz         1611 us
|   \__ 3) vehicle_angular_velocity      620.9 Hz         1611 us
|__ 2) wq:SPI1         
|   \__ 1) icm20602                      562.0 Hz         1779 us
|__ 3) wq:SPI2         
|   \__ 1) icm42688p                     620.9 Hz         1611 us
|__ 4) wq:SPI6         
|   |__ 1) bmi088                        665.1 Hz         1504 us
|   \__ 2) bmi088                        537.1 Hz         1862 us
|__ 5) wq:I2C1         
|   \__ 1) lis3mdl                        70.7 Hz        14139 us
|__ 6) wq:I2C3         
|   \__ 1) voxlpm                         10.0 Hz       100000 us (100000 us)
|__ 7) wq:I2C4         
|   \__ 1) bmp388                         23.1 Hz        43300 us (43300 us)
|__ 8) wq:nav_and_controllers
|   |__ 1) ekf2_selector                 306.9 Hz         3258 us
|   |__ 2) flight_mode_manager            50.0 Hz        20000 us
|   |__ 3) land_detector                 100.0 Hz        10000 us
|   |__ 4) mc_att_control                207.0 Hz         4832 us
|   |__ 5) mc_hover_thrust_estimator     100.0 Hz        10000 us
|   |__ 6) mc_pos_control                100.0 Hz        10000 us
|   |__ 7) sensors                       207.0 Hz         4832 us
|   |__ 8) vehicle_acceleration          207.0 Hz         4832 us
|   |__ 9) vehicle_air_data               23.1 Hz        43300 us
|   |__10) vehicle_gps_position           10.0 Hz        99999 us
|   \__11) vehicle_magnetometer           70.7 Hz        14139 us
|__ 9) wq:INS0         
|   |__ 1) ekf2                          187.3 Hz         5338 us
|   \__ 2) vehicle_imu                   187.3 Hz         5338 us
|__ 10) wq:INS1         
|   |__ 1) ekf2                          207.0 Hz         4832 us
|   \__ 2) vehicle_imu                   207.0 Hz         4831 us
|__ 11) wq:INS2         
|   |__ 1) ekf2                          221.7 Hz         4511 us
|   \__ 2) vehicle_imu                   222.4 Hz         4496 us
|__ 12) wq:hp_default   
|   |__ 1) board_adc                     100.0 Hz        10000 us (10000 us)
|   |__ 2) rc_update                       0.0 Hz            0 us
|   \__ 3) safety_button                  30.3 Hz        33000 us (33000 us)
|__ 13) wq:UART5        
|   \__ 1) rc_input                      250.0 Hz         4000 us (4000 us)
\__ 14) wq:lp_default   
    |__ 1) esc_battery                    50.0 Hz        20000 us
    |__ 2) gyro_calibration               50.0 Hz        20000 us (20000 us)
    |__ 3) gyro_fft                      199.9 Hz         5003 us
    |__ 4) load_mon                        2.0 Hz       500000 us (500000 us)
    \__ 5) send_event                     30.0 Hz        33333 us (33333 us)
```

```
nsh> mavlink status

instance #0:
    mavlink chan: #0
    type:        USB CDC
    flow control: OFF
    rates:
      tx: 0.0 B/s
      txerr: 2341.6 B/s
      tx rate mult: 0.050
      tx rate max: 800000 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
      received from sysid: 255 compid: 190: 1087, lost: 9, last 34073 ms ago
    FTP enabled: YES, TX enabled: YES
    mode: Config
    MAVLink version: 2
    transport protocol: serial (/dev/ttyACM0 @2000000)
    ping statistics:
      last: 26.52 ms
      mean: 24.41 ms
      max: 59243208.00 ms
      min: 0.24 ms
      dropped packets: 59243

instance #1:
    GCS heartbeat valid
    mavlink chan: #1
    type:        GENERIC LINK OR RADIO
    flow control: ON
    rates:
      tx: 21599.2 B/s
      txerr: 0.0 B/s
      tx rate mult: 1.000
      tx rate max: 46080 B/s
      rx: 305.7 B/s
      rx loss: 0.0%
      received from sysid: 1 compid: 197: 597918, lost: 0, last 88 ms ago
      received from sysid: 255 compid: 190: 67, lost: 0, last 54 ms ago
    FTP enabled: YES, TX enabled: YES
    mode: Onboard
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS4 @921600)
    ping statistics:
      last: 14.88 ms
      mean: 59.22 ms
      max: 133.73 ms
      min: 12.94 ms
      dropped packets: 0
```

The mavlink status shows 2 instances. The 1st is usb, and the 2nd is a radio.

top may add some cpu usage, but when flying, the cpu load is consistently above 90%
[image]

dagar, 2021-05-20:
To be honest if it's a reasonably stable 90% during flight in a position control mode (everything in the system running) that doesn't actually worry me. Virtually all work is being done in threads and the priorities are arranged such that even if the system is pegged at 100% you can still fly. In that extreme it will be lower priority things like logger or mavlink that starve.

Suggestions
 - disable gyro_fft `param set IMU_GYRO_FFT_EN 0` (we're not using it by default yet)
 - disable gyro auto cal `param set IMU_GYRO_CAL_EN 0`
 - disable an IMU or reduce multi-ekf from 3 to 2 `param set EKF2_MULTI_IMU 2`
 - disable `volxpm` if you aren't using it (guessing based on `esc_battery` running)
 - reduce IMU_GYRO_RATEMAX to 400
      - the possible value in running faster depends on the size of the vehicle
      - all IMUs are still processing full raw data regardless of this setting
 - Are you using micrortps? We should check that it isn't processing more data than necessary.

benjinne, 2021-05-20:
Ok, I'll give that a try. I am using micrortps and that certainly contributes to higher cpu load and it's sending a couple more uorbs than default. 
[image]

benjinne, 2021-05-20:
Here's my quick unscientific findings from changing those parameters:

| Change | CPU % decrease |
| IMU_GYRO_FFT_EN 0 | 2.1%  |
| IMU_GYRO_CAL_EN    | 3.4% |
| EKF2_MULTI_IMU 2      | 4.5% |
| MAV_1_MODE normal   | 4.0% |
| Voxlpm stop                    | 1.5% |
|--------------------------------|-------------------|

mrpollo, 2021-05-28:
Thanks for posting an update @benjinne, can we consider this issue closed?

benjinne, 2021-06-01:
For my use-case I was able to reduce the cpu to an acceptable value, however the default parameters may need adjusted

mrpollo, 2021-06-01:
Thanks @benjinne

closed 2021-06-01 by mrpollo