PX4/PX4-Autopilot #17615: mpu9250 spi driver error at v1.11.3 on

MartinAtCoventry, 2021-05-19:
**Describe the bug**

I'm testing px4(v1.11.3) on raspberry pi4. And I met a fatal problem about mpu9250 spi driver.
The imu driver I used is at src/driver/imu/invensense/mpu9250. It's just for spi interface. All of
compilation are sucessful, but there is an error info after the program running 3 or 4 minutes.
`ERROR [sensors] Accel #0 fail:  ERR CNT!`, this error 

I don't know what meaning `ERR CNT` is ?  This fault leads to imu and mag-compass crash on
QGC, but in px4 console still show all sensors info after inputting the command `sensors status`.
Who could help me solve the mpu9250 problem, please?

**Screenshots**
```
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
+ BAT1_V_CHANNEL: curr: -1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
bmp280 #0 on SPI bus 2 (devid=0x0)
mpu9250 #0 on SPI bus 2 (devid=0x1)
INFO  [mavlink] mode: Normal, data rate: 5760 B/s on /dev/ttyS2 @ 115200B
INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in PCA9685 mode
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ecl/EKF] 1211103446: reset position to last known position
INFO  [ecl/EKF] 1211103446: reset velocity to zero
INFO  [ecl/EKF] 1218138452: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
ERROR [sensors] Accel #0 fail:  ERR CNT!
```

**Drone (please complete the following information):**
- project version: v1.11.3
- mpu9250 and bmp280 on the same spi bus, but different cs pin.
- Linux debian 10.3

dagar, 2021-05-19:
Could you try current master or v1.12 beta?

MartinAtCoventry, 2021-05-20:
Hi @dagar , I have deployed v1.12.0 beta5 flight stack on my rpi4 board. Unfortunately, that imu fault still exists there.
`ERROR [sensors] Accel #0 fail:  ERR CNT!`

Also, I have run the px4 command `sensors status` to check the status of mpu9250 when it is normal and abnormal.

**Test environment**
- project version: v1.12.0 beta5
- mpu9250 and bmp280 on the same spi bus, but different cs pin.
- Linux debian 10.3
- my rpi4 board with mpu9250 module on it has been able to successfully run Ardupilot flight stack in air.

**Booting and fault Screenshots**
```
INFO  [px4] mlockall() enabled. PX4's virtual address space is locked into RAM.
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
WARN  [uORB] already loaded
INFO  [parameters] BSON document size 147 bytes, decoded 147 bytes
  BAT1_V_CHANNEL: curr: -1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
ERROR [SPI_I2C] Bug: driver bmp280 does not pass the I2C address to I2CSPIDriverBase
bmp280 #0 on SPI bus 2 (devid=0x0)
mpu9250 #0 on SPI bus 2 (devid=0x1)
INFO  [arch_adc] Initializing adc finish!
INFO  [mavlink] mode: Normal, data rate: 5760 B/s on /dev/ttyO4 @ 115200B
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
WARN  [drivers_board] number of outputs too large. Setting to 6
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ecl/EKF] 1544861967: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
ERROR [sensors] Accel #0 fail:  ERR CNT!
```

**Sensors status when the IMU is normal working**
```
INFO  [sensors] selected gyro: 2359570 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator] 	val:  -0.1387, lp:   0.6857 mean dev:   0.0027 RMS:   1.0772 conf:   1.0000
INFO  [data_validator] 	val:  -0.9768, lp:  -0.2656 mean dev:  -0.0011 RMS:   0.4047 conf:   1.0000
INFO  [data_validator] 	val:  -0.1121, lp:   0.1565 mean dev:   0.0005 RMS:   0.2057 conf:   1.0000

INFO  [sensors] selected accel: 2359570 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator] 	val:  -0.2772, lp:   1.0178 mean dev:   0.0024 RMS:   0.5726 conf:   1.0000
INFO  [data_validator] 	val:  -2.8412, lp:   0.5784 mean dev:   0.0034 RMS:   1.4934 conf:   1.0000
INFO  [data_validator] 	val:  -9.1707, lp:  -9.4622 mean dev:   0.0022 RMS:   0.3568 conf:   1.0000

INFO  [vehicle_magnetometer] selected magnetometer: 262418 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator] 	val:   0.7025, lp:   0.7509 mean dev:   0.0063 RMS:   0.0586 conf:   1.0000
INFO  [data_validator] 	val:  -0.2806, lp:  -0.3211 mean dev:   0.0017 RMS:   0.0460 conf:   1.0000
INFO  [data_validator] 	val:   0.2463, lp:   0.0038 mean dev:   0.0053 RMS:   0.1250 conf:   1.0000
INFO  [sensor_calibration] MAG 262418 EN: 1, offset: [ 0.000  0.000  0.000], scale: [ 1.000  1.000  1.000], Internal

INFO  [vehicle_air_data] selected barometer: 4128786 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
[... 53 lines cut ...]
INFO  [data_validator] 	val:  28.9891, lp:  28.9984 mean dev:   0.0209 RMS:   0.0294 conf:   1.0000
INFO  [data_validator] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000

INFO  [sensors] Airspeed status:
INFO  [data_validator] 	no data

INFO  [vehicle_acceleration] selected sensor: 2359570, rate: 1000.0 Hz, estimated bias: [0.0000 0.0000 -0.3668]
INFO  [sensor_calibration] ACC 2359570 EN: 1, offset: [0.0000 0.0000 0.0000] scale: [1.0000 1.0000 1.0000]

INFO  [vehicle_angular_velocity] selected sensor: 2359570, rate: 8002.2 Hz FIFO
INFO  [vehicle_angular_velocity] estimated bias: [0.0110 0.0097 0.0245]
INFO  [sensor_calibration] GYRO 2359570 EN: 1, offset: [0.0000 0.0000 0.0000]

INFO  [vehicle_imu] 0 - IMU ID: 2359570, accel interval: 2496.9 us, gyro interval: 2496.9 us
vehicle_imu: accel data gap: 1 events
vehicle_imu: gyro data gap: 1 events
vehicle_imu: accel update interval: 82778 events, 2499.97us avg, min 776us max 4653us 46.317us rms
vehicle_imu: gyro update interval: 82778 events, 2499.97us avg, min 776us max 4653us 46.317us rms
INFO  [sensor_calibration] ACC 2359570 EN: 1, offset: [0.0000 0.0000 0.0000] scale: [1.0000 1.0000 1.0000]
INFO  [sensor_calibration] GYRO 2359570 EN: 1, offset: [0.0000 0.0000 0.0000]
```
It seems that  the gyroscope is also failed when the accelerometer is crash. By the way, could you tell me what meaning the `ECNT` is ?

I guess whether the spi speed is too fast to mpu9250 or not, because I browsed another famous project `fprime`. There are a few of comment lines to notice that MPU9x50 might be reliable and corrupted when its spi interface run at 20Mhz.
https://github.com/nasa/fprime/blob/devel/Drv/LinuxSpiDriver/LinuxSpiDriverComponentImpl.hpp

dagar, 2021-05-20:
> It seems that the gyroscope is also failed when the accelerometer is crash. By the way, could you tell me what meaning the ECNT is ?

It's an error count coming from the driver. It could be the driver detected corrupted data in the FIFO, or by checking its own configuration. Ideally the driver would be able to cleanly reset and recover, but it depends on what's actually happening.

Can you share `mpu9250 status` and `perf` before and after the "crash"?

> I guess whether the spi speed is too fast to mpu9250 or not, because I browsed another famous project fprime. There are a few of comment lines to notice that MPU9x50 might be reliable and corrupted when its spi interface run at 20Mhz.

We're running the mpu9250 at 10 MHz for that reason, but you may need to lower it even further depending on the physical setup. https://github.com/PX4/PX4-Autopilot/blob/1f40a652104426f902abe7165a3794d2dcb254ae/src/drivers/imu/invensense/mpu9250/InvenSense_MPU9250_registers.hpp#L61

Is the data otherwise good for the first 3 or 4 minutes? Can you share a log from boot (https://logs.px4.io/)? 

Things to try
  - lower the mpu9250 frequency
  - run px4 as root
  - get a log showing good data, then when it turns bad
  - while px4 is running get the output of `top` or `htop` with "show custom thread names" set (htop --pid=`pidof px4`)

MartinAtCoventry, 2021-05-20:
Hi @dagar，could you tell me where log files are usually stored in linux environment?

I just manually execute `px4` in a linux terminal by typing `sudo ./px4 -s px4.config`. So, in the directory where the executable file `px4` is placed has a `log` directory, but nothing in there after finishing `px4`.

MartinAtCoventry, 2021-05-20:
@dagar , which are the command `top` and `htop` you said belong linux system build-in or `px4`build-in ? Because I compiled `px4` without the `top` command module.

dagar, 2021-05-20:
> Hi @dagar，could you tell me where log files are usually stored in linux environment?

Try `find . -name \*.ulg`

dagar, 2021-05-20:
> @dagar , which are the command `top` and `htop` you said belong linux system build-in or `px4`build-in ? Because I compiled `px4` without the `top` command module.

I'm talking about running top or htop outside of PX4 in Linux.

MartinAtCoventry, 2021-05-20:
Hi @dagar , I still can't find any .ulg file on my rpi4 board. I tested both v1.12.0beta5 and v1.11.3 versions, all of them can't generate any .ulg file in `log` directory or where else.

Could you tell which code is about generating .ulg file ?

MartinAtCoventry, 2021-05-21:
Hi @dagar , I have tested v1.12.0 PX4 on my board. I separately tested mpu9250 run at 10MHz, 5MHz and 1MHz. Unfortunately, crash occurred at all of 3 speeds.

**Test Method**
- Execute `px4` as root :  `sudo ./px4 -s px4.config`.
- Respectively generate 3 .ulg log files at 3 different spi bus rate.

**Test Result**
1. The status mpu9250 at normal and crash states
  - normal state
   ```
     INFO  [SPI_I2C] Running on SPI Bus 2
     INFO  [mpu9250] FIFO empty interval: 2500 us (400.0 Hz)
     mpu9250: bad register: 0 events
     mpu9250: bad transfer: 3537 events
     mpu9250: FIFO empty: 0 events
     mpu9250: FIFO overflow: 1 events
     mpu9250: FIFO reset: 2 events
     mpu9250_ak8963: bad transfer: 1 events
     mpu9250_ak09916: magnetic sensor overflow: 0 events
   ```
  - crash state
   ```
     INFO  [SPI_I2C] Running on SPI Bus 2
     INFO  [mpu9250] FIFO empty interval: 2500 us (400.0 Hz)
     mpu9250: bad register: 0 events
     mpu9250: bad transfer: 11392 events
     mpu9250: FIFO empty: 0 events
     mpu9250: FIFO overflow: 1 events
     mpu9250: FIFO reset: 2 events
     mpu9250_ak8963: bad transfer: 1 events
     mpu9250_ak09916: magnetic sensor overflow: 0 events
   ```
2. The `htop` result at 2 different states
  - normal state
   
    [image]

  - crash state
  
    [image]

3. The `work_queue` status
   ```
     Work Queue: 7 threads                        RATE        INTERVAL
     |__ 1) wq:rate_ctrl    
     |   |__ 1) mc_rate_control               400.2 Hz         2499 us
     |   \__ 2) vehicle_angular_velocity      400.1 Hz         2499 us
     |__ 2) wq:SPI2         
     |   |__ 1) bmp280                         23.8 Hz        42073 us
     |   |__ 2) mpu9250                       399.1 Hz         2505 us (2500 us)
     |   \__ 3) mpu9250_ak8963                 50.4 Hz        19849 us
     |__ 3) wq:nav_and_controllers
     |   |__ 1) flight_mode_manager            50.3 Hz        19900 us
     |   |__ 2) land_detector                 100.5 Hz         9950 us
     |   |__ 3) mc_att_control                201.0 Hz         4976 us
     |   |__ 4) mc_hover_thrust_estimator     100.5 Hz         9951 us
     |   |__ 5) mc_pos_control                100.5 Hz         9951 us
     |   |__ 6) sensors                       200.8 Hz         4980 us
     |   |__ 7) vehicle_acceleration          400.4 Hz         2498 us
     |   |__ 8) vehicle_air_data               11.9 Hz        84053 us
     |   \__ 9) vehicle_magnetometer           54.9 Hz        18199 us
     |__ 4) wq:INS0         
     |   |__ 1) ekf2                          200.4 Hz         4991 us
     |   \__ 2) vehicle_imu                   200.2 Hz         4995 us
     |__ 5) wq:hp_default   
     |   |__ 1) battery_status                100.2 Hz         9977 us
     |   |__ 2) board_adc                     100.0 Hz        10000 us (10000 us)
     |   |__ 3) linux_pwm_out                 400.5 Hz         2497 us
     |   \__ 4) rc_update                       0.0 Hz            0 us
     |__ 6) wq:pru_receiver 
     |   \__ 1) pru_receiver                 1000.0 Hz         1000 us (1000 us)
     \__ 7) wq:lp_default   
         \__ 1) load_mon                        2.0 Hz       499263 us (500000 us)
   ```

4. The log data at 3 different SPI bus speeds
    The 3 log data are all from booting to shutdown, which includes crash state. 
   (1) 10Mhz
         https://review.px4.io/plot_app?log=f58e70a7-8aeb-4752-bc6c-292c56961a77

   (2) 5MHz
         https://review.px4.io/plot_app?log=fa2db444-6471-4991-9438-86524053bee2

   (3) 1MHz
         https://review.px4.io/plot_app?log=9451973a-3e69-45a0-8ec3-a78742f7c482

dagar, 2021-05-22:
Any difference if you don't use the mag within the mpu9250?

dagar, 2021-05-23:
Looking at your logs the error count (from bad transfers) is growing constantly.

[image]

We try to detect errors in the transfer by looking at the duplicated accel data in the FIFO. This could be one reason it seems to "work" on ardupilot, but error on PX4.
https://github.com/PX4/PX4-Autopilot/blob/8327335fa48d526dc28be46bf20c5ac27d0b9a4e/src/drivers/imu/invensense/mpu9250/MPU9250.cpp#L552-L571

What happens if you comment this line? https://github.com/PX4/PX4-Autopilot/blob/8327335fa48d526dc28be46bf20c5ac27d0b9a4e/src/drivers/imu/invensense/mpu9250/MPU9250.cpp#L568

The other thing that seems not quite right is that the error count continues to grow, but sensor resets aren't being triggered. Although your transfer errors are regular, they appear to be at a lower rate than the successfully errors. I'll try adjusting this.
https://github.com/PX4/PX4-Autopilot/blob/8327335fa48d526dc28be46bf20c5ac27d0b9a4e/src/drivers/imu/invensense/mpu9250/MPU9250.cpp#L269-L271

dagar, 2021-05-23:
In `htop` setup select "show custom thread names".

[image]

dagar, 2021-05-23:
@MartinAtCoventry I did some testing on px4_fmuv-4 (pixracer) and was able to reproduce this when the mpu9250 magnetometer (ak8963) is enabled. 

[image]

I'll debug further.

dagar, 2021-05-23:
Hopefully fixed by https://github.com/PX4/PX4-Autopilot/pull/17633.

MartinAtCoventry, 2021-05-24:
Thanks for @dagar , I will test the driver on my hardware platform at once.

MartinAtCoventry, 2021-05-24:
Hi @dagar , I modified the mpu9250 code and tested it according to #17633. The mpu9250 run on SPI at 10MHz. Now, the program has had not any error info about mpu9250. Also, I enabled  the `ist8310` driver module for driving a magnetometer on my GPS module, which make the magnetometer on mpu9250 work with `ist8310` sensor.
However, there is a new error info in `ekf2`module about magnetometer as well.
` ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 4 -> 25`

**Htop information**
[image]

**work_queue status**
```
   Work Queue: 8 threads                        RATE        INTERVAL
   |__ 1) wq:rate_ctrl    
   |   |__ 1) mc_rate_control               374.8 Hz         2668 us
   |   \__ 2) vehicle_angular_velocity      374.8 Hz         2668 us
   |__ 2) wq:SPI2         
   |   |__ 1) bmp280                         23.0 Hz        43485 us
   |   |__ 2) mpu9250                       399.8 Hz         2501 us (2500 us)
   |   \__ 3) mpu9250_ak8963                 49.3 Hz        20277 us
   |__ 3) wq:I2C2         
   |   \__ 1) ist8310                        45.5 Hz        21975 us
   |__ 4) wq:nav_and_controllers
   |   |__ 1) flight_mode_manager            50.0 Hz        19998 us
   |   |__ 2) land_detector                  97.2 Hz        10286 us
   |   |__ 3) mc_att_control                199.7 Hz         5007 us
   |   |__ 4) mc_hover_thrust_estimator      99.9 Hz        10005 us
   |   |__ 5) mc_pos_control                 96.9 Hz        10321 us
   |   |__ 6) sensors                       199.5 Hz         5012 us
   |   |__ 7) vehicle_acceleration          374.8 Hz         2668 us
   |   |__ 8) vehicle_air_data               11.5 Hz        86961 us
   |   \__ 9) vehicle_magnetometer           91.0 Hz        10987 us
   |__ 5) wq:INS0         
   |   |__ 1) ekf2                          199.4 Hz         5014 us
   |   \__ 2) vehicle_imu                   199.7 Hz         5008 us
   |__ 6) wq:hp_default   
   |   |__ 1) battery_status                100.0 Hz        10000 us
   |   |__ 2) board_adc                     100.0 Hz        10000 us (10000 us)
   |   |__ 3) linux_pwm_out                 374.8 Hz         2668 us
   |   \__ 4) rc_update                       0.0 Hz            0 us
   |__ 7) wq:pru_receiver 
   |   \__ 1) pru_receiver                  999.6 Hz         1000 us (1000 us)
   \__ 8) wq:lp_default   
       \__ 1) load_mon                        2.0 Hz       499464 us (500000 us)
```

**sensors status**
```
   INFO  [sensors] selected gyro: 2359570 (0)
   INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
   INFO  [data_validator] sensor #0, prio: 50, state: OK
   INFO  [data_validator] 	val:   0.0116, lp:   0.0094 mean dev:   0.0004 RMS:   0.9328 conf:   1.0000
   INFO  [data_validator] 	val:  -0.0097, lp:  -0.0108 mean dev:   0.0002 RMS:   1.2671 conf:   1.0000
   INFO  [data_validator] 	val:   0.0249, lp:   0.0238 mean dev:  -0.0000 RMS:   0.5921 conf:   1.0000

   INFO  [sensors] selected accel: 2359570 (0)
   INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
   INFO  [data_validator] sensor #0, prio: 50, state: OK
   INFO  [data_validator] 	val:  -0.5925, lp:  -0.5446 mean dev:  -0.0025 RMS:   1.4183 conf:   1.0000
   INFO  [data_validator] 	val:  -0.1917, lp:  -0.2489 mean dev:   0.0009 RMS:   1.1379 conf:   1.0000
   INFO  [data_validator] 	val: -10.1375, lp: -10.1263 mean dev:  -0.0008 RMS:   0.9879 conf:   1.0000

   INFO  [vehicle_magnetometer] selected magnetometer: 397073 (0)
   INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
   INFO  [data_validator] sensor #0, prio: 50, state: OK
   INFO  [data_validator] 	val:  -2.4674, lp:  -2.4679 mean dev:  -0.0081 RMS:   0.2832 conf:   1.0000
   INFO  [data_validator] 	val:  -2.9402, lp:  -2.9392 mean dev:  -0.0097 RMS:   0.2669 conf:   1.0000
   INFO  [data_validator] 	val:   2.3492, lp:   2.3567 mean dev:   0.0089 RMS:   0.1840 conf:   1.0000
   INFO  [data_validator] sensor #1, prio: 50, state: OK
   INFO  [data_validator] 	val:  -0.3094, lp:  -0.3023 mean dev:  -0.0025 RMS:   0.0871 conf:   1.0000
   INFO  [data_validator] 	val:  -0.7894, lp:  -0.7853 mean dev:  -0.0014 RMS:   0.0856 conf:   1.0000
   INFO  [data_validator] 	val:  -0.2359, lp:  -0.2392 mean dev:  -0.0012 RMS:   0.0732 conf:   1.0000
   INFO  [sensor_calibration] MAG 397073 EN: 1, offset: [ 0.000  0.000  0.000], scale: [ 1.000  1.000  1.000], Internal
[... 28 lines cut ...]
**MPU9250 status**
```
   INFO  [SPI_I2C] Running on SPI Bus 2
   INFO  [mpu9250] FIFO empty interval: 2500 us (400.0 Hz)
   mpu9250: bad register: 0 events
   mpu9250: bad transfer: 0 events
   mpu9250: FIFO empty: 3 events
   mpu9250: FIFO overflow: 12 events
   mpu9250: FIFO reset: 12 events
   mpu9250_ak8963: bad transfer: 0 events
   mpu9250_ak09916: magnetic sensor overflow: 0 events
```

**ist8310 status**
```
   INFO  [SPI_I2C] Running on I2C Bus 2
   ist8310: reset: 0 events
   ist8310: bad register: 0 events
   ist8310: bad transfer: 0 events
   pxh> mpu9250 status
```

**Flight log**
mpu9250 at 10MHz on SPI & ist8310 on I2C
https://review.px4.io/plot_app?log=1aa48731-0a16-4a02-9e3d-92533360e951

MartinAtCoventry, 2021-05-24:
The log data is too big, its size is 220MB.

dagar, 2021-05-24:
Thanks for testing, from what I can see that all looks good.

That ekf2 magnetometer error is harmless and will be silenced shortly. https://github.com/PX4/PX4-Autopilot/pull/17639

If you have a good external mag I would consider not even running the mag on the mpu9250.

MartinAtCoventry, 2021-05-24:
I have another problem. After running the new compiled flight stack and connect to QGC, the QGC always throw out a parameter missing notice when I go into `Vehicle Setup`.
[image]

So, I don't know where is something wrong ? I checked the start script `px4.config`. I found I have set the parameter `SYS_AUTOSTART` to 4001.

dagar, 2021-05-24:
The init scripts for these Linux boards is still off to the side and hasn't been unified with the rest of the system and "normal" startup.

For now try adding `param set SYS_AUTOCONFIG 4001` in your px4.config. If you're interested in the slightly bigger/"proper" solution I'd be happy to discuss it in detail later.

MartinAtCoventry, 2021-05-24:
Thanks for your help, that QGC problem has been solved. Other questions, I will discuss with you at `PX4 FORUM`.

commit e18b1c03 (2021-05-24): mpu9250: tolerate missing/mismatching FIFO samples if I2C slave mag active

 - when the mpu9250 mag is active it disrupts IMU sampling
 - this change modifies the simple mpu9250 FIFO integrity checks (comparing duplicate accel samples) to tolerate these samples that don't match if the mag is active
 - fixes #17615
closed 2021-05-24 by MartinAtCoventry