PX4/PX4-Autopilot #17616: problem with micrortps.html

b0hne, 2021-05-20:
in https://docs.px4.io/master/en/middleware/micrortps.html#agent-in-an-offboard-fast-dds-interface-ros-independent:
cd build/<target-platform>/src/modules/micrortps_bridge/micrortps_client/micrortps_agent
does not exist.
When trying:
build/<target-platform>/src/modules/micrortps_bridge/micrortps_agent
i get:
_bridge/micrortps_agent/src/microRTPS_timesync.h:46:10: fatal error: Timesync_Publisher.h: No such file or directory
   46 | #include "Timesync_Publisher.h"

tugbakara, 2021-11-02:
I got smilar problem, did you find the solution?

b0hne, 2021-11-02:
I think it worked after using the raspberry pi environment setup.
https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#raspberry-pi

zenglongGH, 2022-01-28:
I got smilar problem, did you find the solution? @tugbakara 
it not work using the raspberry pi environment setup. @b0hne

tugbakara, 2022-02-03:
Could you share the logs if not a problem I want to compare with mine.

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: ***@***.***>
Gönderilme: 28 Ocak 2022 Cuma 11:01
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [PX4/PX4-Autopilot] problem with micrortps.html (#17616)

I got smilar problem, did you find the solution? @tugbakara<https://github.com/tugbakara>
it not work using the raspberry pi environment setup. @b0hne<https://github.com/b0hne>