PX4/PX4-Autopilot #17623: VTOL tailsitter doesn't transition to forward flight

Etruscian, 2021-05-21:
**Describe the bug**
Sometimes the aircraft snaps to a pitch angle of 0 before pitching forward during the forward transition. On rare occasions, the aircraft snaps to a pitch angle of 0 and never starts the transition.

**To Reproduce**
Steps to reproduce the behavior:
1. Take off with a VTOL tailsitter in MC mode
2. Trigger the forward transition

**Expected behavior**
When a forward transition is triggered, the aircraft performs a forward transition starting from the current pitch angle

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=c0c8dba9-4d3a-41ed-9e4d-fc77f77b08b8

**Drone (please complete the following information):**
- VTOL tailsitter with 4 motors in + configuration

**Additional context**
I've added a log showing both successful forward transitions, forward transitions that started from a pitch angle of 0 and two transitions where the aircraft does not perform a transition at all. In this log, the mc_virtual_attitude_setpoint topic is logged, as well as the scheduled vtol mode, actual vtol mode, calculated tilt, and the time since transition start, respectively in debug_key_value 0, 1, 2, and 3.

Next to this, I've added q_trans_start as a logged message.

In this log, the transition snapped back to 0 pitch angle and/or didn't happen a total of 5 times. Each of those times, q_trans_start was taken as a quaternion where the 2nd and 3rd term are 0. In all correct transitions, these terms have values not equal to 0. During the transition attempts where the transition did not happen at all, the calculated tilt was logged as NaN.

Looking at the timestamps, when an incorrect transition occurs, the position controller publishes a new setpoint before q_trans_start is logged. This could indicate a race condition where for some reason, the vtol_att_control module is delayed and thus can't take the correct attitude setpoint as a starting point of the transition. During this delay, the flight task is already switched from position to transition, and the setpoint from the latter flight task is published.

The second issue is that it appears that when a wrong initial setpoint is taken, sometimes the normalization of q_trans_sp fails and the arc cosine is taken of a number larger than 1. This causes the aircraft to never pitch forward.

Adding the extra logging fields may have increased the frequency at which the failed transitions occur, but it was also present when the extra log fields were not in the firmware.

The log itself contains timestamps in the description explaining what happened when.

xdwgood, 2021-05-21:
As far as I know, the latest master should have solved this problem

xdwgood, 2021-05-21:
here: https://github.com/PX4/PX4-Autopilot/pull/17310

Etruscian, 2021-05-21:
Ah, I can't believe that I didn't find this! Thanks for pointing it out!

closed 2021-05-21 by Etruscian