PX4/PX4-Autopilot #17635: Navigation Errors during FW Flight

ryanjAA, 2021-05-24:
Pixhawk 4
1.11.3 - Fixed Wing

After some debate on this I decided to open an issue.

Navigation between waypoints is not consistent and does not follow the expected direction (more detail below). The direction is circuitous.

In the first log, it was clear there was some mag interference (this was seen on the ground prior to takeoff - approximately 10 degrees could not be corrected even after multiple calibrations) but the issue was that the trajectory from one waypoint to the next does not make sense (in parts of the flight). Instead of going from one point to the next, there are large turns between them.

I questioned if this was due to an insufficient SD card speed for reading the mission. Testing the SD card speed in QGC gave an average speed of 63kb/s (much too slow but this was only changed after these flights) and there are logging dropouts at some of the points in the log. Right around 22 min where the location looks to be sideways, it was not. In the telemetry replay it does a coordinated turn. This turn at least makes sense, that is the direction is was supposed to go. That changes though later in the same flight.

At 22:46 it looks like the aircraft starts to circle down. Maybe to hit the altitude set-point but that does not make too much sense seems it looks like it doesnt actually miss it. After it does circle down though it then just decides to go left (no errors still) and then circle around again. Mag issue? There is no reason for a turn to be in that direction though. The question becomes is if this is mag related and or SD card read access speed and mission related.

Log 1 - 
https://review.px4.io/plot_app?log=72ac3453-0872-447d-8716-3fcca37c1e9e

The below log had a new mag installed and there weren't any offsets observed or mag ratios, etc but at 13:17 instead of turning left, the aircraft made a roll to the right before banking to the left to turn. Additionally, it went directly through the waypoint before turning instead of starting at the L1 distance and making the typical rounded turn. From 13:18-13:25 the log looks like the aircraft is going to the wrong direction but there is also a log dropout there, in the telemetry log the orientation and direction looks correct in terms of how the aircraft flew but not why it still turned so late and went right then left. 

Log 2-
https://review.px4.io/plot_app?log=1e2b37e4-b557-45d3-9f73-6fb099bf0fa4

dagar, 2021-05-29:
There are definitely estimator issues, but I'm not sure if that explains your issue. Let's dig a little deeper.

ryanjAA, 2021-05-29:
Ok - definitely strange behavior that doesn’t seem to make sense. There are some other logs from the same days as well.

Antiheavy, 2021-05-30:
is this amount of CPU load variability normal for your version of PX4 and your vehicle?  I'm used to seeing something much more steady (although we are on older PX4 version), there are logging dropouts that correlate with the higher CPU load sections:
[image]

Do you often command 55 degree roll angles?  That is pretty high roll angle for most small aircraft.  I wonder if the synthetic side-slip math isn't appropriate for turns with very high bank angles?
[image]

ryanjAA, 2021-05-30:
@Antiheavy no, that amount of cpu usage is not typical and I saw in some of the preflight logs and on qgc a cpu fail warning due to high cpu usage so I rebooted but these are the same settings as usually flown. I changed the logging at the last moment to get an issue with mission transferring to work but that might have been sd card related (seems like somethings though at this point). Logging is on from boot so all the logs from boot are also on hand. 

As for the high roll angle, we leave it at that so it can turn sharply if need be and the large deflection surfaces make it possible to achieve but maybe that is an issue with sss math. I’m not sure. I believe one log we had  steady 10-15 mph wind and higher gusts (I can check the measurements we took that day) but the whole scenario of these flights doesnt fit anything I’ve seen. 

Was using unmodified qgc and unmodified 1.11.3 for this. 
[image]

ryanjAA, 2021-06-02:
Logged on bench from right now with same SD. While connected (wirelessly) and running top, log_writer_file hits 61% way too high.

Changing nothing else except the SD to a SanDisk Extreme U3 32GB and log_writer_file stays in single digits (maybe hits 10 once?) 

This only happens when armed (as right now I have this aircraft set to log only when armed) but I wonder if the slow SD card is backing up the CPU and creating high CPU usage which was then not giving enough resources to things like the navigator from being able to operate correctly.

Here are the two logs:

Existing SD card with CPU usage high.
[HERE](https://review.px4.io/plot_app?log=73468f1c-5bed-4fa8-918b-2f1f1ffec99a)

The dropouts coincide with high CPU usage (which again, I wonder if was also effecting the navigation). 

New SD card and CPU low.
[HERE](https://review.px4.io/plot_app?log=9afb6749-15ad-46d1-9ab8-4a7cdf6bfbce)

If this is confirmed (@dagar) then maybe we need to think about something in the docs @hamishwillee given this is actually pretty severe or can cause very bad unintended consequences if the SD card is really bad (and these are Amazon bought class 10 cards but still just junk IMO).

[image]

ryanjAA, 2021-06-03:
The more I think about this the more I think this shouldn’t be able to happen. If something like logging can almost bring down a vehicle shouldn’t there be a safeguard or circuit breaker in here where if resources (cpu, load, etc) from something non-critical spikes then it gets shutdown? That makes most sense. Maybe you lose part of the log but you have your vehicle, etc and a flag and message saying whatever it was.

Antiheavy, 2021-06-07:
>  If something like logging can almost bring down a vehicle shouldn’t there be a safeguard or circuit breaker

seems like a smart idea huh?    

We stopped using Class 10 SDcards long ago and only use Sandisk Extreme U3 rated cards.   We found the logger was bringing down the system in similar and unpredictable ways, like what you saw when we had Class 10 SDcards.   Even now, someone in the lab will not pay attention, put in some Class 10 SDcard, go out flying or even just parking lot testing, and get weird unexpected behavior and sensor warnings.    Throw them Class 10 cards out!

ryanjAA, 2021-06-07:
Absolutely! It shouldn’t even be possible!

This is the first time we’ve ever seen this happen but it will hopefully be the last. Running sd card speed tests is pretty telling (from qgc) so these are all going in the trash.

ThomasRigi, 2021-10-15:
I think I had the same navigation error on a simulator, v1.12.1: https://logs.px4.io/plot_app?log=2a3afbae-e556-450e-83ab-8353aac0a0f2

Around 33:54 it happens a first time, at 35:15 a second time and somewhat differently at 38:30-38:45. 
No idea why it happened. I flew the same mission a second time without restarting the simulator and all went well again. It's a mission we've flown many times on real vehicles too.

- it wasn't a slow SD card issue (I still think the safeguard makes sense just in case)
- CPU varies a lot but I don't know how that is calculated on a simulated drone.
- altitude was good, wasn't some sort of loiter-to-altitude issue
- no watchdog, no reason why mag would have been bad