PX4/PX4-Autopilot #17641: Makefile:226: recipe for target 'px4_sitl' failed make: *** [px4_sitl] Error 1 FAILED: ROMFS/romfs_extract.stamp

Dagu12s, 2021-05-24:
Hello, im having a lot of troubles trying to install PX4-Autopilot with gazebo. Im in windows 10 with a virtual machine with ubuntu 18 and my goal is to install PX4-autopilot and run it with Gazebo and ROS (1). I've tryed with bash ./PX4-Autopilot/Tools/setup/ubuntu.sh and running it separatly downloading the px4-autopilot repository and runing 
`make px4_sitl jmavsim` . At first if i run this make` px4_sitl jmavsim` it works and i can simulate it but when i run `make px4_sitl gazebo` it generates an error 
`Generating romfs_extract.stamp
FAILED: ROMFS/romfs_extract.stamp 
cd /home/david/PX4-Autopilot/build/px4_sitl_default/etc && /usr/bin/cmake -E remove_directory /home/david/PX4-Autopilot/build/px4_sitl_default/etc/* && /usr/bin/cmake -E tar xf /home/david/PX4-Autopilot/build/px4_sitl_default/romfs_files.tar && /usr/bin/cmake -E touch /home/david/PX4-Autopilot/build/px4_sitl_default/ROMFS/romfs_extract.stamp
CMake Error: cmake version 3.10.2 `

Afer that i can not run `make px4_sitl jmavsim`. I tryed it with differents virtual machines and it didnt work

Thank you

tourist0714, 2021-05-27:
Accrodding to "Troubleshooting->General Build Errors":

> Many build problems are caused by either mismatching submodules or an incompletely cleaned-up build environment. Updating the submodules and doing a distclean can fix these kinds of errors:

so i run `git submodule update --recursive` and then run `make distclean`.Now i solve the problem.

Dagu12s, 2021-05-28:
Thank you very much. 
I tried a lot of things but it seems that this `git submodule update --recursive` was the solution.
Again, thank you very much

closed 2021-05-28 by Dagu12s