PX4/PX4-Autopilot #17644: latest tbb is causing `make px4_sitl gazebo` failed

tenggyut, 2021-05-25:
**Describe the bug**

the newest tbb has removed `tbb::task` api, gazebo11 homebrew bottle depends on tbb@2020 as a workaround, but `make px4_sitl gazebo` still use header file in `/usr/local/include/tbb` which is the latest tbb.

**To Reproduce**
brew tap PX4/px4
brew install px4-dev
python3 -m pip install --user pyserial empy toml numpy pandas jinja2 pyyaml pyros-genmsg packaging
brew install --cask xquartz
brew install px4-sim-gazebo

**Expected behavior**
```
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:39:58: error: expected class name
    class GZ_TRANSPORT_VISIBLE PublishTask : public tbb::task
                                                         ^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:54:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:162:52: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                PublishTask *task = new(tbb::task::allocate_root())
                                        ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:165:17: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                tbb::task::enqueue(*task);
                ^~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/net/if_var.h:229:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
8 errors generated.
ninja: build stopped: subcommand failed.
```

**Additional context**
homebrew related issue  [#issue1477](https://github.com/osrf/homebrew-simulation/issues/1477)

julianoes, 2021-05-25:
@tenggyut what's your suggestion how to fix this?

tenggyut, 2021-05-25:
currently, my workaround is link tbb@2020 include dir in /usr/local/include with a different name, like tbb2020. And change `#include <tbb/task.h>` to `#include <tbb2020/task.h>`. but this is my local workaround, may not suitable for more users...

the best solutioin that I can think of right now is maintain a copy of tbb2020 header files in this repo and change `CMakeList.txt` to use tbb hearder files in this repo instead of the one in `/usr/local/inlcude` 

Qt6 will also break the `make px4_sitl gazebo` command, because `init` is removed from `QStyleOption` since qt6, so a qt version requirement may need be documented if not documented already.

TSC21, 2021-06-05:
We'll have to wait for OSRF to fix their homebrew bottles.

daxaxelrod, 2021-06-19:
Same issue here, Tried @tenggyut's work around but didnt work for me

ryanjAA, 2021-07-06:
The fix for this (on a Mac and for the time being) is to remove the symlink for tbb so:

rm  /usr/local/include/tbb

Then add a new symlink to tbb@2020_u3:

ln -s /usr/local/Cellar/tbb@2020_u3/2020_U3/include/tbb /usr/local/include/tbb

Then sitl gazebo runs fine

jony4, 2021-07-08:
> The fix for this (on a Mac and for the time being) is to remove the symlink for tbb so:
> 
> rm /usr/local/include/tbb
> 
> Then add a new symlink to tbb@2020_u3:
> 
> ln -s /usr/local/Cellar/tbb@2020_u3/2020_U3/include/tbb /usr/local/include/tbb
> 
> Then sitl gazebo runs fine

In file included from /Users/jony4/dev/drone/PX4-Autopilot/Tools/sitl_gazebo/include/gazebo_user_camera_plugin.h:46:
/usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:46:11: error: no member named 'init' in 'QStyleOption'
      opt.init(this);
      ~~~ ^
1 error generated.
ninja: build stopped: subcommand failed.
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build
cd /Users/jony4/dev/drone/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.20.5/bin/cmake --build /Users/jony4/dev/drone/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 10
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

ryanjAA, 2021-07-08:
@jony4 that’s a different issue. One I’ve seen lately actually which iirc is quasi build environment specific (more to do with which version of dependencies the scripts install, hence the quasi). 

What version of PX4 are you trying to build?

There actually are some issues with the current versions of the toolchain setup and stable release (1.11.3) but not beta and master (1.12.0), reverting to older versions fixes it but must be done manually (and depending on your os version, aren’t really possible).

jony4, 2021-07-09:
thanks for your tips!

I change the px4's version to v1.11.3 and had a new error message:

 make px4_sitl gazebo

`
jony4@nq ~/d/d/PX4-Autopilot ((v1.11.3))> make px4_sitl gazebo
[65/404] Building CXX object src/modules/commander/CMakeFiles/modules__commander.dir/accelerometer_calibration.cpp.o
FAILED: src/modules/commander/CMakeFiles/modules__commander.dir/accelerometer_calibration.cpp.o
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"commander\" -DPX4_MAIN=commander_app_main -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/ecl -I../../src/modules/commander/Arming/PreFlightCheck -I../../src/modules/commander/Arming/ArmAuthorization -I../../src/modules/commander/Arming/HealthFlags -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.3.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wno-logical-op -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -std=gnu++14 -MD -MT src/modules/commander/CMakeFiles/modules__commander.dir/accelerometer_calibration.cpp.o -MF src/modules/commander/CMakeFiles/modules__commander.dir/accelerometer_calibration.cpp.o.d -o src/modules/commander/CMakeFiles/modules__commander.dir/accelerometer_calibration.cpp.o -c ../../src/modules/commander/accelerometer_calibration.cpp
/Users/jony4/dev/drone/PX4-Autopilot/src/modules/commander/accelerometer_calibration.cpp:166:15: fatal error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Wnon-c-typedef-for-linkage]
typedef struct  {
              ^
               accel_worker_data_t
/Users/jony4/dev/drone/PX4-Autopilot/src/modules/commander/accelerometer_calibration.cpp:167:31: note: type is not C-compatible due to this default member initializer
        orb_advert_t    *mavlink_log_pub{nullptr};
                                        ^~~~~~~~~
/Users/jony4/dev/drone/PX4-Autopilot/src/modules/commander/accelerometer_calibration.cpp:170:3: note: type is given name 'accel_worker_data_t' for linkage purposes by this typedef declaration
} accel_worker_data_t;
  ^
1 error generated.
[78/404] Building CXX object src/modules/simulator/CMakeFiles/modules__simulator.dir/simulator_mavlink.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
`

mattes-bru, 2021-07-14:
> The fix for this (on a Mac and for the time being) is to remove the symlink for tbb so:
> 
> rm /usr/local/include/tbb
> 
> Then add a new symlink to tbb@2020_u3:
> 
> ln -s /usr/local/Cellar/tbb@2020_u3/2020_U3/include/tbb /usr/local/include/tbb
> 
> Then sitl gazebo runs fine

I could archive the same result with following commands:

```bash
brew unlink tbb
brew link tbb@2020
```

ryanjAA, 2021-07-14:
> > The fix for this (on a Mac and for the time being) is to remove the symlink for tbb so:
> > rm /usr/local/include/tbb
> > Then add a new symlink to tbb@2020_u3:
> > ln -s /usr/local/Cellar/tbb@2020_u3/2020_U3/include/tbb /usr/local/include/tbb
> > Then sitl gazebo runs fine
> 
> I could archive the same result with following commands:
> 
> ```shell
> brew unlink tbb
> brew link tbb@2020
> ```

Yes. Does the same thing.

ryanjAA, 2021-08-12:
> thanks for your tips!
>
> I change the px4's version to v1.11.3 and had a new error message:
>
> make px4_sitl gazebo
>
> ```
>
> jony4@nq ~/d/d/PX4-Autopilot ((v1.11.3))> make px4_sitl gazebo
> [65/404] Building [...]

This is due to your Apple Clang version. Mostly likely you’re using something relatively new (Apple Clang 12.0-12.0.5). That will generate an error on PX4 1.11.3. On PX4 1.12.x it works fine. If you want to do sitl on 1.11.x on a Mac you need to use Apple clang version 11 which means you’ll need a Mac OS operating system around 10.15.3 and Command line tools for xcode 11.3. I tested this today for good measure and can confirm. 

It may be possible for PX4 1.11.x to work with the newest Apple clang for sitl but I haven’t been able to get it to work ever. I suppose you could always use docker but that would be much much slower if you are doing any dev work you want to see on gazebo given the time to compile via docker, etc.

richtong, 2021-09-09:
Thanks @hamishwillee I think you can close this one now :-)

hamishwillee, 2021-11-11:
@julianoes @richtong  So to workaround all this stuff we added the following commands to the instructions here: http://docs.px4.io/master/en/dev_setup/dev_env_mac.html#gazebo-simulation
```
brew unlink tbb
brew install tbb@2020
brew link tbb@2020
```
The note says that the commands should be removed when the issue is fixed. Is it actually fixed? Or do these commands need to remain in the docs forever?

LiamTal, 2023-04-16:
building for macOS (using macOS Ventura) I get :
Error: tbb@2020 has been disabled because it is not supported upstream!
cannot find a workaround

mkarklins, 2023-04-16:
Without the workaround and just using `brew install tbb` I get `px4/sitl/src/sitl_led.c:53:23: fatal error: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]` when running `make px4_sitl gazebo` though not sure if the error is related to this particular issue.

hamishwillee, 2023-05-24:
Thanks, so I guess "not fixed"

julianoes, 2023-05-26:
I haven't seen this problem in years :smile:

CalvinKoung, 2023-06-01:
> 

I encounter same problem, did you fix it ?

bousborne, 2023-06-02:
I also have this problem now. I haven't found a fix, but just to get more info out there from "brew edit tbb@2020" it says it is disabled recently:

disable! date: "2023-02-07", because: :unsupported

julianoes, 2023-06-02:
Right. So we're relying on an old version of tbb on macOS. Who wants to dig why that's a dependency and why it can't be updated? I'm happy to merge PRs but I can't do the archeology work myself without a Mac on my desk.

bousborne, 2023-06-02:
I ended up fixing it locally. I first removed any tbb related stuff (I don't think it actually matters, maybe just skip the brew install tbb stuff). Then, when I ran "make px4_sitl jmavsim" as per the isntructions, I was getting a bunch of errors related to -Wstrict-prototypes. So, I followed the errors and added in the voids, and commented out one variable that was assigned but not used, related to -Wassigned-but-not-used error. It now seems like it works (at least for building and running the jmavsim stuff). It ended up only being a few items. I apparently can’t attach a file with txt extension, so here is the pasted patch:

`
diff --git a/boards/px4/sitl/src/sitl_led.c b/boards/px4/sitl/src/sitl_led.c
index c9d3f4d053..f86e877764 100644
--- a/boards/px4/sitl/src/sitl_led.c
+++ b/boards/px4/sitl/src/sitl_led.c
@@ -50,7 +50,7 @@ __END_DECLS
 
 static bool _led_state[2] = { false, false };
 
-__EXPORT void led_init()
+__EXPORT void led_init(void)
 {
 	PX4_DEBUG("LED_INIT");
 }
diff --git a/platforms/common/work_queue/hrt_thread.c b/platforms/common/work_queue/hrt_thread.c
index 8438b0ae2c..8da8b415d1 100644
--- a/platforms/common/work_queue/hrt_thread.c
+++ b/platforms/common/work_queue/hrt_thread.c
@@ -113,7 +113,7 @@ static void _sighandler(int sig_num)
  *
  ****************************************************************************/
 
-static void hrt_work_process()
+static void hrt_work_process(void)
 {
 	struct wqueue_s *wqueue = &g_hrt_work;
 	volatile struct work_s *work;
diff --git a/platforms/common/work_queue/work_thread.c b/platforms/common/work_queue/work_thread.c
index b1e0e72355..5109b6ad8a 100644
--- a/platforms/common/work_queue/work_thread.c
+++ b/platforms/common/work_queue/work_thread.c
@@ -338,7 +338,7 @@ int work_usrthread(int argc, char *argv[])
 
 #endif /* CONFIG_SCHED_USRWORK */
 
-uint32_t clock_systimer()
+uint32_t clock_systimer(void)
 {
 	//printf("clock_systimer: %0lx\n", hrt_absolute_time());
 	return (0x00000000ffffffff & hrt_absolute_time());
diff --git a/platforms/posix/include/hrt_work.h b/platforms/posix/include/hrt_work.h
index 6dd2ab5a09..7799d3815c 100644
--- a/platforms/posix/include/hrt_work.h
+++ b/platforms/posix/include/hrt_work.h
@@ -48,13 +48,13 @@ int hrt_work_queue(struct work_s *work, worker_t worker, void *arg, uint32_t usd
 void hrt_work_cancel(struct work_s *work);
 
 static inline void hrt_work_lock(void);
-static inline void hrt_work_lock()
+static inline void hrt_work_lock(void)
 {
 	px4_sem_wait(&_hrt_work_lock);
 }
 
 static inline void hrt_work_unlock(void);
-static inline void hrt_work_unlock()
+static inline void hrt_work_unlock(void)
 {
 	px4_sem_post(&_hrt_work_lock);
 }
diff --git a/src/lib/rc/rc_tests/RCTest.cpp b/src/lib/rc/rc_tests/RCTest.cpp
index 07c45d33c0..3fa2cd41aa 100644
--- a/src/lib/rc/rc_tests/RCTest.cpp
+++ b/src/lib/rc/rc_tests/RCTest.cpp
@@ -359,7 +359,7 @@ bool RCTest::sbus2Test()
 	bool sbus_frame_drop;
 	uint16_t max_channels = sizeof(rc_values) / sizeof(rc_values[0]);
 
-	int rate_limiter = 0;
+//	int rate_limiter = 0;
 	unsigned last_drop = 0;
 
 	while (EOF != (ret = fscanf(fp, "%f,%x,,", &f, &x))) {
@@ -390,7 +390,7 @@ bool RCTest::sbus2Test()
 			last_drop = sbus_frame_drops + sbus_frame_resets;
 		}
 
-		rate_limiter++;
+//		rate_limiter++;
 	}
 
 	ut_test(ret == EOF);
`

Copy and paste that to a file called diff.patch. Then run:

git apply patchfile.patch

It should build then (hopefully!). 

I now think this is not really related to the tbb as I initially thought, I think this code is just old and was originally compiled with gcc version 8 or 9. Now, the newer versions are stricter on forcing warnings to errors. So, the tbb dependency may not be needed.

bousborne, 2023-07-04:
Workaround:

I actually found out that you can just comment out the "disable" line in the brew files. Do the following for the workaround:

`brew unlink tbb`
`brew edit tbb@2020`
Comment out the following line
`disable! date: "2023-02-07", because: :unsupported`
Save and exit back to cli.
`brew install tbb@2020`
`brew link tbb@2020`

julianoes, 2023-07-04:
@bousborne thanks for the note! Would it be possible to fix this with a PR? I'm not quite sure where the edit is required that you mention.

bousborne, 2023-07-10:
@julianoes, no problem! I created a pull request for the "tbb@2020 not installing" problem here: https://github.com/PX4/PX4-user_guide/pull/2617

That is actually a something that needs to be done in the setup, so that pull request went to the PX4-user_guide repo.

julianoes, 2023-07-11:
Thanks @bousborne, I guess we can now close this issue.

closed 2023-07-11 by julianoes

shivam5594, 2023-07-28:
@bousborne ditto!
```
  # Original deprecation date: 2020-04-02
  # Temporarily undeprecated from 2022-07-29 to 2022-11-07
  # disable! date: "2023-02-07", because: :unsupported
```

and it still fails with 
```
% brew install tbb@2020
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
==> Updated Homebrew from 4.0.28 (da01f8f0b) to 4.1.2 (3d44b9311).
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
bfs                            llm                            plog                           python-cryptography            tzdiff
govulncheck                    mvfst                          pop                            quictls
killport                       pgrok                          prettierd                      terragrunt-atlantis-config
==> New Casks
4k-video-downloaderplus        herd                           lm-studio                      ripx                           showmeyourhotkeys
fedistar                       keyclu                         replay                         sfm
Warning: Calling the `appcast` stanza is deprecated! Use the `livecheck` stanza instead.
Please report this issue to the adoptopenjdk/openjdk tap (not Homebrew/brew or Homebrew/homebrew-core), or even better, submit a PR to fix it:
  /usr/local/Homebrew/Library/Taps/adoptopenjdk/homebrew-openjdk/Casks/adoptopenjdk15.rb:9

You have 12 outdated formulae and 1 outdated cask installed.
Warning: Calling the `appcast` stanza is deprecated! Use the `livecheck` stanza instead.
Please report this issue to the adoptopenjdk/openjdk tap (not Homebrew/brew or Homebrew/homebrew-core), or even better, submit a PR to fix it:
  /usr/local/Homebrew/Library/Taps/adoptopenjdk/homebrew-openjdk/Casks/adoptopenjdk15.rb:9

Warning: Calling the `appcast` stanza is deprecated! Use the `livecheck` stanza instead.
Please report this issue to the adoptopenjdk/openjdk tap (not Homebrew/brew or Homebrew/homebrew-core), or even better, submit a PR to fix it:
  /usr/local/Homebrew/Library/Taps/adoptopenjdk/homebrew-openjdk/Casks/adoptopenjdk15.rb:9

The 4.1.0 release notes are available on the Homebrew Blog:
  https://brew.sh/blog/4.1.0
The 4.1.2 changelog can be found at:
  https://github.com/Homebrew/brew/releases/tag/4.1.2
Error: tbb@2020 has been disabled because it is not supported upstream!
```

ch-greams, 2023-08-02:
@shivam5594 try installing it like this:

```sh
HOMEBREW_NO_INSTALL_FROM_API=1 brew install tbb@2020 
```

bousborne, 2023-08-15:
@shivam5594 Did @ch-greams suggestion work? I think this is trying to install without fetching the latest from the developers pushes, but rather from local versions to the homebrew site (or something along those lines).

If that doesn't work, if any tbb was installed in the past, homebrew can sometimes keep a local cache of the homebrew formula, and will automatically used the cached version instead of the front facing version you edited.

Therefore, try to clear the homebrew cache, and reinstall like this:

`rm -rf $(brew --cache)/tbb@2020--*
brew install ./tbb@2020.rb
`

shivam5594, 2023-08-24:
@ch-greams thanks for that. it worked 👍

mengchaoheng, 2024-03-01:
on a new pc I have installed development environment, and run `make px4_sitl jmavsim` successful.

but when I checkout to an old version px4 (maybe v1.11.0 around), something wrong:
```Console
➜  PX4-Autopilot git:(jint) make px4_sitl jmavsim                  
-- PX4 version: v1.11.0-alpha1
-- PX4 config file: /Users/mch/Proj/Mac_DF/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- The C compiler identification is AppleClang 15.0.0.15000100
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.12", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-506-g99575d7
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default
[0/775] git submodule src/drivers/gps/devices
[3/775] Building C object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o
FAILED: boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o 
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/cc -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/boards/px4/sitl/src -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/platforms/posix/src/px4/common/include -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/src/lib -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/platforms/common/include -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/src -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/include -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/lib -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/lib/matrix -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/platforms/posix/include -I/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/external/Install/include -O2 -g -DNDEBUG -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-common -Wbad-function-cast -Wnested-externs -Wstrict-prototypes -std=gnu11 -MD -MT boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o -MF boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o.d -o boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o -c /Users/mch/Proj/Mac_DF/PX4-Autopilot/boards/px4/sitl/src/sitl_led.c
/Users/mch/Proj/Mac_DF/PX4-Autopilot/boards/px4/sitl/src/sitl_led.c:53:23: fatal error: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
__EXPORT void led_init()
                      ^
                       void
1 error generated.
[13/775] Generating git version header
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
➜  PX4-Autopilot git:(jint) 
```

mengchaoheng, 2024-03-01:
> on a new pc I have installed development environment, and run `make px4_sitl jmavsim` successful.
> 
> but when I checkout to an old version px4 (maybe v1.11.0 around), something wrong:

fix by [set MacOSX12.sdk](https://github.com/PX4/PX4-Autopilot/issues/20550#issuecomment-1317151899) and [set -Wno-deprecated-non-prototype](https://github.com/madler/zlib/issues/633#issuecomment-1507872127) on cmake/px4_add_common_flags.cmake中add_compile_options

mengchaoheng, 2024-03-02:
Warning: No available formula with the name "tbb@2020".

```Console 
 ~ brew unlink tbb

Error: No such keg: /usr/local/Cellar/tbb
➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb

sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
➜  ~ brew install tbb@2020

==> Downloading https://formulae.brew.sh/api/formula.jws.json
################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
################################################################################### 100.0%
Warning: No available formula with the name "tbb@2020".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for tbb@2020.
➜  ~ brew link tbb@2020

Error: No such keg: /usr/local/Cellar/tbb@2020
➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb

sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
➜  ~ rm -rf $(brew --cache)/tbb@2020--*

zsh: no matches found: /Users/mch/Library/Caches/Homebrew/tbb@2020--*
➜  ~ HOMEBREW_NO_INSTALL_FROM_API=1 brew install tbb@2020 
Running `brew update --auto-update`...
==> Tapping homebrew/core
Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-core'...
remote: Enumerating objects: 1888230, done.
remote: Counting objects: 100% (1404/1404), done.
remote: Compressing objects: 100% (574/574), done.
remote: Total 1888230 (delta 984), reused 1177 (delta 830), pack-reused 1886826
Receiving objects: 100% (1888230/1888230), 715.28 MiB | 3.63 MiB/s, done.
Resolving deltas: 100% (1344441/1344441), done.
Tapped 3 commands and 7022 formulae (7,370 files, 784.5MB).
==> Tapping homebrew/cask
Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask'...
remote: Enumerating objects: 897531, done.
remote: Counting objects: 100% (2082/2082), done.
remote: Compressing objects: 100% (850/850), done.
remote: Total 897531 (delta 1416), reused 1884 (delta 1231), pack-reused 895449
Receiving objects: 100% (897531/897531), 364.87 MiB | 3.52 MiB/s, done.
Resolving deltas: 100% (647163/647163), done.
Tapped 4397 casks (4,472 files, 393MB).
Warning: No available formula with the name "tbb@2020".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for tbb@2020.
➜  ~ brew install tbb@2020

==> Downloading https://formulae.brew.sh/api/formula.jws.json
-=O=#   #   #     #                                                                      
==> Downloading https://formulae.brew.sh/api/cask.jws.json
-=#=-#   #   #                                                                           
Warning: No available formula with the name "tbb@2020".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for tbb@2020.
➜  ~ brew unlink tbb

Error: No such keg: /usr/local/Cellar/tbb
➜  ~ brew edit tbb@2020

Warning: edit is a developer command, so Homebrew's
developer mode has been automatically turned on.
To turn developer mode off, run:
  brew developer off

Usage: brew edit [options] [formula|cask|tap ...]

Open a formula, cask or tap in the editor set by EDITOR or
HOMEBREW_EDITOR, or open the Homebrew repository for editing if no argument is
provided.

      --formula, --formulae        Treat all named arguments as formulae.
      --cask, --casks              Treat all named arguments as casks.
      --print-path                 Print the file path to be edited, without
                                   opening an editor.
  -d, --debug                      Display any debugging information.
  -q, --quiet                      Make some output more quiet.
  -v, --verbose                    Make some output more verbose.
  -h, --help                       Show this message.

Error: Invalid usage: tbb@2020 doesn't exist on disk.
Run brew create --set-name tbb@2020 $URL to create a new formula!
➜  ~ HOMEBREW_NO_INSTALL_FROM_API=1 brew install tbb@2020 
Running `brew update --auto-update`...
Warning: No available formula with the name "tbb@2020".
==> Searching for similarly named formulae and casks...
Error: No formulae or casks found for tbb@2020.
➜  ~ brew update
Already up-to-date.
➜  ~ brewupgrade
zsh: command not found: brewupgrade
➜  ~ brew upgrade
➜  ~ brew doctor
Your system is ready to brew.
➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb

sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
➜  ~ 
```

kadirozgun, 2024-03-21:
> Warning: No available formula with the name "tbb@2020".
> 
> ```
>  ~ brew unlink tbb
> 
> Error: No such keg: /usr/local/Cellar/tbb
> ➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb
> 
> sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
> ➜  ~ brew install tbb@2020
> 
> ==> Downloading https://formulae.brew.sh/api/formula.jws.json
> ################################################################################### 100.0%
> ==> Downloading https://formulae.brew.sh/api/cask.jws.json
> ################################################################################### 100.0%
> Warning: No available formula with the name "tbb@2020".
> ==> Searching for similarly named formulae and casks...
> Error: No formulae or casks found for tbb@2020.
> ➜  ~ brew link tbb@2020
> 
> Error: No such keg: /usr/local/Cellar/tbb@2020
> ➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb
> 
> sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
> ➜  ~ rm -rf $(brew --cache)/tbb@2020--*
> 
> zsh: no matches found: /Users/mch/Library/Caches/Homebrew/tbb@2020--*
> ➜  ~ HOMEBREW_NO_INSTALL_FROM_API=1 brew install tbb@2020 
> Running `brew update --auto-update`...
> ==> Tapping homebrew/core
> Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-core'...
> remote: Enumerating objects: 1888230, done.
> remote: Counting objects: 100% (1404/1404), done.
> remote: Compressing objects: 100% (574/574), done.
> remote: Total 1888230 (delta 984), reused 1177 (delta 830), pack-reused 1886826
> Receiving objects: 100% (1888230/1888230), 715.28 MiB | 3.63 MiB/s, done.
> Resolving deltas: 100% (1344441/1344441), done.
> Tapped 3 commands and 7022 formulae (7,370 files, 784.5MB).
> ==> Tapping homebrew/cask
> Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask'...
> remote: Enumerating objects: 897531, done.
> remote: Counting objects: 100% (2082/2082), done.
> remote: Compressing objects: 100% (850/850), done.
> remote: Total 897531 (delta 1416), reused 1884 (delta 1231), pack-reused 895449
> Receiving objects: 100% (897531/897531), 364.87 MiB | 3.52 MiB/s, done.
> Resolving deltas: 100% (647163/647163), done.
> Tapped 4397 casks (4,472 files, 393MB).
> Warning: No available formula with the name "tbb@2020".
> ==> Searching for similarly named formulae and casks...
> Error: No formulae or casks found for tbb@2020.
> ➜  ~ brew install tbb@2020
> 
> ==> Downloading https://formulae.brew.sh/api/formula.jws.json
> -=O=#   #   #     #                                                                      
> ==> Downloading https://formulae.brew.sh/api/cask.jws.json
> -=#=-#   #   #                                                                           
> Warning: No available formula with the name "tbb@2020".
> ==> Searching for similarly named formulae and casks...
> Error: No formulae or casks found for tbb@2020.
> ➜  ~ brew unlink tbb
> 
> Error: No such keg: /usr/local/Cellar/tbb
> ➜  ~ brew edit tbb@2020
> 
> 
> Warning: edit is a developer command, so Homebrew's
> developer mode has been automatically turned on.
> To turn developer mode off, run:
>   brew developer off
> 
> Usage: brew edit [options] [formula|cask|tap ...]
> 
> Open a formula, cask or tap in the editor set by EDITOR or
> HOMEBREW_EDITOR, or open the Homebrew repository for editing if no argument is
> provided.
> 
>       --formula, --formulae        Treat all named arguments as formulae.
>       --cask, --casks              Treat all named arguments as casks.
>       --print-path                 Print the file path to be edited, without
>                                    opening an editor.
>   -d, --debug                      Display any debugging information.
>   -q, --quiet                      Make some output more quiet.
>   -v, --verbose                    Make some output more verbose.
>   -h, --help                       Show this message.
> 
> Error: Invalid usage: tbb@2020 doesn't exist on disk.
> Run brew create --set-name tbb@2020 $URL to create a new formula!
> ➜  ~ HOMEBREW_NO_INSTALL_FROM_API=1 brew install tbb@2020 
> Running `brew update --auto-update`...
> Warning: No available formula with the name "tbb@2020".
> ==> Searching for similarly named formulae and casks...
> Error: No formulae or casks found for tbb@2020.
> ➜  ~ brew update
> Already up-to-date.
> ➜  ~ brewupgrade
> zsh: command not found: brewupgrade
> ➜  ~ brew upgrade
> ➜  ~ brew doctor
> Your system is ready to brew.
> ➜  ~ sed -i.bak '/disable! date:/s/^/  /; /disable! date:/s/./#/3' $(brew --prefix)/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb
> 
> sed: /usr/local/Library/Taps/homebrew/homebrew-core/Formula/tbb@2020.rb: No such file or directory
> ➜  ~ 
> ```

I also have same problem on Apple Silicon M1 Max with X86 build Homebrew

Daywithck, 2024-08-17:
> I also have same problem on Apple Silicon M1 Max with X86 build Homebrew

I am seeing similar error on Apple Silicon M2. Any work around or fix?