PX4/PX4-Autopilot #17646: Battery Percentage Incorrect for 12S

DoppleGangster, 2021-05-25:
I'm running a 12S battery configuration and my battery percentage scale does not appear to function correctly. I've tested the 6S setting and I get the correct scaling. I'm currently using the basic battery configuration.

When set to 12S the the high and low battery voltage appear correct however the percentage stays at 100% while the measured voltage continues to drop.

[image]
[image]

Thanks!

dagar, 2021-05-29:
Can you share a log? https://logs.px4.io

DoppleGangster, 2021-06-08:
Sure:
[Flight Lig](https://logs.px4.io/plot_app?log=b9a79237-9d6d-46b8-8040-b7dde50e17bc)

I have my low voltage and high voltage range adjusted so that it reports the battery percentage correctly.