PX4/PX4-Autopilot #17648: Quaternion normalization doesn't normalize under all conditions

Etruscian, 2021-05-26:
**Describe the bug**
Under certain conditions, the quaternion normalize function returns a quaternion with a norm larger than 1. See #17310 and similar issues.

**Expected behavior**
The normalize function returns a normalized quaternion with a norm of 1.

This is now solved everywhere with workarounds, but it is a potential risk for people that do not know that this can happen.

bresch, 2021-05-26:
Hi @Etruscian , 

I assume you're talking about this function: https://github.com/PX4/PX4-Matrix/blob/b8568a89db8455d0ab2e2e391e2149ba2e5e10dd/matrix/Vector.hpp#L100-L102

Could you please provide an example showing this function returning a quaternion with a norm larger than 1?

Etruscian, 2021-05-26:
Hi @bresch,

That is indeed the correct function. Examples of quaternions that fail are:
0.5419441               0               0      -0.8404146
0.5846435               0               0      -0.8112904

Both of these return NaN in the tilt calculation if the clamping is removed, indicating that the value going in the acos does not lay between -1 and 1. https://github.com/PX4/PX4-Autopilot/blob/b264577daaed7c28efc16475568378aef43114c0/src/modules/vtol_att_control/tailsitter.cpp#L228

When I add the quaternions to the unit tests however, they do return a norm of 1.0f. My own floating point knowledge isn't good enough to know why this could be happening, maybe @nidhishraj could shine a better light on when/why it fails.

closed 2025-12-01 by github-actions[bot]