PX4/PX4-Autopilot #17652: Crossfire Micro RX (CRSF) on CUAV V5+ with most recent stable firmware

drtrigon, 2021-05-26:
**Describe the bug**
Today I set-up a CUAV V5+ with most recent stable firmware on our hexacopter. Then I connected a Crossfire Micro RX to it in order to remote control the model and get telemetry data for GCS as well. Followed: https://oscarliang.com/crossfire-betaflight/

The issue here is that when trying to calibrate radio I get “0 channels detected. PixHawk needs at least 5”. But it starts working when I go over to the MAVLinkConsole and enter: `rc_input start -d /dev/ttyS3` (the RX is on UART4)

So why does the firmware not detect it during boot and start rc_input??

(May be another observation is that it does not detect telemetry - that would then be my next question as soon as this bug/issue is solved... ;)

**To Reproduce**
Steps to reproduce the behavior - I don't know if you can not having my hardware (?):
1. Use CUAV V5+ with most recnet firmware
2. Configure frame (hex x)
3. Connect Crossfire Micro RX Ch1+2 to UART4 RX+TX
4. Start the FC and connect via USB with QGC or MissionPlanner
5. See error (and have fun)

**Expected behavior**
It should just work as I can conclude from the sparse docu about this topic.

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- Hex X config - user built frame
- CUAV V5+ with GPS, BEC and Crossfire Micro RX on UART4

**Additional context**
This is part of a research project and currently we are moving from the aged pixhawk v1 to this v5 FMU.

bkueng, 2021-05-28:
The default RC port on that board goes through the IO, which does not enable CRSF. You're right, could be mentioned in the docs.
The port isn't configurable there either, so `extras.txt` is a viable option (although indeed not the nicest). If you're on latest master/beta you should also have telemetry.

drtrigon, 2021-05-29:
Thanks a lot for the clarification! So would it be possible to change that? Introduce a parameter that allows to enable it?

I also browsed the sources a bit and look at various things. I found from the docs that the RC/DSM port is a serial port too `/dev/ttyS4` (but a soft serial one which does not work for crsf as I understand). Or some boards use `/dev/ttyS5` which would be the debug port here. So actually I would be fine using the debug port for that!

I am considering now to use MAVLink instead would that work on this board? Including telemetry? What port?

(However the preferred solution would be to make CRSF work...)

bkueng, 2021-05-31:
> So would it be possible to change that?

We already have boards that expose a configurable RC port. See https://github.com/PX4/PX4-Autopilot/blob/master/boards/holybro/kakutef7/default.cmake#L16 for example. V5 is not among them as it has a dedicated RC port and you're the first one who wants to use another port. Although possible, I'm a bit hesitant in making that configurable on V5 as it adds complexity, particularly in combination with IO.

> I am considering now to use MAVLink instead would that work on this board? Including telemetry? What port?

I've never tried that combination, you can test it. You'd just have to enable and configure mavlink on any of the serial ports.

drtrigon, 2021-05-31:
Basically what I am asking is whether it is possible to add the option to use CSRF with CUAV v5+?

(I will test MAVLink as alternative as soon as I have access to the system again)

drtrigon, 2021-06-09:
MAVLink works (no need for additional SBUS); config 2 pins to be MAVLink RX and TX on the receiver and an UART on the FC to be MAVLink (e.g. UART4).

What you get is telemetry on OpenTX and a bluetooth link to PC, Tablet or Phone with telemetry also.

Confer: https://m.youtube.com/watch?v=b55rdaDCKaM

(The question now is what is the advantage of CSRF over MAVLink? Speed? Does it have additional features?)

closed 2025-12-01 by github-actions[bot]