PX4/PX4-Autopilot #17660: Generator for UORB message docs

hamishwillee, 2021-05-27:
We need a generator to create docs of UOR message definitions for inclusion in the PX4 User Guide.

The expected end result in a new folder in the docs (**en/uorb_msg/**) that contains a README.md and a **.md** file for every **.msg** file in [/msg](https://github.com/PX4/PX4-Autopilot/tree/master/msg).
- The readme will simply be a list of links to the the contained ".msg" files in alphabetical order - link text is the name of the msg file: [/msg](https://github.com/PX4/PX4-Autopilot/tree/master/msg). For example
   ```
   # UORB Messages

  [actuator_armed.msg](actuator_armed.md)
  [actuator_controls.msg](actuator_controls.md)
  ...
  [wind.msg](wind.md)
  [yaw_estimator_status.msg](yaw_estimator_status.md)
- The individual msg file markdown files will initially/minimally include: a link to the file in source, and the content of the raw files in a code block. Note below I show this as a py code block to highlight the comments as comments. So the **actuator_armed.md** might look like this:
  <pre>
  # actuator_armed (UORB msg)
  
  source file: [/msg/actuator_armed.msg](https://github.com/PX4/PX4-Autopilot/blob/master/msg/actuator_armed.msg)

  ```py
  uint64 timestamp	# time since system start (microseconds)

  bool armed		# Set to true if system is armed
  bool prearmed		# Set to true if the actuator safety is disabled but motors are not armed
  bool ready_to_arm	# Set to true if system is ready to be armed
  bool lockdown		# Set to true if actuators are forced to being disabled (due to emergency or HIL)
  bool manual_lockdown    # Set to true if manual throttle kill switch is engaged
  bool force_failsafe	# Set to true if the actuators are forced to the failsafe position
  bool in_esc_calibration_mode # IO/FMU should ignore messages from the actuator controls topics
  bool soft_stop # Set to true if we need to ESCs to remove the idle constraint
  ```
  </pre>

Note:
- There is some discussion that we might need to modify this to add a description. My thinking is we could do this as comments - e.g. something like:
  ```
  #Armed state: track and set armed state
  bool armed		# Set to true if system is armed
  ...
  ```
  or, if we think there might be having other metadata in future?
  ```
  # @description Armed state: track and set armed state
  # @rtsp True   //i.e. to indicate that this is published for rtsp by default
  # @ros2 True //ie to indicate that is s published for ROS2
  bool armed		# Set to true if system is armed
  ...
  ```
- And we might need to further play with the rendering - ie have a table that splits on the comments marker.
- My point here is that I am not sure whether or not it is worth building in the capability for that parsing in first version.

@bkueng Is this something you could do for me? If not, I am happy to start an implementation. However can you point me to the model you'd preferred used? I.e. we could just have a basic parser in https://github.com/PX4/PX4-Autopilot/tree/master/msg/tools that creates the required folder to a sub page. However it seems to me that copying the airframe parsing stuff might make this more flexible long term, and allow us to hook into jenkins more easily.

Either way, if I write the parser, I would need your help hooking into CI. Please advise.

mrpollo, 2021-05-28:
Hey @hamishwillee does it really make sense to have a file per message? wouldn't it be better to render a single table with all the messages? or to have them all render on a single page?

Regarding the parsing, I would prefer we re-use as much code as possible, if there's any code we already have to parse them using python, I would suggest we split that into a base class we can re-use in this tool, or a similar approach.

TSC21, 2021-05-28:
@mrpollo the idea is to present a similar format to what the ROS message specs provide.

dagar, 2021-05-29:
How about starting with exactly what ROS uses? http://wiki.ros.org/rosdoc_lite

mkeyno, 2024-07-03:
@hamishwillee , may I ask why PX4 programmer decided  to include file such as **PX4-Autopilot\msg\VehicleCommandAck.msg**  in the following format 

` 
#include <uORB/topics/vehicle_command_ack.h>
 

uORB::Publication<vehicle_command_ack_s>        _command_ack_pub{ORB_ID(vehicle_command_ack)};

`