PX4/PX4-Autopilot #17663: How to debug the PX4 firmware in ubuntu while running px4 sitl

rkv006, 2021-05-27:
I want to debug the px4 firmware using GDB in Linux(ubuntu) 
I followed as same as in the documentation 
https://dev.px4.io/v1.11_noredirect/en/debug/simulation_debugging.html

and I started simulation using the following command 
$ make  px4_sitl_default jmavsim___gdb
it was stopped, till here its fine. 
now I want to put a breakpoint in any file like suppose pid.cpp
(gdb) b pid.cpp:125
I have given like this 
but its not going to that line 
how can solve this, give instructions to do proper debugging of PX4 firmware

kistlin, 2021-06-01:
This happened to me too but with a different setup. A breakpoint was not even necessary.
According to https://gnobal.net/591/what-is-sig32-during-a-gdb-debug-session  this can happen.
After that error you described, I had some more. I put the following lines into `.gdbinit`.
```
handle SIGCONT nostop noprint pass
handle SIGPIPE nostop noprint pass
handle SIG32 nostop noprint pass
```
After that the debug session ran as usual.
As long as you are not debugging threading issues, it should be alright I guess :).

rkv006, 2021-06-02:
> This happened to me too but with a different setup. A breakpoint was not even necessary.
> According to https://gnobal.net/591/what-is-sig32-during-a-gdb-debug-session this can happen.
> After that error you described, I had some more. I put the following lines into `.gdbinit`.
> 
> ```
> handle SIGCONT nostop noprint pass
> handle SIGPIPE nostop noprint pass
> handle SIG32 nostop noprint pass
> ```
> 
> After that the debug session ran as usual.
> As long as you are not debugging threading issues, it should be alright I guess :).
> 
> Maybe this helps?
> And it would be interesting to know why it happens. But I didn't take the time to search yet. So if someone knows :).

Thanks @kistlin for your reply, as long as I'm not putting any breakpoints in px4 debugging, its working fine.
and I'm facing issue while debugging the PX4 SITL,
I have done the Ardupilot SITL debugging but I didn't faced this issue
can you please specify where you have used the above mentioned lines   
you mentioned .gdbinit is it in (gdb)? or any specific file?

kistlin, 2021-06-02:
Yes `.gdbinit` is a file. Here is the man page https://man7.org/linux/man-pages/man5/gdbinit.5.html.
Under description  it says
```
 ~/.gdbinit
           User initialization file.  It is executed unless user
           specified GDB options "-nx", "-n" or "-nh".

 ./.gdbinit
           Initialization file for current directory.  It may need to be
           enabled with GDB security command "set auto-load
           local-gdbinit".
```
So if you create that file in your home directory with mentioned content, it should be applied during debugging. At least in my case it did.
Be aware that this gets applied every time you debug now. If you have other projects, you may want to use ` ./.gdbinit` next to your executable.
It should also be possible to enter the lines manually in `gdb` before you start debugging.

rkv006, 2021-06-02:
> Yes `.gdbinit` is a file. Here is the man page https://man7.org/linux/man-pages/man5/gdbinit.5.html.
> Under description it says
> 
> ```
>  ~/.gdbinit
>            User initialization file.  It is executed unless user
>            specified GDB options "-nx", "-n" or "-nh".
> 
>  ./.gdbinit
>            Initialization file for current directory.  It may need to be
>            enabled with GDB security command "set auto-load
>            local-gdbinit".
> ```
> 
> So if you create that file in your home directory with mentioned content, it should be applied during debugging. At least in my case it did.
> Be aware that this gets applied every time you debug now. If you have other projects, you may want to use ` ./.gdbinit` next to your executable.
> It should also be possible to enter the lines manually in `gdb` before you start debugging.
> 
> `gdb` also has an option to read commands from a file. https://linux.die.net/man/1/gdb
> Under Options you will find
> 
> ```
> All the options and command line arguments you give are processed in sequential order.
> The order makes a difference when the '-x' option is used.
> 
> ...
> 
> -x file
> Execute GDB commands from file file.
> ```
> 
> But maybe you start with putting it in a file in your home directory.

Thanks @kistlin for your info, still I'm not able to examine the pid.cpp (or) ekf2.cpp file, can you guess what's the reason behind this issue I was unable to locate what's the issue. if you can give any other suggestions I would be happy for that

kistlin, 2021-06-02:
When you run `make px4_sitl_default jmavsim___gdb` do you see the output
```
...

Reading symbols from /home/<user>/PX4-Autopilot/build/px4_sitl_default/bin/px4...
Buildfile: /home/<user>/PX4-Autopilot/Tools/jMAVSim/build.xml
warning: File "/home/<user>/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
        add-auto-load-safe-path /home/<user>/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/.gdbinit
line to your configuration file "/home/<user>/.gdbinit".
To completely disable this security protection add
        set auto-load safe-path /
line to your configuration file "/home/<user>/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
        info "(gdb)Auto-loading safe path"
...
```
This meas gdb will not load the `.gdbinit` already present at `"/home/<user>/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/.gdbinit"` due to security mechanisms.

Now what could help in your case is put the lines
```
handle SIGCONT nostop noprint pass
handle SIGPIPE nostop noprint pass
handle SIG32 nostop noprint pass
run
```
in the already existing file `PX4-Autopilot/build/Tools/posix.gdbinit` (clear the other lines in the file first)
A script `PX4-Autopilot/Tools/sitl_run.sh` will copy `posix.gdbinit` and replace the one at `/home/<user>/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/.gdbinit`. It is part of the make command, no need to execute it yourself.

Then put (change the path to where your repository is)
```
add-auto-load-safe-path /home/<user>/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/.gdbinit
```
into
```
/home/<user>/.gdbinit
```
and remove all other lines that you may have added.

After that try `make px4_sitl_default jmavsim___gdb` again.

rkv006, 2021-06-03:
> When you run `make px4_sitl_default jmavsim___gdb` do you see the output
>
> ```
> ...
>
> Reading symbols from /home/<user>/PX4-Autopilot/build/px4_sitl_default/bin/px4...
> Buildfile: [...]

well @kistlin that's a great guess!, now I'm able to examine ekf.cpp but not pid.cpp while I'm trying to put the breakpoint it was going to simulation mode and it's awaiting for commands(like commander takeoff, commander land), problem is not hitting the breakpoint in pid.cpp library. any solutions?

kistlin, 2021-06-03:
Well good to hear that you got further. But for that problem you need someone with more insight, sorry.
Not hitting the breakpoint could have several reasons.
- You are already past the break point
- The code is maybe not executed during standby (if you not have sent a command to actually fly or whatever)
- The code is not used (unlikely but could be)

rkv006, 2021-06-03:
> Well good to hear that you got further. But for that problem you need someone with more insight, sorry.
> Not hitting the breakpoint could have several reasons.
>
> * You are already past the break point [...]
Yes @kistlin I'm also suspecting the same, I also need more exposure on this with the help of someone like you, let me look into it. and thanks for your time! have a great day.

rkv006, 2021-06-07:
>     * You are already past the break point

Greetings!, i have been checking repeatedly. what i found is px4 is not running PID code while doing SITL simulation that's why I'm not able to debug it(it is in libraries)? might be wrong? , i want to understand the PX4-Autopilot Copter code flow can any one give any idea about it, I'm familiar with Ardupilot

rkv006, 2021-06-07:
> well @kistlin that's a great guess!, now I'm able to examine ekf.cpp but not pid.cpp while I'm trying to put the breakpoint it was going to simulation mode and it's awaiting for commands(like [...]

 sh px4sil_jmavgdb.sh 
/home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default
[1/1] Cleaning all built files...
Cleaning... 1132 files.
/home/rakeshvangaveti/PX4-Autopilot/build/px4_fmu-v5_multicopter
[1/1] Cleaning all built files...
Cleaning... 0 files.
/home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_test
[1/1] Cleaning all built files...
Cleaning... 0 files.
/home/rakeshvangaveti/PX4-Autopilot/build/px4_fmu-v5_default
[1/1] Cleaning all built files...
Cleaning... 0 files.
Entering 'Tools/flightgear_bridge'
Entering 'Tools/jMAVSim'
Removing out/
Entering 'Tools/jsbsim_bridge'
Entering 'Tools/sitl_gazebo'
Entering 'boards/atlflight/cmake_hexagon'
Entering 'mavlink/include/mavlink/v2.0'
Entering 'platforms/nuttx/NuttX/apps'
Entering 'platforms/nuttx/NuttX/nuttx'
Entering 'platforms/qurt/dspal'
Entering 'src/drivers/gps/devices'
Entering 'src/drivers/uavcan/libuavcan'
Entering 'src/drivers/uavcan_v1/legacy_data_types'
Entering 'src/drivers/uavcan_v1/libcanard'
Entering 'src/drivers/uavcan_v1/public_regulated_data_types'
Entering 'src/drivers/uavcannode_gps_demo/libcanard'
Entering 'src/drivers/uavcannode_gps_demo/public_regulated_data_types'
Entering 'src/lib/ecl'
Entering 'src/lib/matrix'
Entering 'src/modules/micrortps_bridge/micro-CDR'
[3/812] git submodule src/drivers/gps/devices
[4/812] git submodule src/lib/ecl
[5/812] git submodule mavlink/include/mavlink/v2.0
[246/812] git submodule Tools/jMAVSim
[811/812] cd /home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default/tmp && /home/rakeshvangaveti... none /home/rakeshvangaveti/PX4-Autopilot /home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: gdb
program: jmavsim
model: none
world: none
src_path: /home/rakeshvangaveti/PX4-Autopilot
build_path: /home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/rakeshvangaveti/PX4-Autopilot"/test_data
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
/home/rakeshvangaveti/.gdbinit:3: Error in sourced command file:
No executable file specified.
Use the "file" or "exec-file" command.
Reading symbols from /home/rakeshvangaveti/PX4-Autopilot/build/px4_sitl_default/bin/px4...
[Thread debugging using libthread_db enabled]
[... 870 lines cut ...]
[Thread 0x7fffe77fe700 (LWP 11407) exited]
[New Thread 0x7fffe77fe700 (LWP 11409)]
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
[New Thread 0x7fffe67fc700 (LWP 11411)]
[Thread 0x7fffe77fe700 (LWP 11409) exited]
[Thread 0x7fffe67fc700 (LWP 11411) exited]
[New Thread 0x7fffe67fc700 (LWP 11413)]
[Thread 0x7fffe67fc700 (LWP 11413) exited]
[New Thread 0x7fffe67fc700 (LWP 11415)]
[Thread 0x7fffe67fc700 (LWP 11415) exited]
[New Thread 0x7fffe67fc700 (LWP 11418)]
[New Thread 0x7fffe77fe700 (LWP 11419)]
[Thread 0x7fffe77fe700 (LWP 11419) exited]
--Type <RET> for more, q to quit, c to continue without paging--c
[Switching to Thread 0x7fffe67fc700 (LWP 11418)]

Thread 80 "px4" hit Breakpoint 2, EKF2::EKF2 (this=0x7fffd00021b0, multi_mode=true, config=..., replay_mode=false) at ../../src/modules/ekf2/EKF2.cpp:48
48	EKF2::EKF2(bool multi_mode, const px4::wq_config_t &config, bool replay_mode):
(gdb) s
[New Thread 0x7fffe77fe700 (LWP 11430)]
[Thread 0x7fffe77fe700 (LWP 11430) exited]
164		_param_ekf2_gsf_tas_default(_params->EKFGSF_tas_default)
(gdb) 

when im trying to examine EKF2 it was working fine(able to examine step by step), but if im trying to pid then it was not even going inside....

kistlin, 2021-06-07:
> Greetings!, i have been checking repeatedly. what i found is px4 is not running PID code while doing SITL simulation that's why I'm not able to debug it(it is in libraries)? might be wrong? , i want [...]

Did you already watch the Youtube videos that are put out? There they give an overview.
[Overview of multicopter control from sensors to motors — PX4 Developer Summit Virtual 2020](https://youtu.be/orvng_11ngQ)
[PX4 Flight Task Architecture Overview - Dennis Mannhart, Matthias Grob - PX4 Developer Summit 2019](https://youtu.be/-dkQG8YLffc)

rkv006, 2021-06-07:
> Did you already watch the Youtube videos that are put out? There they give an overview.
> [Overview of multicopter control from sensors to motors — PX4 Developer Summit Virtual [...]

Thanks @kistlin this videos and those slides helped me to understand flow of PX4 Autopilot code, It has given to me some clarifications, i think we can debug the modules like ekf, sensors, pos_control, att_control not the libraries, now im able to examine the mc_position_control.cpp 

#PX4DevloperSummit2020 slides:
https://static.sched.com/hosted_files/px4developersummitvirtual2020/1b/PX4%20Developer%20Summit%202020%20-%20Overview%20of%20multicopter%20control%20from%20sensors%20to%20motors.pdf