PX4/PX4-Autopilot #17669: #Takeoff airspeed reached issue

SamLin55587, 2021-05-28:
When I attempt to fly the planes,they start to out of control.
like following video
I am not sure whether I set the parameters or  launch file wrong.
Please tell me what I can do to fix the problem.
Sincerely appreciate.

https://user-images.githubusercontent.com/83861951/119969473-2ee15c00-bfe1-11eb-8864-0213adeb2dc1.mp4

dagar, 2021-05-29:
Which version of PX4 is this? Can you share the flight log? https://logs.px4.io

SamLin55587, 2021-05-29:
My px4 version is 1.11.0 dev.  But I don't know how to open my flight log

SamLin55587, 2021-05-29:
And this is my flight report in QGC 
[image]
[image]

SamLin55587, 2021-05-29:
This is my flight code in terminal

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=1030
process[uav0/plane_0_spawn-5]: started with pid [13287]
INFO  [param] selected parameter default file eeprom/parameters_1030
[param] Loaded: eeprom/parameters_1030
process[uav0/mavros-6]: started with pid [13294]
 parameter default: BAT_N_CELLS 0 -> 4
 parameter default: CBRK_SUPPLY_CHK 0 -> 894281
 parameter default: COM_CPU_MAX 90.00 -> -1.00
 parameter default: COM_RC_IN_MODE 0 -> 1
 parameter default: EKF2_REQ_GPS_H 10.00 -> 0.50
 parameter default: EKF2_MULTI_IMU 0 -> 3
 parameter default: SENS_IMU_MODE 1 -> 0
 parameter default: EKF2_MULTI_MAG 0 -> 2
 parameter default: SENS_MAG_MODE 1 -> 0
 parameter default: SDLOG_MODE 0 -> 1
 parameter default: SDLOG_PROFILE 1 -> 131
 parameter default: SDLOG_DIRS_MAX 0 -> 7
 parameter default: TRIG_INTERFACE 4 -> 3
 parameter default: COM_POS_FS_DELAY 1 -> 5
 parameter default: COM_POS_FS_EPH 5.00 -> 15.00
 parameter default: COM_POS_FS_EPV 10.00 -> 30.00
 parameter default: COM_POS_FS_GAIN 10 -> 0
 parameter default: COM_POS_FS_PROB 30 -> 1
 parameter default: COM_VEL_FS_EVH 1.00 -> 5.00
 parameter default: EKF2_ARSP_THR 0.00 -> 8.00
 parameter default: EKF2_FUSE_BETA 0 -> 1
 parameter default: EKF2_GPS_CHECK 245 -> 21
 parameter default: EKF2_MAG_ACCLIM 0.50 -> 0.00
 parameter default: EKF2_MAG_YAWLIM 0.25 -> 0.00
 parameter default: EKF2_REQ_EPH 3.00 -> 10.00
 parameter default: EKF2_REQ_EPV 5.00 -> 10.00
 parameter default: EKF2_REQ_HDRIFT 0.10 -> 0.50
 parameter default: EKF2_REQ_SACC 0.50 -> 1.00
 parameter default: EKF2_REQ_VDRIFT 0.20 -> 1.00
 parameter default: RTL_TYPE 0 -> 1
 parameter default: RTL_RETURN_ALT 60.00 -> 100.00
 parameter default: RTL_DESCEND_ALT 30.00 -> 100.00
 parameter default: RTL_LAND_DELAY 0.00 -> -1.00
 parameter default: MIS_DIST_WPS 900.00 -> 5000.00
 parameter default: MIS_LTRMIN_ALT -1.00 -> 25.00
 parameter default: MIS_TAKEOFF_ALT 2.50 -> 25.00
 parameter default: PWM_MAIN_RATE 400 -> 50
 parameter default: GPS_UBX_DYNMODEL 7 -> 8
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[ INFO] [1622264892.157033951]: FCU URL: udp://:14540@localhost:14550
[ INFO] [1622264892.158742991]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1622264892.158813795]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622264892.159001564]: GCS bridge disabled
[ INFO] [1622264892.167181739]: Plugin 3dr_radio loaded
[ INFO] [1622264892.168141081]: Plugin 3dr_radio initialized
[ INFO] [1622264892.168208820]: Plugin actuator_control loaded
[ INFO] [1622264892.169668657]: Plugin actuator_control initialized
[ INFO] [1622264892.172148969]: Plugin adsb loaded
[ INFO] [1622264892.173610151]: Plugin adsb initialized
[ INFO] [1622264892.173686958]: Plugin altitude loaded
[ INFO] [1622264892.174162558]: Plugin altitude initialized
[ INFO] [1622264892.174220869]: Plugin cam_imu_sync loaded
[ INFO] [1622264892.174543731]: Plugin cam_imu_sync initialized
[ INFO] [1622264892.174623421]: Plugin command loaded
[ INFO] [1622264892.177649760]: Plugin command initialized
[ INFO] [1622264892.177731060]: Plugin companion_process_status loaded
[ INFO] [1622264892.178671816]: Plugin companion_process_status initialized
[ INFO] [1622264892.178732830]: Plugin debug_value loaded
[ INFO] [1622264892.180590025]: Plugin debug_value initialized
[... 1802 lines cut ...]
[ INFO] [1622265222.712366965, 79.216000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265223.710204956, 79.524000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265223.713147672, 79.528000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265224.712822777, 79.836000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265224.713360763, 79.840000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265225.711779681, 80.088000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265225.753918859, 80.092000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265226.714265979, 80.344000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265226.800582998, 80.348000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265227.709425914, 80.420000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265227.756896343, 80.420000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265227.865964739, 80.432000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265227.873823154, 80.440000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265228.716566046, 80.608000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265228.731989324, 80.612000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265228.870629882, 80.648000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265228.872272232, 80.652000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265229.464848522, 80.852000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265229.467585030, 80.856000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265229.707287490, 80.916000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265229.729927497, 80.920000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1622265230.708447657, 81.308000000]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1622265230.710791628, 81.312000000]: udp0: Remote address: 127.0.0.1:14580
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to baro