PX4/PX4-Autopilot #17680: PX4-Autpilot SITL : Build fails with no proper error message

adbidwai, 2021-05-31:
**System OS** : Ubuntu 18.04 dual booted alongside Windows
**ROS Version** : ROS Melodic

I am trying to set up PX4 SITL in Gazebo simulator. I am following [PX4's official guide](https://docs.px4.io/master/en/simulation/gazebo.html)

I first built MAVLink and MAVROS successfully, then I downloaded the PX4-Autopilot Firmware package in the ```src``` folder of my workspace and tried to build it using ```make px4_sitl_default_gazebo```

After that I am sourcing my packages and trying to build everything using ```catkin build```, I get the following issue : 
```
Warnings   << px4:check /home/vaibhav/sop_ws/logs/px4/build.check.000.log                                                                     
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:41 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:48 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

cd /home/vaibhav/sop_ws/build/px4; catkin build --get-env px4 | catkin env -si  /usr/bin/make cmake_check_build_system; cd -
..............................................................................................................................................
______________________________________________________________________________________________________________________________________________
Errors     << px4:make /home/vaibhav/sop_ws/logs/px4/build.make.001.log                                                                       
  File "/home/vaibhav/sop_ws/src/PX4-Autopilot/src/lib/version/px_update_git_header.py", line 119
    """
      ^
SyntaxError: invalid syntax
make[2]: *** [src/lib/version/build_git_version.h] Error 1
make[1]: *** [src/lib/version/CMakeFiles/ver_gen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [uORB/topics/uORBTopics.hpp] Error 1
make[1]: *** [msg/CMakeFiles/uorb_headers.dir/all] Error 2
Traceback (most recent call last):
  File "/home/vaibhav/sop_ws/src/PX4-Autopilot/src/lib/parameters/px_process_params.py", line 50, in <module>
    from px4params import srcscanner, srcparser, injectxmlparams, xmlout, markdownout, jsonout
  File "/home/vaibhav/sop_ws/src/PX4-Autopilot/src/lib/parameters/px4params/jsonout.py", line 35
    def get_typed_value(value: str, type_name: str):
                             ^
SyntaxError: invalid syntax
make[2]: *** [parameters.xml] Error 1
make[1]: *** [src/lib/parameters/CMakeFiles/parameters_xml.dir/all] Error 2
make: *** [all] Error 2
cd /home/vaibhav/sop_ws/build/px4; catkin build --get-env px4 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................................
Failed     << px4:make                     [ Exited with code 2 ]                                                                             
Failed    <<< px4                          [ 19.7 seconds ]                                                                                   
[build] Summary: 11 of 12 packages succeeded.                                                                                                 
[build]   Ignored:   None.                                                                                                                    
[build]   Warnings:  1 packages succeeded with warnings.                                                                                      
[build]   Abandoned: None.                                                                                                                    
[build]   Failed:    1 packages failed.                                                                                                       
[build] Runtime: 29.7 seconds total.
```
I'm not sure what exactly is causing this error, any leads and help would be highly appreicable,
TIA

TSC21, 2021-05-31:
Why did you build with `make` and then using `catkin`? What are you trying to do? Note that building the firmware using catkin hasn't been tested or validated for a long time already, so it's not currently considered a valid way builiding the PX4 firmware (for SITL).

adbidwai, 2021-05-31:
@TSC21 So am I just supposed to build it using ```make``` and leave it ? In case I need to build other packages in the workspace (other than PX4-Autopilot) it will always show a build error. Is that how it's supposed to be ?

TSC21, 2021-05-31:
You are not supposed to put it inside of a catkin workspace. This repo is not functioning correctly right now as a catkin package for a long time now, so you are advised to build it standalone, outside of and isolated from your ROS environment.

adbidwai, 2021-06-01:
@TSC21 Thanks a lot for the insights

TSC21, 2021-06-01:
You're welcome. Closing this as solved.

closed 2021-06-01 by TSC21