PX4/PX4-Autopilot #17685: Yaw setpoint spike, when landing triggered via Mavlink command MAV_CMD_NAV_LAND

ecmnet, 2021-06-01:
In 1.12.beta, triggering landing via MAV_CMD_NAV_LAND leads to a spike in yawn the beginning of the landing phase, which should not occur. This issue is independent from parameter 4 of MAV_CMD_NAV_LAND (setting it to NaN or current yaw):

[image]

[image]

ULog: [https://logs.px4.io/plot_app?log=07daba2b-95db-4f4b-b5e9-423ebb3f5fa0](https://logs.px4.io/plot_app?log=07daba2b-95db-4f4b-b5e9-423ebb3f5fa0)

dagar, 2021-06-01:
Looks like its jumping to your current heading.

[image]

ecmnet, 2021-06-02:
Possible reason might be that the yaw Setpoint is set to the current yaw measured:

[image]

LorenzMeier, 2021-06-02:
@MaEtUgR Do you have an idea what change set might have introduced this?

MaEtUgR, 2021-06-02:
I'm not aware of any particular change that would cause this. Navigator indeed just sets the setpoint to the current yaw estimate but there is(/was?) a slewrate applied on top that always results in feasible turn speeds configured by the maximum yawrate.

MaEtUgR, 2021-06-02:
I would first check if the slewrate is configured and initialized the correct way...

ecmnet, 2021-06-15:
I do not believe anymore that this is due to stepping changes. I saw that kind of spikes now during takeoff also.

Kim-zhi-jiang, 2021-06-16:
VTOL  v_master 
[image]

ecmnet, 2021-06-16:
UPDATE: Might be related to HIGH_RES SD logging. First checks show, that after setting the logging mode from 17 to 1, the spikes do not occur anymore.

ecmnet, 2021-06-17:
I can confirm now, that this issue is related to the switch of the yaw Setpoint to the current yaw. Might be solved by better tuning of the yaw controller.

MaEtUgR, 2021-06-21:
I haven't had time to look into details yet. My suspicion is that the yaw rate limit for waypoint
https://github.com/PX4/PX4-Autopilot/blob/155d3c7d367fced2830ff57031e346c2faa037c8/src/modules/flight_mode_manager/tasks/Auto/FlightTaskAuto.cpp#L93-L124
is not initialized the right way and that's why you might run into the issue.

ecmnet, 2021-07-07:
I do not understand why the setpoint is updated even if yaw is aligned

`if (!PX4_ISFINITE(_yawspeed_setpoint) && (_deltatime > FLT_EPSILON)) { 
 			// Create a feedforward 
 			_yawspeed_setpoint = dyaw / _deltatime; 
 		} `

It is hard to reproduce the issue in SITL but if the aligned condition is added, no setpoint change happens:

`if (!PX4_ISFINITE(_yawspeed_setpoint) && (_deltatime > FLT_EPSILON) && !_yaw_sp_aligned) {
			// Create a feedforward
			_yawspeed_setpoint = dyaw / _deltatime;
		}`

ecmnet, 2021-07-22:
Closing this as the spike does not occur anymore with v1.12-stable.

closed 2021-07-22 by ecmnet