PX4/PX4-Autopilot #17688: Autopilot enters Failsafe upon connecting to QGC

Eli-Voly, 2021-06-01:
**Describe the bug**
With 1.12-beta5 loaded the autopilot goes into failsafe and beeps with https://docs.px4.io/master/en/getting_started/tunes.html#battery-warning-fast tune continuously. This issue is the same as https://px4.slack.com/archives/C0W2KUFFT/p1622234288061900

**To Reproduce**
Steps to reproduce the behavior:
1. Power drone with either USB or battery
2. Open QGroundControl to connect to a GCS
3. To stop failsafe mode, turn on RC transmitter

**Expected behavior**
I don't expect the autopilot to enter failsafe mode when connecting to QGC.

**Log Files and Screenshots**
Flight Review: https://review.px4.io/plot_app?log=7b8ce0d1-52c8-4c32-ad14-55b48b99d5d7

**Drone (please complete the following information):**
- Pixhawk 4 with Pixhawk 4 GPS and magnetometer
- SBUS RC input

Eli-Voly, 2021-06-03:
I was also able to reproduce the same issue by plugging in a joystick then connecting to QGC. I think there is an issue with the way manual_control_setpoint is being handled causing it to failsafe. I am not sure if this is a problem with QGC or with PX4 firmware

rizvanovic, 2021-10-04:
Similar issue appears when using joystick (not virtual). The drone enters failsafe mode and has to be armed and disarmed for the beeping to stop. Currently looking into if it is safe to operate the drone with the failsafe having been triggered.

closed 2025-11-30 by github-actions[bot]