PX4/PX4-Autopilot #17691: px4_sitl gazebo ran out of RAM

cjffly, 2021-06-02:
I have similar issue with #14971, while I am using gazebo11 on Ubuntu 20.04. Seems make here does not recognize '-2'.

/PX4-Autopilot$ make px4_sitl gazebo -2
make: invalid option -- '2'
Usage: make [options] [target] ...

My PC has rayzon 5900, which is 12 core, with 16gb RAM. (Side question: should I upgrade to 32gb?)

dagar, 2021-06-03:
The argument is `-j2`.

cjffly, 2021-06-04:
The reason I put as `-2` is according to the last post in #14747. 

 I've also tried `-j2`, the result is ninja ignores the `-j2`, and use all the cores and got out of ram

cjffly, 2021-06-05:
Update: upgrading my ram from 16gb to 32gb did solve my problem. But there should be different ways to solve this...

offjangir, 2021-06-06:
The build is not over yet, let the machine run for a while I have good hardware specs it took about 40-45 mins after it hangs . make worked without specifying cores. Sit back and let it process. It will take a while  , it may take 1-1.5 hours for some people but works perfectly later.

cjffly, 2021-06-06:
> The build is not over yet, let the machine run for a while I have good hardware specs it took about 40-45 mins after it hangs . make worked without specifying cores. Sit back and let it process. It will take a while , it may take 1-1.5 hours for some people but works perfectly later.

If you mean to the gazebo build may take 40-45 min, that's too long, specially it freeze the PC and you can't do anything else... After upgrade the ram, the build only take me about 30 seconds. Sure I can let it run for 1 hour and hopefully it will work, but if lower the core counts would make the build much faster, why not?

offjangir, 2021-06-06:
Cause even after reducing the core count  it didnt work for me.
As mentioned by other people on different threads even -jN didnt work for them.
So what better way is there ?  Other then incresing your ram?

cjffly, 2021-06-06:
That is why I raised this issue: stating -jN didn't work. I am just saying increase ram is one quick but expensive way.

PR #18047 (2021-11-07): Allow overriding sitl_gazebo build_cores with an environment variable

Closes #17691

This is far from a perfect solution, it simply tries to offer a minimum of flexibility when deploying automated builds.

This allows the possibility of overriding the parallelism when building sitl_gazebo, which is memory intensive and can result in oom ( see #17691 , other issues have been closed in the past but not actually solved: #14747 #14430  ). This is especially the case as new CPUs have more and more logical cores, but systems still generally have 16GB or less of RAM. In my 16GB system, `master` curr [... cut]
closed 2021-11-07 by dagar