PX4/PX4-Autopilot #17700: Bulding custom firmware with FW_AIRSPD_MAX value above 40m/s on Cube Black

sa7ina, 2021-06-03:
Hello my friends

I am struggling with an issue.
I have a big bird that i can't unleash it's full potential due to firmware limitations.
This birdie: [](https://uavfactory.com/en/penguin-b-uav)
In manual flight mode i can reach 55ish m/s speed. but on all automated modes i have a ceiling of 40m/s.
When i try to change FW_AIRSPD_MAX above 40, QGC warns me and lets me save higher values. But during flight it ignores it completely. Closing and opening the throttle intermittently maintaining max 40m/s speed.

So i forked this repo and searched for a few hours where this limitation comes from but i cant find it anywhere in the code.
Changed the airframe value and 'make' completed without issues. But still... PX4 ignores it.
Found where to change the default value in: [](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_params.c) but 'make' fails because i exceeded the max: 40 value that i can't find anywhere in the code.

Any help will be much appreciated.

LOVE

dagar, 2021-06-03:
You should be able to force set the value higher in QGC, the min & max are soft limits only enforced by the GCS. https://github.com/PX4/PX4-Autopilot/blob/db1657fa843f943ccd3f5e9a517b970caa69ef61/src/modules/fw_pos_control_l1/fw_pos_control_l1_params.c#L456-L457

What does the QGC parameter tree show after you set the value about 40 and refresh? You can also do it directly the command line (`param set FW_AIRSPD_MAX 55`).

sa7ina, 2021-06-06:
When i set it to 55 it says that it above the max but lets me save the value.
When i use `param set FW_AIRSPD_MAX 55` it's doing the same.
But during flight i can't fly over 40. As soon i reach 40 throttle closing and opening maintaining 40.

I'm trying to set the max airspeed higher than 40 but i cant build it:
`Default value is larger than max: FW_AIRSPD_MAX default:55.0 max:40
[21/1341] Performing configure step for 'px4io_firmware'
-- PX4 version: v1.11.3-4-g9334573099
-- PX4 config file: /home/ei8ht/PX4-Autopilot/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3")
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-404-g395c8d3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ei8ht/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware
ninja: build stopped: subcommand failed.
Makefile:217: recipe for target 'px4_fmu-v3_default' failed
make: *** [px4_fmu-v3_default] Error 1
` 

Where does the "max:40" comes from?

dagar, 2021-06-06:
> Where does the "max:40" comes from?

Right here in the parameter metadata. https://github.com/PX4/PX4-Autopilot/blob/db1657fa843f943ccd3f5e9a517b970caa69ef61/src/modules/fw_pos_control_l1/fw_pos_control_l1_params.c#L456-L457

sa7ina, 2021-06-07:
OK 10x a lot!
I guessed since it was commented values came from elsewhere.

Solved!

closed 2021-06-07 by sa7ina