PX4/PX4-Autopilot #17705: link libssh library to Makefile of px4_sitl

ZineddineOuldimam, 2021-06-04:
Hi, 

I want to execute libssh library inside the px4_simple_app code to pass command from px4 stack to my host machine Ubuntu. I create a C code using libssh, and to be compiled I should link the code with libssh like this: `gcc test.c -o test -lssh `. The code works when I complied it outside px4 with gcc command in my terminal. However, it raises error when I launch px4 with the command `make px4_sitl_default jmavsim `. Those errors are the same when I run my code in my terminal without option `-lssh`. So how can add this option or flag to my compiler in Cmakefile of px4 ?

please help me

dagar, 2021-06-04:
As a quick fix you can add it here. https://github.com/PX4/PX4-Autopilot/blob/ad9288536b39183c6ec69e05c70b69de58073acf/platforms/posix/CMakeLists.txt#L58-L61

It can also be done at the module level DEPENDS (CMakelists.txt).

ZineddineOuldimam, 2021-06-04:
Thank you very much, that resolved my problem and made my day.

closed 2021-06-04 by ZineddineOuldimam