PX4/PX4-Autopilot #17719: PX4 SITL Custom model shaking instead of flying

Swell974, 2021-06-05:
Hello,
I wanted to use a custom model in Gazebo with PX4 SITL but it was weirdly shaking when I asked it to takeoff. So I wanted to ensure that I was able to fly a simple copy of the iris model (found in /PX4-autopilot/Tools/sitl_gazebo/models) but I got the same symptoms.

**To Reproduce**
I followed the indications given by antonerasm in https://discuss.px4.io/t/create-custom-model-for-sitl/6700/4, naming the iris copy 'iris2'.

> 1. Create a folder under Tools/sitl_gazebo/models for your model, let’s call it my_vehicle
> 2. Create the following files under Tools/sitl_gazebo/models/my_vehicle: model.config and my_vehicle.sdf (these can be based off the iris or solo models in Tools/sitl_gazebo/models)
> 3. Create a world file in Tools/sitl_gazebo/worlds called my_vehicle.world (Again, this can be based off the iris or solo world
> 4. Create an airframe file under ROMFS/px4fmu_common/init.d-posix (Yet again, this can be based off the iris or solo airframe files), give it a number (for example 4229) and name it 4229_my_vehicle
> 5. Add the airframe name (my_vehicle) to the file platforms/posix/cmake/sitl_target.cmake in the command set(models …

Since it is just a copy of the iris model, the behavior can't come from the SDF file or from bad controller gains. I don't understand what is happening.

Jaeyoung-Lim, 2021-07-07:
@Swell974 How are you starting the model?

Jaeyoung-Lim, 2021-07-15:
Closing as the issuer didn't respond. Please reopen if this is still an issue

closed 2021-07-15 by Jaeyoung-Lim