PX4/PX4-Autopilot #17722: Miltiple HW issues on CubeOrange

aet2230, 2021-06-06:
We use cube-orange on our drones, PX4 - v1.11.3
We encounter the following issues, we did not see on pixhawk cube-black

1. Local position invalid
- Drone may be armed, but do not takeoff. In stabilize/manual mode motors do not increase thrust with throttle motion
- mavlink messages of position and attitude are not published
- When connecting via nsh and checking ekf2 status we get local position invalid
- Software reboot do not help - only power off then on
2. RC do not respond
- Drone does not respond RC commands at all (modes, switches, arm)
- RC receiver led is green
- Tried to connect via QGC, reflash version and re-upload parameters. I got errors which point that dshot module is not running (unknown parameter DSHOT_CONFIG, DSHOT_MIN ...)
- Software reboot do not help - only power off then on
3. Uncontrolled flight
- Drone responded like its PID are very high (while before it flew ok)
- Software reboot do not help - only power off then on 

**Additional context**
Add any other context about the problem here.

dagar, 2021-06-06:
As a start please try v1.12 beta or master on the Cube Orange where quite a few issues have been fixed and it's now being regularly used.

From that point if you have any remaining problems please share a flight log and we can discuss in detail. https://logs.px4.io/

aet2230, 2021-06-14:
I cannot build it, I get the following errors

```
/src/lib/version/px_update_git_header.py", line 119
    """
      ^
SyntaxError: invalid syntax
```
what should be done?

mrpollo, 2021-08-31:
@aet2230 have you tried the latest stable are you still experiencing the problems from this issue?

aet2230, 2021-09-12:
@mrpollo , yes.
We did not try to reproduce all issues on v1.12, but reproduced Issue #2 above (RC do not respond).
Before trying v1.12, we found that all above issues occur after "SW reboot" via https://mavlink.io/en/messages/common.html#MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN message (sent from QGroundControl or from other application).
Our HW architecture allowed us to design power on-off to cube only, and this way there are no problems, so there is definitely a problem with the reboot via mavlink message.