PX4/PX4-Autopilot #17726: CUAV x7 Pro The master firmware enables camera triggering, causing no other PWM output

CUAVcaijie, 2021-06-07:
**Describe the bug**
There is no output from the PWM channel after the camera trigger is enabled

**test environment**
Test hardware: 
CUAV X7 Pro

Test firmware version:
FW git-hash: 66dbc1e25f6fc173cb0ca54f73723b7bbc460cf8

Test model: eight-axis
SYS_AUTOSTART = 8001
[image]

parameter settings:
CBRK_USB_CHK = 197848（(It can be unlocked normally when connected to the USB)）
CBRK_SUPPLY_CHK = 894281（It can be unlocked normally without connecting the ammeter）
RC_MAP_AUX1 = Channel5
RC_MAP_AUX2 = Channel6
RC_MAP_AUX3 = Channel7
RC_MAP_AUX4 = Channel8

**Test comparison**

1. PWM output when camera trigger is not enabled
Before unlocking, only the first eight channels have output (normal)
[image]

After unlocking: the first eight channels plus 4 RC channels (normal)
[image]

2. Enable PWM output when camera is triggered
Camera trigger settings:
[image]

DEBUG error message after enabling camera trigger:
[image]

Enable camera trigger-not unlocked: The 6th channel becomes the camera trigger channel. The first eight channels of PWM control have no output
[image]

Enable camera trigger-unlock: Same as unlocked, PWM has no output.
[image]

**The cause of this phenomenon**

1. Camera trigger channels should be the last few
2. In line 393 of file ROMFS/px4fmu_common/init.d/rcS. It is caused by setting the output mode to none. 

**modified code test**
https://github.com/cuav/Firmware/tree/camera_trigger_test
Set the camera to 2000 after trigger
Test Results：PWM OUT OK, camera trigger OK
[image]

CUAVcaijie, 2021-06-07:
Because the modification involves various motherboards, I hope to get help on the modification.

davids5, 2021-06-10:
@CUAVcaijie please test https://github.com/PX4/PX4-Autopilot/pull/17748

closed 2021-06-15 by CUAVcaijie