PX4/PX4-Autopilot #17732: Hexacopter crashed by just falling down (wq:attitude_ctrl low on stack)

drtrigon, 2021-06-08:
Last week I did a test flight in order to finish setting up the new FC (CUAV V5+) on our hexacopter.

While flying at one point the drone just dropped from sky (as if it was turned off).

I am looking at the logs now in order to try to figure out what happened. I still don't know what was the cause but noticed some things that look wrong to me:
1. we get warnings `[load_mon] wq:attitude_ctrl low on stack! (268 bytes left)` in the log file that covers the crash but the log files after the crash do not show this (also log files that are more in the past do not show this) - might that be the cause of the crash? (is this related to https://github.com/PX4/PX4-Autopilot/issues/14195?)
2. the log file does NOT show "ground_contact" and also does NOT show disarming of the FC
(I used PlotJuggler for analysis - also Flight Review shows landing detected only after enabling "Show additional data")

[log_21_2021-6-3-10-35-52.zip](https://github.com/PX4/PX4-Autopilot/files/6616402/log_21_2021-6-3-10-35-52.zip)

Help in figuring out what happened is needed. Can anyone help me figuring this out?

dagar, 2021-06-08:
For reference. https://logs.px4.io/plot_app?log=394c7e25-9867-4473-b274-280cfbbd7d9a

When the log stops you're still in the air with about 47% throttle.

[image]

> 1. we get warnings `[load_mon] wq:attitude_ctrl low on stack! (268 bytes left)` in the log file that covers the crash but the log files after the crash do not show this (also log files that are more in the past do not show this) - might that be the cause of the crash? (is this related to #14195?)

Unlikely the cause of the crash. These are harmless warnings as long as the number is > 0. It's fairly safe at this point to upgrade to v1.12 beta (about to be v1.12 stable) to silence the warning.

When the vehicle was on the ground did it still have power? Did you have a telemetry connection?

drtrigon, 2021-06-08:
> When the vehicle was on the ground did it still have power? Did you have a telemetry connection?

I did not check. But it is possible that power was gone. But I am a bit suprised by that, because if you look at the battery voltage is was fairly stable at that time and way higher than the 12 V drop in the beginning. Further more the FC was on a BEC and thus I would assume to see at least some part of the voltage drop down to 5 V on the log (even though the motors stopped).

What would you expect from your experience?

dagar, 2021-06-16:
> What would you expect from your experience?

I don't want to generalize without seeing your vehicle, but some kind of electrical or hardware problem leading to the flight controller briefly losing power is not uncommon.

Unless the firmware has been customized in some way I would be surprised if the either unmodified v1.11 or v1.12 crashed in flight like this. On the sd card do you see any hard fault logs?

Have you done any additional flight testing with v1.12?

drtrigon, 2021-06-28:
Thanks for your answer!

Meanwhile I was able to do more flight testing, and it show strong indication that you are right.

The thing that suprises me is that power just cut off that sudden. But it is possible due to me not caring to much for FC power during that test and thus beeing a bit wacky an that part.

Thanks!

closed 2022-03-21 by drtrigon