PX4/PX4-Autopilot #17745: cubepilot_cubeorange does not configure & build properly in VSCode GUI

Ionstream, 2021-06-10:
When selecting build targets in VSCode GUI, `cubepilot_cubeorange` target points to `cubepilot_cubeorange_test`. Current workaround is to call `make cubepilot_cubeorange` directly in terminal. Issue requested by @hamishwillee.

hamishwillee, 2021-06-16:
FYI, when this get fixed we can remove the note I put in for tracking in https://github.com/PX4/PX4-user_guide/commit/9775c2c4c4c236919604a1c073115e99af1982f8

dagar, 2021-06-16:
This is because `cubepilot_cubeorange_test` includes the serial console which many people use while debugging.

You can change this to `cubepilot_cubeorange_default` if that's what you prefer.

cubepilot_cubeorange_test = cubepilot_cubeorange_default + serial console + test material

Ionstream, 2021-06-16:
Ah, that explains it. The orange cube does not expose it's debug port without some disassembly and a special cable, so I guess that's where my efforts went wrong. Thanks for the clarification!

That said, I do have trouble with the build configuration when attempting to build `cubepilot_cubeorange_test`. Is there anything I'm missing here?
```
[cmake] Configuring incomplete, errors occurred!
[cms-driver] Error during CMake configure: 
	Error: Configuration failed.
	at CMakeServerClient._onMessage (/home/ionstream/.vscode/extensions/ms-vscode.cmake-tools-1.7.3/dist/main.js:44271:33)
	at CMakeServerClient._onMoreData (/home/ionstream/.vscode/extensions/ms-vscode.cmake-tools-1.7.3/dist/main.js:44244:18)
	at Socket.emit (events.js:315:20)
	at addChunk (internal/streams/readable.js:309:12)
	at readableAddChunk (internal/streams/readable.js:284:9)
	at Socket.Readable.push (internal/streams/readable.js:223:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:188:23)
```

dagar, 2021-06-16:
I don't think that error means much. Try wiping the build directory (or `make distclean`), then freshly open vscode and click on the cubepilot variant to initiate the configure.

PR #21042 (2023-02-01): vscode: change cubeorange build target
- Fixes typo in Cube Orange build target config

Fixes #17745
closed 2023-02-01 by dagar