PX4/PX4-Autopilot #17746: Heading timeout in FW flight

VTOLDavid, 2021-06-10:
**Describe the bug**
RTL is trigered in FW flight because timeout for heading in MC triggers, sending this messages:
[navigator] unable to reach heading within timeout
[commander] Mission cannot be completed

**To Reproduce**
Steps to reproduce the behavior:
Firmware: Tailsitter v1.11.3.
1. Set MIS_YAW_TMT to a value lower than the time to turn 180 degree in FW flight.
In mi case I had MIS_YAW_TMT set to 1.
2. Fly a mision with several alt loiters

You wil get a RTL with this messages:
[navigator] unable to reach heading within timeout
[commander] Mission cannot be completed

Seems that this error is triggered here:
[https://github.com/PX4/PX4-Autopilot/blob/release/1.11/src/modules/navigator/mission_block.cpp#L398](https://github.com/PX4/PX4-Autopilot/blob/release/1.11/src/modules/navigator/mission_block.cpp#L398)

**Expected behavior**
Ignore the time out since is FW flight.
Whould be nice also improve the description of MIS_YAW_TMT in the parameter reference, is not clear what a value of 0 does.
Whould be also nice to be able to ignore the transition heading for VTOLs.

**Log Files and Screenshots**
[https://logs.px4.io/plot_app?log=92979fd1-0fc9-44f9-9a14-2c80e6ed66d3](https://logs.px4.io/plot_app?log=92979fd1-0fc9-44f9-9a14-2c80e6ed66d3)

ryanjAA, 2022-01-30:
Is this still relevant?

sfuhrer, 2022-02-10:
I wasn't able to reproduce, so seems to have been fixed in the mean time.

closed 2022-02-10 by sfuhrer