PX4/PX4-Autopilot #17764: Build of v1.12.0-beta5 fail

aet2230, 2021-06-14:
I cannot build it, I get the following errors

```
/src/lib/version/px_update_git_header.py", line 119
    """
      ^
SyntaxError: invalid syntax
```

On other PC I get 

```
In file included from NuttX/nuttx/include/nuttx/irq.h:123:0,
                 from NuttX/nuttx/include/nuttx/sched.h:55,
                 from NuttX/nuttx/include/sched.h:49,
                 from NuttX/nuttx/include/stdio.h:48,
                 from ../../platforms/common/include/px4_platform_common/log.h:121,
                 from msg/topics_sources/debug_key_value.cpp:38:
NuttX/nuttx/include/arch/irq.h:54:10: fatal error: arch/chip/irq.h: No such file or directory
 #include <arch/chip/irq.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
Is there any  list of requirements??

mrpollo, 2021-06-14:
What commands did you try? Did you follow our user guide?

aet2230, 2021-06-15:
Read again
and now I get
./Tools/docker_run.sh make -j8 px4_fmu-v3_default upload
```
guessing PX4_DOCKER_REPO based on input
PX4_DOCKER_REPO: px4io/px4-dev-nuttx-focal:2021-04-29
Starting with UID : 1000
[0/1082] Performing build step for 'px4io_firmware'
[1/229] Generating nuttx/binfmt/libbinfmt.a
FAILED: NuttX/nuttx/binfmt/libbinfmt.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find binfmt -type f -name *.o -delete && make -C binfmt --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:82: libbinfmt.a] Error 127
[2/229] Generating nuttx/libs/libxx/libxx.a
FAILED: NuttX/nuttx/libs/libxx/libxx.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find libs/libxx -type f -name *.o -delete && make -C libs/libxx --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=n EXTRAFLAGS=
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:90: libxx.a] Error 127
[3/229] Generating apps/libapps.a
FAILED: NuttX/apps/libapps.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps && find /home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps -name *.o -delete && make --quiet --no-print-directory TOPDIR="/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx" > nuttx_apps.log
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[3]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[3]: *** [/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/Application.mk:145: archive] Error 127
make[2]: *** [Makefile:99: /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/libapps.a] Error 2
make[1]: *** [Makefile:53: all] Error 2
[4/229] Generating nuttx/boards/libboards.a
FAILED: NuttX/nuttx/boards/libboards.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find boards -type f -name *.o -delete && make -C boards --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:100: libboards.a] Error 127
[5/229] Generating nuttx/drivers/libdrivers.a
FAILED: NuttX/nuttx/drivers/libdrivers.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find drivers -type f -name *.o -delete && make -C drivers --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:120: libdrivers.a] Error 127
[6/229] Generating nuttx/mm/libmm.a
FAILED: NuttX/nuttx/mm/libmm.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find mm -type f -name *.o -delete && make -C mm --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=n EXTRAFLAGS=
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:73: libmm.a] Error 127
[7/229] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a 
cd /home/lab3/workspace/PX4-Autopilotr/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find fs -type f -name *.o -delete && make -C fs --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-ar: Command not found
make[1]: *** [Makefile:97: libfs.a] Error 127
[8/229] Generating nuttx/arch/arm/src/libarch.a
FAILED: NuttX/nuttx/arch/arm/src/libarch.a 
cd /home/lab3/workspace/PX4-Autopilot/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find arch/arm/src -type f -name *.o -delete && make -C arch/arm/src --quiet --no-print-directory all TOPDIR=/home/lab3/workspace/ForMerge/Their/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__ [... cut ...]
ninja: build stopped: subcommand failed.
make: *** [Makefile:221: px4_fmu-v3_default] Error 1
```

I have appropriate version:
arm-none-eabi-gcc-7.2.1 --version
```
arm-none-eabi-gcc-7.2.1 (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

dagar, 2021-06-16:
You have to quote the entire build command, and the upload isn't going to work from within the container unless you figure out the USB passthrough.

Try

``` Console
./Tools/docker_run.sh 'make cubepilot_cubeorange_default'
```

aet2230, 2021-06-16:
Solved! Thanks!

closed 2021-06-16 by aet2230