PX4/PX4-Autopilot #17767: QGC param download broken

julianoes, 2021-06-15:
**Describe the bug**
Something is broken with the MAVLink connection. I'm no longer able to connect properly to QGC. I get warnings about param and mission download failed.

**To Reproduce**
Steps to reproduce the behavior:
1. `make px4_fmu-v4_default upload`
2. Open QGC 4.1.3, connected via USB

**Expected behavior**
This used to just work.

**Log Files and Screenshots**

[image]

taileron, 2021-06-15:
I can confirm it. Master Firmware and QGC Daily. (generic Quadcopter Airframe 4001)
USB or Wifi Connection doesn´t change something. 
QGC 3.4.4 still works with the current PX4 master not depending on selected Airframe.
Various Vtol Airframes seem still working in any combination.

julianoes, 2021-06-15:
@taileron thanks for the comment! That's interesting, older 4.0.5 QGC does indeed still work with latest master PX4.

My `git bisect`ing was inconclusive:

```
723e803ab9813717847ebaad69fe735b798b139a is the first bad commit
commit 723e803ab9813717847ebaad69fe735b798b139a
Author: PX4 BuildBot <bot@px4.io>
Date:   Sun May 16 12:39:14 2021 +0000

    Update submodule public_regulated_data_types to latest Sun May 16 12:39:14 UTC 2021
    
        - public_regulated_data_types in PX4/Firmware (b00f416229a5bcef2bdad4ec29c90e4846c701af): https://github.com/UAVCAN/public_regulated_data_types/commit/1337b1c86fee5bd3f3c3c0f1027bcf19e5c08aae
        - public_regulated_data_types current upstream: https://github.com/UAVCAN/public_regulated_data_types/commit/0a773b93ce5c94e1d2791b180058cb9897fab7e1
        - Changes: https://github.com/UAVCAN/public_regulated_data_types/compare/1337b1c86fee5bd3f3c3c0f1027bcf19e5c08aae...0a773b93ce5c94e1d2791b180058cb9897fab7e1
    
        0a773b9 2021-05-10 Kalyan Sriram - reg/drone: remove redundant timestamp in Velocity3Var (#117)

 src/drivers/uavcan_v1/public_regulated_data_types | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

julianoes, 2021-06-16:
I tried again and this time I went further back and the problem didn't go away. I eventually tried v1.11, same problem. I tried another USB port (without hub), same problem.

I then plugged out my joystick, and all was good. And now I can't reproduce it anymore, even when plugging in the joystick again... :confused:

Pedro-Roque, 2021-06-16:
> I tried again and this time I went further back and the problem didn't go away. I eventually tried v1.11, same problem. I tried another USB port (without hub), same problem.
>
> I then plugged out my [...]

USB driver issue ?

taileron, 2021-06-16:
With virtual joystick and (ESP) wifi connection it was the same and after one time virtual joystick off and on the problem seems temporarely gone.

mrpollo, 2021-06-16:
@julianoes can you please have a second-pass and reproduce it with a virtual joystick, it would be great to know if this is a QGC or PX4 issue, it has the potential to block the v1.12 release

julianoes, 2021-06-16:
Looks like this is the same as reported here: https://github.com/mavlink/qgroundcontrol/issues/9022

julianoes, 2021-06-16:
It definitely looks like a QGC issue. As soon as I connect a joystick over USB and enable it, there is no more MAVLink traffic from QGC to PX4.

antonio-sc66, 2021-06-29:
Hi @julianoes, it's definitely a QGC issue, I've experienced the same problem: https://github.com/mavlink/qgroundcontrol/issues/9611. I think you can close this issue.