PX4/PX4-Autopilot #17769: Accel calibration gets stuck

julianoes, 2021-06-16:
**Describe the bug**
I tried to calibrate the accelerometer of my Pixracer and it seems to get stuck at the first step:

```
INFO  [commander] [cal] calibration started: 2 accel
INFO  [commander] [cal] pending: back front left right up down
INFO  [commander] [cal] hold vehicle still on a pending side
```

QGC says:
```
Start the individual calibration steps by clicking one of the buttons to the left.
[cal] calibration started: 2 accel
[cal] pending: back front left right up down
[cal] hold vehicle still on a pending side
[cal] calibration failed: reading sensor
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Pixracer with USB
2. Hit accel calibration in QGC

**Expected behavior**
It should work :man_shrugging:.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

mrpollo, 2021-06-16:
I successfully calibrated a mRo PixRacer Pro using QGC Daily, and the latest PX4 was recently pulled from upstream.

1. Connect pixracer with USB
2. Flash firmware `make mro_pixracerpro upload` (QGC flashing also works)
3. Ran accel calibrations

Note: that it took a while for the first step to begin.

You can download my log from here in case its helpful https://logs.px4.io/plot_app?log=7f5fd267-4b90-453e-9420-90e62a415409

julianoes, 2021-06-17:
```
update: 1s, num topics: 51
TOPIC NAME              INST #SUB RATE #Q SIZE
actuator_armed             0   13    2  1   16 
adc_report                 0    2  100  1   96 
battery_status             0    9  100  1  160 
commander_state            0    1    2  1   16 
cpuload                    0    5    2  1   16 
log_message                0    1    1  4  136 
manual_control_setpoint    0    3   25  1   64 
safety                     0    2    1  1   16 
sensor_accel               0    4  801  8   48 
sensor_accel               1    3  799  8   48 
sensor_baro                0    2   75  1   32 
sensor_gyro                0    4  801  8   48 
sensor_gyro                1    3  799  8   48 
sensor_mag                 0    3  146  4   48 
sensor_preflight_mag       0    1  146  1   16 
sensors_status_imu         0    1  100  1   88 
system_power               0    2  100  1   40 
telemetry_status           0    2    1  1   88 
telemetry_status           1    2    1  1   88 
telemetry_status           2    2    2  1   88 
tune_control               0    1   90  4   24 
vehicle_air_data           0   13   19  1   40 
vehicle_control_mode       0   17    2  1   24 
vehicle_land_detected      0    6    1  1   24 
vehicle_magnetometer       0    3   49  1   40 
vehicle_status             0   30    2  1   80 
vehicle_status_flags       0    5    2  1   48 
nsh> 
nsh> 
nsh> 
nsh> sensors status
INFO  [sensors] selected gyro: 0 (0)
INFO  [data_validator] validator: best: -1, prev best: -1, failsafe: NO (0 events)

INFO  [sensors] selected accel: 0 (0)
INFO  [data_validator] validator: best: -1, prev best: -1, failsafe: NO (0 events)

INFO  [vehicle_magnetometer] selected magnetometer: 65546 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator]  val:   0.2317, lp:   0.2334 mean dev:  -0.0001 RMS:   0.0020 conf:   1.0000
INFO  [data_validator]  val:  -0.0853, lp:  -0.0820 mean dev:   0.0007 RMS:   0.0033 conf:   1.0000
INFO  [data_validator]  val:   0.3817, lp:   0.3857 mean dev:  -0.0009 RMS:   0.0040 conf:   1.0000
INFO  [sensor_calibration] MAG 65546 EN: 1, offset: [ 0.000  0.000  0.000], scale: [ 1.000  1.000  1.000], Internal

INFO  [vehicle_air_data] selected barometer: 3997714 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 0, state: OK
INFO  [data_validator]  val: 96794.9922, lp: 96794.7891 mean dev:  15.3167 RMS:  37.7052 conf:   1.0000
INFO  [data_validator]  val:  40.1200, lp:  40.1319 mean dev:   0.0412 RMS:   0.1786 conf:   1.0000
INFO  [data_validator]  val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000

INFO  [sensors] Airspeed status:
INFO  [data_validator]  no data

INFO  [vehicle_acceleration] selected sensor: 0, rate: nan Hz, estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] ACC 0 EN: 1, offset: [0.0000 0.0000 0.0000] scale: [1.0000 1.0000 1.0000]

INFO  [vehicle_angular_velocity] selected sensor: 0, rate: nan Hz 
INFO  [vehicle_angular_velocity] estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] GYRO 0 EN: 1, offset: [0.0000 0.0000 0.0000]
```

closed 2021-06-17 by dagar