PX4/PX4-Autopilot #17770: No response for roll and pitch channel in manual flight mode

jiajianc, 2021-06-16:
**Describe the bug**
At begin, the RC worked well. Later, there are RC inputs in roll and pitch channel but the actuator control has no response.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=ab757bbe-0c5f-4445-a060-3ba2921bdd75

**Drone**
Tarrot 650

julianoes, 2021-06-18:
I can see that the flight mode is toggling between loiter and position control. Are you toggling that on purpose?

jiajianc, 2021-06-18:
> I can see that the flight mode is toggling between loiter and position control. Are you toggling that on purpose?

Yes, I tried to test the stability of loiter mode. I fixed this problem by tightening the socket on PPM sum receiver. It's weird that the pixhawk can receive the RC command but there is no input for PID tuning. If the PPM receiver didn't connect well, I tought it will lose all signals.

julianoes, 2021-08-30:
Sorry for the late reply. Did you figure this out in the meantime?