PX4/PX4-Autopilot #17780: How to spawn iris_downward_depth_camera using single_vehicle_spawn.launch ?

adbidwai, 2021-06-19:
Hello,
I want to spawn a drone using a downward-facing depth camera for altitude-based localization.
I think the model iris_downward_depth_camera from PX4-Autopilot/Tools/sitl_gazebo/models would be the most useful for me. I am using the multi_uav_mavros_sitl.launch file which inherently uses single_vehicle_spawn.launch. This launch file can spawn the iris drone but when I set the vehicle argument to iris_downward_depth_camera it results in an error saying it can't find the jinja.sdf file (which actually doesn't exist in the given path)

Help is really appreciated.

TIA

hcws, 2024-08-02:
I have the same question , do you already solve it？

adbidwai, 2024-08-03:
Hi @hcws !
It's been a while since I posted this question and I don't remember how I found the solution to this question. Sorry!