PX4/PX4-Autopilot #17787: Encountered "[IO] mixer upload fail" message and no PWM signal output

NAmmann, 2021-06-21:
**Bug description**
A few seconds after powering the UAV the ESCs made the typical beeping sound meaning that there is no valid PWM signal. Furthermore, we encountered the "[IO] mixer upload fail" message in QGC.

I doubt that this is a software bug. But please explain the conditions in which this failure might occur. Is it a SD card failure or some sort hardware failure caused by hot temperature? Is it possible that this kind of failure also occurs during flight? What is the safest way to proceed flight operation?

**To Reproduce**
The issue occured a few seconds after powering the UAV. It was a hot day with nearly 30°C, direct sun light and on an asphalt landing strip.

The issue could not be reproduced yet. We tried to reproduce even in similar weather conditions, with no success.

**Expected behavior**
The normal behaviour would be to not display the info message and all ESCs receive a valid PWM signal so they do not make the typical beeping sound.

**Log Files and Screenshots**
Since the issue occured prior to arming the UAV no log file is available.

**Drone:**
- DJI M600 airframe
- Pixhawk 4 flight controller
- PX4 v1.11.3

ThomasRigi, 2021-08-10:
I have something similar, but not exactly the same error message. Not sure if related or not. VTOL quadplane with Pixhawk 4 running v1.12.1.

After a PX4 reboot (sent over mavros) the MC mixer failed to load (MC mixer is on the IO) and the FW mixer did not apply the trims. I was still able to arm to get a log: https://logs.px4.io/plot_app?log=2b39c9d2-f5f6-4843-8417-6f87c3db5fb2. No motors spinning as no mixer loaded. 

From QGC mavlink console I got some additional info (using `dmesg`):

[image]
[image]
[image]
[image]

It looks like some characters got missing when transferring the mixer file to the IO. 

@bkueng would you have any idea what could have messed up?

bkueng, 2021-08-12:
@NAmmann the error comes from a failed data write from FMU to the IO. If you had the `dmesg` output that could contain more clues.

@ThomasRigi this one looks familiar. The IO can go into a bad state on a soft reboot. @davids5 did you ever look into this?

davids5, 2021-09-14:
This should have been fixed in https://github.com/PX4/PX4-Autopilot/pull/18147

bkueng, 2022-10-25:
Closing, please re-open if still an issue