PX4/PX4-Autopilot #17796: make: CMAKE_AR-NOTFOUND: Command not found

WangJintao1998, 2021-06-24:
I met this problem when I was using px4_fmu-v3 on ubuntu 16.04

[0/1024] Performing build step for 'px4io_firmware'
[1/229] Generating nuttx/boards/libboards.a
FAILED: NuttX/nuttx/boards/libboards.a 
cd /home/jin/Firmware_ws/src/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find boards -type f -name *.o -delete && make -C boards --quiet --no-print-directory all TOPDIR=/home/jin/Firmware_ws/src/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:100: recipe for target 'libboards.a' failed
make[1]: *** [libboards.a] Error 127

WangJintao1998, 2021-07-06:
Can anyone help me with this quetion?

dagar, 2021-07-06:
The toolchain hasn't been tested on 16.04 for quite some time, so it's not really supported. I would upgrade to 20.04 or you can build within docker.

With docker installed simply run.

``` Console
./Tools/docker_run.sh 'make px4_fmu-v3_default'
```

https://hub.docker.com/repository/docker/px4io/px4-dev-nuttx-focal/tags?page=1&ordering=last_updated

closed 2021-07-06 by dagar

WangJintao1998, 2021-07-07:
> The toolchain hasn't been tested on 16.04 for quite some time, so it's not really supported. I would upgrade to 20.04 or you can build within docker.
> 
> With docker installed simply run.
> 
> ```
> ./Tools/docker_run.sh 'make px4_fmu-v3_default'
> ```
> 
> https://hub.docker.com/repository/docker/px4io/px4-dev-nuttx-focal/tags?page=1&ordering=last_updated

I upgrade to 18.04 but it sitll happens. I have run bash ./ubuntu.sh and make px4_fmu-v3_default.

Jaeyoung-Lim, 2021-07-07:
@WangJintao1998 I am on 18.04 and I have no issues on building with `make px4_fmu-v3_default` 

Could you share the log of the problem you are experiencing?

WangJintao1998, 2021-07-07:
> @WangJintao1998 I am on 18.04 and I have no issues on building with `make px4_fmu-v3_default`
> 
> Could you share the log of the problem you are experiencing?

Sure. Though I don't really know where to find the log, I saw tow logs under nuttx, one is nuttx_context.log and the other is nuttx_olddefconfig.log. 
nuttx_content.log:
Create .version
Create version.h
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/../nuttx-config/include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/../nuttx-config/src
LN: arch/arm/src/chip to arch/arm/src/stm32
LN: /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/drivers/platform to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/drivers/dummy
LN: platform/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/platform/dummy
LN: platform/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/platform/dummy

nuttx_olddefconfig.log:
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/../nuttx-config/include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/../nuttx-config/src
LN: arch/arm/src/chip to arch/arm/src/stm32
LN: /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/drivers/platform to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx/drivers/dummy
LN: platform/board to /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/platform/dummy
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/examples
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/gpsutils
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/system/libuv
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/system
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/fsutils
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/interpreters
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/testing
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/graphics
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/netutils
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/canutils
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/wireless/ieee802154
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/wireless/bluetooth
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/wireless
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/industry
mkkconfig in /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps
DEBUG: kconfiglib kconfig-conf wrapper, arguments: --olddefconfig Kconfig

Jaeyoung-Lim, 2021-07-07:
@WangJintao1998 No, more on what you see on your terminal when the build fails

WangJintao1998, 2021-07-07:
> @WangJintao1998 No, more on what you see on your terminal when the build fails. As you shared initially

OK. 
[0/1296] Performing build step for 'px4io_firmware'
[1/229] Generating nuttx/boards/libboards.a
FAILED: NuttX/nuttx/boards/libboards.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find boards -type f -name *.o -delete && make -C boards --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:100: recipe for target 'libboards.a' failed
make[1]: *** [libboards.a] Error 127
[2/229] Generating nuttx/binfmt/libbinfmt.a
FAILED: NuttX/nuttx/binfmt/libbinfmt.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find binfmt -type f -name *.o -delete && make -C binfmt --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:82: recipe for target 'libbinfmt.a' failed
make[1]: *** [libbinfmt.a] Error 127
[3/229] Generating nuttx/libs/libxx/libxx.a
FAILED: NuttX/nuttx/libs/libxx/libxx.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find libs/libxx -type f -name *.o -delete && make -C libs/libxx --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=n EXTRAFLAGS=
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:90: recipe for target 'libxx.a' failed
make[1]: *** [libxx.a] Error 127
[4/229] Generating apps/libapps.a
FAILED: NuttX/apps/libapps.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps && find /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps -name *.o -delete && make --quiet --no-print-directory TOPDIR="/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx" > nuttx_apps.log
make[3]: CMAKE_AR-NOTFOUND: Command not found
make[3]: *** [archive] Error 127
make[2]: *** [/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/apps/libapps.a] Error 2
make[1]: *** [all] Error 2
[5/229] Generating nuttx/drivers/libdrivers.a
FAILED: NuttX/nuttx/drivers/libdrivers.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find drivers -type f -name *.o -delete && make -C drivers --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:120: recipe for target 'libdrivers.a' failed
make[1]: *** [libdrivers.a] Error 127
[6/229] Generating nuttx/mm/libmm.a
FAILED: NuttX/nuttx/mm/libmm.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find mm -type f -name *.o -delete && make -C mm --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=n EXTRAFLAGS=
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:73: recipe for target 'libmm.a' failed
make[1]: *** [libmm.a] Error 127
[7/229] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find fs -type f -name *.o -delete && make -C fs --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:97: recipe for target 'libfs.a' failed
make[1]: *** [libfs.a] Error 127
[8/229] Generating nuttx/sched/libsched.a
FAILED: NuttX/nuttx/sched/libsched.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find sched -type f -name *.o -delete && make -C sched --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:60: recipe for target 'libsched.a' failed
make[1]: *** [libsched.a] Error 127
[9/229] Generating nuttx/arch/arm/src/libarch.a
FAILED: NuttX/nuttx/arch/arm/src/libarch.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find arch/arm/src -type f -name *.o -delete && make -C arch/arm/src --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:158: recipe for target 'libarch.a' failed
make[1]: *** [libarch.a] Error 127
[10/229] Generating nuttx/libs/libc/libc.a
FAILED: NuttX/nuttx/libs/libc/libc.a 
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx && find libs/libc -type f -name *.o -delete && make -C libs/libc --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/ [... cut ...]
cd /home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/NuttX/nuttx && find libs/libc -type f -name *.o -delete && make -C libs/libc --quiet --no-print-directory all TOPDIR=/home/jin/Firmware/f/Firmware/build/px4_fmu-v3_default/NuttX/nuttx KERNEL=n EXTRAFLAGS=
make[1]: CMAKE_AR-NOTFOUND: Command not found
Makefile:138: recipe for target 'libc.a' failed
make[1]: *** [libc.a] Error 127
ninja: build stopped: subcommand failed.
Makefile:219: recipe for target 'px4_fmu-v3_default' failed
make: *** [px4_fmu-v3_default] Error 1

ninja-hu, 2021-07-15:
I am observing the same error on Ubuntu 20.04. Here is the log:
```
-- PX4 version: v1.12.0-1887-gf70b6fbf0b
-- PX4 config file: /home/jadhav/PX4-Autopilot/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding boards/px4/fmu-v4/bootloader/px4_fmu-v4_bootloader.bin -> /etc/extras/bootloader.bin
-- ROMFS:  Adding platforms/nuttx/init/stm32/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v4/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v4/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v4/init/rc.board_extras -> /etc/init.d/rc.board_extras
-- ROMFS:  Adding boards/px4/fmu-v4/init/rc.board_mavlink -> /etc/init.d/rc.board_mavlink
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jadhav/PX4-Autopilot/build/px4_fmu-v4_default
Scanning dependencies of target git_nuttx
Scanning dependencies of target git_nuttx_apps
Scanning dependencies of target flighttasks_generated
Scanning dependencies of target uorb_headers
[  0%] git submodule platforms/nuttx/NuttX/nuttx
[  0%] git submodule platforms/nuttx/NuttX/apps
[  0%] Generating Flight Tasks
[  0%] Generating uORB topic headers
[  0%] Built target git_nuttx_apps
[  0%] Built target git_nuttx
Scanning dependencies of target events_header
Scanning dependencies of target generate_topic_listener
[  0%] Built target flighttasks_generated
[  0%] Generating combined event json file
[  0%] Generating listener_generated.cpp
Scanning dependencies of target px4_config_file_target
[  0%] Built target px4_config_file_target
Scanning dependencies of target parameters_xml
[  0%] Generating serial_params.c
[  0%] Built target generate_topic_listener
[  0%] Generating px4 event header file
Scanning dependencies of target ver_gen
[  0%] Generating git version header
[  0%] Built target events_header
Scanning dependencies of target copy_bootloader_bin
[  0%] Copying /home/jadhav/PX4-Autopilot/boards/px4/fmu-v4/bootloader/px4_fmu-v4_bootloader.bin to ROMFS extras
[  0%] Built target copy_bootloader_bin
Scanning dependencies of target mixer_gen_6dof
[  0%] Generating mixer_multirotor_6dof.generated.h
[  0%] Generating parameters.xml
[  0%] Built target mixer_gen_6dof
Scanning dependencies of target mixer_gen
[  0%] Generating mixer_multirotor.generated.h
[... 43 lines cut ...]
[  1%] Generating nuttx/boards/libboards.a
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[4]: CMAKE_AR-NOTFOUND: Command not found
make[4]: *** [Makefile:100: libboards.a] Error 127
make[3]: *** [NuttX/CMakeFiles/nuttx_boards_build.dir/build.make:4643: NuttX/nuttx/boards/libboards.a] Error 2
make[2]: *** [CMakeFiles/Makefile2:19039: NuttX/CMakeFiles/nuttx_boards_build.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  1%] Generating nuttx/libs/libxx/libxx.a
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  2%] Generating nuttx/binfmt/libbinfmt.a
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  2%] Generating builtins_clean.stamp
[  3%] Built target builtins_clean_target
make[4]: CMAKE_AR-NOTFOUND: Command not found
make[4]: *** [Makefile:90: libxx.a] Error 127
make[3]: *** [NuttX/CMakeFiles/nuttx_xx_build.dir/build.make:85: NuttX/nuttx/libs/libxx/libxx.a] Error 2
make[2]: *** [CMakeFiles/Makefile2:19616: NuttX/CMakeFiles/nuttx_xx_build.dir/all] Error 2
make[4]: CMAKE_AR-NOTFOUND: Command not found
make[4]: *** [Makefile:82: libbinfmt.a] Error 127
make[3]: *** [NuttX/CMakeFiles/nuttx_binfmt_build.dir/build.make:111: NuttX/nuttx/binfmt/libbinfmt.a] Error 2
make[2]: *** [CMakeFiles/Makefile2:19262: NuttX/CMakeFiles/nuttx_binfmt_build.dir/all] Error 2
make[1]: *** [Makefile:140: all] Error 2
make: *** [Makefile:226: px4_fmu-v4_default] Error 2

```