PX4/PX4-Autopilot #17804: variation sensor_combined/accelerometer_integral_dt

NicolasM0, 2021-06-25:
**Describe the bug**
During some of my tests, I have done very long logs (I work on a tether drone that is able to fly several days continuously).

I noticed a strange patern in the accelrometer vibration metric while my autopilot did not move as it was on my desk during the weekend:
[image]

Digging a big, I have found that it comes from a non-constant accelerometer integration duration causing difference between accumalated acceleration:
[image]

I have repeated 3 long logs on the same autopilot (CUAV V5+) with px4 v1.11.3 : the integration duration variation appear at exactly 8590sec after boot on the three logs.
I have tested v1.12 in the same condition on another autopilot (still CUAV v5+) : the same variations appear at about 8590sec after boot but I also have some variation before that : 
[image]

It seems that the average integration duration is constant. I suppose that in averge, the correct amount of accelerometers sample is used.

I have nerver noticed that on v1.10 (not checked in a lot of logs).

Is it safe to fly more than 8590sec in v1.11 and v1.12 ?
I suppose yes as the integration is correct but what could happen if the variation continue to increase ?

dagar, 2021-06-25:
Are you able to share the full log (v1.12 ideally)?

In v1.12 we can check this across different IMUs (`vehicle_imu`) and I'd also like to know which sensor was your primary in the plots above.

NicolasM0, 2021-06-25:
> Are you able to share the full log (v1.12 ideally)?
> 
> In v1.12 we can check this across different IMUs (`vehicle_imu`) and I'd also like to know which sensor was your primary in the plots above.

The log is downloading but I'm not sure will be able to share it today (very bad connexion...) 
It appears on the three estimators:
[image]

NicolasM0, 2021-06-25:
estimator_selector_status_0.accel_device_id = 3932170

dagar, 2021-06-25:
> 3932170

3932170 = 0x3C000A

0x3C is the device type for the icm20689.
https://github.com/PX4/PX4-Autopilot/blob/9d3e18dbe10700a13f3048f2807dc4b1a62eee62/src/drivers/drv_sensor.h#L93

NicolasM0, 2021-06-25:
I will share the log monday.

here is the vehicle_imu_X.delta_velocity_dt:
[image]

zoom:
[image]

imu 2 is has some varition from the begining but is seems worst after 8590sec
imu 0 and 1 are have both the issue

dagar, 2021-06-25:
Ok, we'll dig into it more with the full log. One thing to note with the CUAV v5+ is that all 3 IMUs are on the same bus (SPI1), so it's possible they're interacting somewhat.

NicolasM0, 2021-06-25:
> > 3932170
> 
> 3932170 = 0x3C000A
> 
> 0x3C is the device type for the icm20689.
> 
> https://github.com/PX4/PX4-Autopilot/blob/9d3e18dbe10700a13f3048f2807dc4b1a62eee62/src/drivers/drv_sensor.h#L93

Thank you for this !

dagar, 2021-06-25:
The perf counters (`perf`) might be another clue before looking at the full log.

If the problem originates in the sensor itself the thing to start thinking about is if/how we can catch it at that level and trigger a reset.

https://github.com/PX4/PX4-Autopilot/blob/9d3e18dbe10700a13f3048f2807dc4b1a62eee62/src/drivers/imu/invensense/icm20689/ICM20689.cpp#L242-L264

dagar, 2021-06-25:
I haven't been able to reproduce this yet on my CUAV v5+.

dagar, 2021-06-26:
Still no failure on my v5+ after about 2 days. Did you reproduce this on multiple units?

NicolasM0, 2021-06-26:
Reproduced 100% but only over 2 units. I can try on another one.

I will try the patch monday. And share the logs.

NicolasM0, 2021-06-28:
here is the 1.12-beta5 log: https://drive.google.com/file/d/1SzOaOcWYg99Y-BOUoyc_pKfHt1Gutokz/view?usp=sharing

NicolasM0, 2021-06-28:
and one 1.11 log (with higher variation as it is longer): https://drive.google.com/file/d/1rn7GjICs4cpv_uJbhwRm-VdSup6YEipG/view?usp=sharing

closed 2021-07-02 by NicolasM0