PX4/PX4-Autopilot #17806: ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

SEUZTh, 2021-06-25:
When I run ```bash ubuntu_sim_ros_melodic.sh``` in a clean Ubuntu18.04, I met this bug.
```
Requirement already satisfied: six>=1.5 in /usr/lib/python2.7/dist-packages (from python-dateutil>=2.5.0->pandas) (1.11.0)
Installing collected packages: python-dateutil, pytz, pandas, MarkupSafe, jinja2, pyserial, pyyaml
Successfully installed MarkupSafe-1.1.1 jinja2-2.11.3 pandas-0.24.2 pyserial-3.5 python-dateutil-2.8.1 pytz-2021.1 pyyaml-5.4.1
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting pyulog
  Downloading pyulog-0.9.0.tar.gz (38 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-fuZlXO/pyulog/setup.py'"'"'; __file__='"'"'/tmp/pip-install-fuZlXO/pyulog/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-eJi4c1
         cwd: /tmp/pip-install-fuZlXO/pyulog/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-fuZlXO/pyulog/setup.py", line 14, in <module>
        import versioneer
      File "versioneer.py", line 1739
        file=sys.stderr)
            ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Installing jMAVSim simulator dependencies
Reading package lists... Done

```

885288421, 2021-08-20:
hi,I meet the same problem.What shall I do?

SEUZTh, 2021-08-21:
> hi,I meet the same problem.What shall I do?

```bash
sudo pip install future
sudo apt install python3-pip
pip3 install pyulog
```

closed 2021-08-21 by SEUZTh

Aiwenqiuyu, 2023-04-04:
hi,friends,I have try the three commands but there is still the problem,what should i do?