PX4/PX4-Autopilot #17807: Camera Hotshoe feedback not working on Cube Orange

divyanshu1234, 2021-06-26:
Setup 1:

FCU: Cube Orange
Firmware: PX4 master build (Board: cubepilot_cubeorange)
Trigger Pin: Aux 5
Feedback Pin: Aux 6
CAM_CAP_FBACK = 1
[log file](https://review.px4.io/plot_app?log=527184c2-b27e-46cb-84b3-ac9cbd77e01e)

Setup 2:

FCU: Holybro Pixhawk4 (FMUv5)
Firmware: PX4 master build (Board: px4_fmu-v5_multicopter)
Trigger Pin: Aux 5
Feedback Pin: Aux 6
CAM_CAP_FBACK = 1
[log file](https://review.px4.io/plot_app?log=9924248b-c6ee-47c4-aa75-0890232c8f3a)

Both the missions were run in SIH. No `CAMERA_CAPTURE` messages were generated in the case of the Cube, but the exact same setup works perfectly with the FMUv5. We have also tried the release/1.12 and stable release/1.11 build on the Cube, but the camera feedback still doesn't work.

To make sure that there is no fault in the hardware setup of the Cube, we flashed ArduPilot firmware on it and the `CAMERA_FEEDBACK` message was being generated.

mlaiacker, 2021-12-06:
Hi,
I had the same problem and the issue is that in board_config.h for the cube orange

    #define DIRECT_INPUT_TIMER_CHANNELS  6

is missing.

mlaiacker, 2022-01-27:
I created a pull request for that https://github.com/PX4/PX4-Autopilot/pull/19082 
and compiled it for me and can be found here 
https://github.com/mlaiacker/PX4-Autopilot/releases/tag/untagged-d22d4ed6cd804607b48c 

https://github.com/mlaiacker/PX4-Autopilot/releases/download/untagged-d22d4ed6cd804607b48c/cubepilot_cubeorange_default.px4

mlaiacker, 2022-01-28:
it is fixed in master

mlaiacker, 2022-01-28:
and working with my hotfix release based on v1.11 

  
    nsh> camera_capture status
    INFO  [camera_capture] Capture enabled : YES
    INFO  [camera_capture] Frame sequence : 2
    INFO  [camera_capture] Last trigger timestamp : 52305422 (752 ms ago)
    INFO  [camera_capture] Number of overflows : 0
    nsh>

mlaiacker, 2022-01-28:
to make it work in master:
AUX5 used for trigger the camera
AUX6 Hotshoe input 

param TRIG_PINS set to 5

elektra-sven, 2022-02-03:
I flashed your hotfix and set

param TRIG_PINS to 5
param CAM_CAP_FBACK to 1
param CAM_CAP_EDGE to Falling edge

and connected the trigger to aux5 and the capture to aux6 and rebooted. Triggering works fine but "camera_capture status" gives me:

[image]

"camera_capture on" doesn't work and I have no clue why not:

[image]

mlaiacker, 2022-02-03:
For v1.11 you have to set trig pins to 56

elektra-sven, 2022-02-03:
I set trig pins to 56 and went outside to get GPS 3D Lock but the problem persists:

[image]

elektra-sven, 2022-02-21:
It turned out that the camera_capture-module only works properly if triggering with pin 5 is NOT PWM-based (TRIG_INTERFACE < 4). For TRIG_INTERFACE=4, other pins (e.g. AUX2) have to be used for capturing as pin5 and pin6 are using the same internal timer.

closed 2025-11-30 by github-actions[bot]